sinara-hw/Booster #278: Macconfig default seems broken

hartytp, 2019-09-06:
On the new booster v1.4s we have (latest fw) running the vcp ‘macconfig default’ command changed the MAC address to something else. Running it seems to give the same MAC address for all boosters.

Cc @dnadlinger

dnadlinger, 2019-09-06:
In particular, it seems like it might pick up the wrong part of the device id or something like that. I've run the command on four newly refurbished ones we received, with device ids and `macconfig default` addresses as follows:

```
TechnoSystem | 003B00383037470535353239 | 37:30:39:32:35:35
TechnoSystem | 003B002E3037470535353239 | 37:30:39:32:35:35
creoTECH | 002500333436510A32323838 | 36:34:38:38:32:32
creoTECH | 003200223436510B32323838 | 36:34:38:38:32:32
```

This is with firmware version `RFPA v1.4.19248 2d079d8`.

dnadlinger, 2019-09-06:
Alternatively, this might be a documentation/CLI help issue – both @hartytp and I had assumed that the "default" MAC address is supposed the one the device is shipped with.

wizath, 2019-09-07:
I found the issue. Default mac address is set with CPU UUID bytes 6-11. I'll change it to reflect UUID from EEPROM at monday.

hartytp, 2019-09-07:
Thanks! I thought it was something like that. Can you prioritise finishing the docs on Monday so I can finish testing?

wizath, 2019-09-09:
Fixed

closed 2019-09-09 by wizath

hartytp, 2019-09-14:
@wizath one one of the v1.3 Boosters that technosystem shipped us a while ago (maybe 6 months), the MAC address after shipping was `36:34:39:35:38:32` after running `macconfig default` it's now `80:1F:12:57:1E:F3`.

Is that concerning? Or is this just because the older firmware used a MAC address based on the CPU UID rather than the one in the EEPROM?

wizath, 2019-09-14:
Nope, older ID was subset of CPU UUID, now is from EEPROM

hartytp, 2019-09-14:
makes sense. So, for our older units default won't be what I have in our logs. but, that's fine.

thanks for clarifying!