sinara-hw/Booster #281: version string

hartytp, 2019-09-10:
I'm not sure I understand this: https://github.com/sinara-hw/Booster/wiki/Remote-programming-interface#idn

My version string is `RFPA v1.4.19248 2d079d8, built Sep  5 2019 14:57:58, id 30374705, hv rev 4`

- Is `<version string>` really the firmware version string?
- are all the fields listed under **version string** part of the firmware version string (`<version string>`) or just general parts of the version string? e.g. is the second element really the device ID? AFAICT they are a mixture of the fields 
- should  `hv rev` be `hw rev`? Should that be 1.4 rather than 4 to match with the version on the scehmatics

wizath, 2019-09-11:
* `version string` is `RFPA v1.4.19248 2d079d8`. The ID field in this table was off, fixed.
* `hv rev` - fixed to `hw rev`
* Hardware revision is read from HWREV pins 0-2. I can add it to display like `1.4` with fixed `1.`

hartytp, 2019-09-11:
Thanks, that makes sense. 

> I can add it to display like 1.4 with fixed 1.

Yes, please! It's a small thing but it's good to be consistent with the hardware revisions. (Although arguably we should have called those v1, v2, ... but it's too late to change that now!)

hartytp, 2019-09-11:
So, `19248` is the `build number` in the table, which is actually a build date according to the wiki description, right?

wizath, 2019-09-11:
Ok, now looks fine. I'll add that `1.`

hartytp, 2019-09-11:
What about the build number? The description on the wiki says it's actually a build date as an integer. Is that correct? If so what's the point of including it when we have the full date string later on?

wizath, 2019-09-11:
Hmm, In fact there is no need to include this, since there is commit hash

hartytp, 2019-09-11:
Can you delete it from the fw and wiki then please?

wizath, 2019-09-11:
done

closed 2019-09-11 by hartytp