sinara-hw/Booster #283: commands should not return until action complete

hartytp, 2019-09-10:
```
    def test_enable(self):
        delay = 0.1
        for _ in range(100):
            self.dev.set_enabled(1, True)
            time.sleep(delay)
            self.assertTrue(self.dev.get_enabled(1))

            self.dev.set_enabled(1, False)
            time.sleep(delay)
            self.assertFalse(self.dev.get_enabled(1))

            self.dev.set_enabled(1, True)
            time.sleep(delay)
            self.assertTrue(self.dev.get_enabled(1))
```

This test fails with delay = 0, but passes with delay=0.1

It's really useful to have a guarantee that set commands only return a response after the action is completed so that users can establish a guarantee of causality.

wizath, 2019-09-11:
I don't get it - response is returned after command is completed, so there is need to wait a little bit

hartytp, 2019-09-11:
> I don't get it - response is returned after command is completed, so there is need to wait a little bit

I don't follow you here? Why do we need to wait a little bit? I would have thought that as soon as `set_enabled(ch, en)` returns `ok` `get_enabled(ch)` should return `en`, right?

wizath, 2019-09-11:
I can make it return response immediately, then eg. enable channels. Since the enable procedure has delays built in

hartytp, 2019-09-11:
@wizath to be clear, I do *not* want this to return immediately. I want to return only after the channel has been enabled.

The problem is that right now:
```
set_enabled(0, True)
assert get_enabled(0) == True
set_enabled(0, False)
assert get_enabled(0) == False
```
does not work. It only works if I add a delay between calling the `set` and `get` functions.

wizath, 2019-09-11:
But simple `set_enabled(0, True)` command will take about 100 ms to produce `OK` response, and that's why you have wait for `get_enabled(0)` to get results, because commands will line up in buffers and they will be executed one by one.

hartytp, 2019-09-11:
Sorry, I should have posted my driver. In pseudocode my driver does something like:
```python
def set_enabled(channel, enabled):
    self.socket.write( ... )
    response = self.socket.readline( ... )
    assert response.decode().strip().lower() == 'ok'
```

In other words, my driver blocks until it receives the 'ok' response, so the `get_enabled` function is not called until after the `ok` response has already been received.

hartytp, 2019-09-11:
I can post a full driver if that helps with your testing

wizath, 2019-09-11:
> I can post a full driver if that helps with your testing
Yes please

hartytp, 2019-09-11:
okay, one second.

hartytp, 2019-09-11:
```python
import serial
import collections

Version = collections.namedtuple("Version", ["hw_rev"])

class Booster:
    """ Booster 8-channel RF PA """
    def __init__(self, device):
        self.dev = serial.serial_for_url("socket://{}:5000".format(device))
        self.dev.flushInput()
        assert self.ping()

    def get_version(self):
        """ Returns the device version information as a named tuple """
        self.dev.write(b"*IDN?\n")
        idn = self.dev.readline().decode().strip().lower().split(',')

        if not idn[0].startswith("rfpa"):
            raise Exception(
                "Unrecognised device identity string: {}".format(idn))

        return Version(hw_rev = 1.4)  # WIP!

    def ping(self):
        """ Returns True if we are connected to a Booster """
        try:
            self.get_version()
        except Exception:
            return False
        return True

    def _cmd(self, cmd, channel, arg=None):

        if channel not in range(8):
            raise ValueError("invalid channel number {}".format(channel))

        if arg is None:
            cmd = "{} {}\n".format(cmd, channel)
        else:
            cmd = "{} {}, {}\n".format(cmd, channel, arg)
        self.dev.write(cmd.encode())

        response = self.dev.readline().decode().lower().strip()

        if '?' in cmd and not "error" in response:
            return response
        elif response == "ok":
            return

        raise Exception(
            "Unrecognised response to '{}': '{}'".format(cmd, response))

    def _query_bool(self, cmd, channel, arg=None):
        resp = self._cmd(cmd, channel, arg)
        if resp == "0":
            return False
        elif resp == "1":
            return True
        else:
            raise Exception(
                "Unrecognised response to {}: '{}'".format(cmd, resp))

    def _query_float(self, cmd, channel, arg=None):
        resp = self._cmd(cmd, channel, arg)
        try:
            return float(resp)
        except ValueError:
            raise Exception(
                "Unrecognised response to {}: '{}'".format(cmd, resp))

    def set_enabled(self, channel, enabled=True):
        """ Enables/disables a channel """
        cmd = "CHAN:ENAB" if enabled else "CHAN:DISAB"
        self._cmd(cmd, channel)

    def get_enabled(self, channel):
        """ Returns True is the channel is enabled """
        return self._query_bool("CHAN:ENAB?", channel)
```

wizath, 2019-09-11:
It seems like a marginal race condition, since
```
    def test_enable(self):
        t1 = time.time()
        self.dev.set_enabled(0, True)
        t2 = time.time()
        print('execution time {}'.format(t2-t1))

        t1 = time.time()
        self.assertTrue(self.dev.get_enabled(0))
        t2 = time.time()
        print('execution time {}'.format(t2 - t1))
```

Runs ok without delays on my PC

hartytp, 2019-09-11:
How many times did you try it?

Try this:
```
class TestBooster(unittest.TestCase):

    @classmethod
    def setUpClass(cls):
        cls.dev = Booster(ip)

   def test_enable(self):
        for _ in range(1000):
            chan = random.randint(0, 7)
            en = bool(random.getrandbits(1))
            print(chan, en)
            self.dev.set_enabled(chan, en)
            time.sleep(0.1)
            self.assertEqual(self.dev.get_enabled(chan), en)

if __name__ == "__main__":
    # to do: write tidy up method!
    unittest.main()
```

wizath, 2019-09-11:
It runs fine with delay. No red LEDs.

hartytp, 2019-09-11:
sorry, try without delay

hartytp, 2019-09-11:
```python
class TestBooster(unittest.TestCase):

    @classmethod
    def setUpClass(cls):
        cls.dev = Booster(ip)

   def test_enable(self):
        for _ in range(1000):
            chan = random.randint(0, 7)
            en = bool(random.getrandbits(1))
            self.dev.set_enabled(chan, en)
            self.assertEqual(self.dev.get_enabled(chan), en)

if __name__ == "__main__":
    # to do: write tidy up method!
    unittest.main()
```

wizath, 2019-09-11:
It fails without delay tho

hartytp, 2019-09-11:
good! Glad you can reproduce this. As you say, probably a race somewhere...

wizath, 2019-09-11:
HA! I got it. SCPI `chan:enab?` takes info from `channel[i].enabled` struct. Information for this class is provided by measurement task which runs in about 50 ms intervals. I've added updating `enabled` and `sigon` fields in `enable` and `disable` procedures and problem is solved (from my initial tests it seems so)

hartytp, 2019-09-11:
Good, I thought it would be something like that!

hartytp, 2019-09-11:
Can you check if this also affects other methods like interlock clearing, interlock threshold updating etc etc and then post a new fw binary for me to check plz?

wizath, 2019-09-11:
Interlocks are fine. Any other command just return measurements which are updated constantly

[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3600773/eth-scpi.dfu.zip)

hartytp, 2019-09-11:
ever seen this before:
```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0483
Match product ID from file: df11
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
file contains 1 DFU images
parsing DFU image 1
image for alternate setting 0, (1 elements, total size = 162776)
parsing element 1, address = 0x08000000, size = 162768
Download	[=                        ]   7%        12288 bytesdfu-util: dfuse_download: libusb_control_transfer returned -9
```

hartytp, 2019-09-11:
hmm...power cycled and now flashed correctly

hartytp, 2019-09-11:
Expanded randomized testing (WIP):
```python
import unittest
import time
import random
from oxart.devices.booster.driver import Booster

ip = "xxx.xxx.xxx.xxx"

class TestBooster(unittest.TestCase):

    @classmethod
    def setUpClass(cls):
        cls.dev = Booster(ip)
        print("Testing on: {}".format(cls.dev.get_version()))

    def _cmd(self, cmd):
        self.dev.dev.write((cmd+'\n').encode())
        resp = self.dev.dev.readline().decode().strip().lower()
        if "?" in cmd:
            return resp

        self.assertEqual(resp, "ok")

    def test_channel_validation(self):
        for chan in [0.1, -1, 8, "test", False, "1e0"]:

            self.dev.dev.write("CHAN:ENAB? {}\n".format(chan).encode())
            self.assertTrue("error" in self.dev.dev.readline().decode()
                                           .lower())

            # test a command that doesn't accept all as this follows a
            # slightly different codepath in firmware
            self.dev.dev.write("CHAN:DIAG? {}\n".format(chan).encode())
            self.assertTrue("error" in self.dev.dev.readline().decode()
                                           .lower())

    def test_enable(self):
        chan = random.randint(0, 8)
        en = bool(random.getrandbits(1))
        if chan == 8:
            if en:
                self._cmd("CHAN:ENAB all")
            else:
                self._cmd("CHAN:DISAB all")
            resp = self._cmd("CHAN:ENAB? all")
            expected = "0" if en is False else "255"
            self.assertEqual(resp, expected)
        else:
            self.dev.set_enabled(chan, en)
            self.assertEqual(self.dev.get_enabled(chan), en)
            self.dev.get_enabled(chan)

    def test_detect_channels(self):
        for chan in range(9):
            if chan < 8:
                self.assertTrue(self.dev.get_detected(chan))
            else:
                self.assertEqual(self._cmd("CHAN:DET? aLl"), "255")

    def test_fuzz(self):
        tests = [method_name for method_name in dir(self)
                  if callable(getattr(self, method_name)) \
                  and method_name.startswith("test_") \
                  and method_name != "test_fuzz"]
        num_iterations = 100
        for test_num in range(num_iterations):
            test_idx = random.randint(0, len(tests)-1)
            test = tests[test_idx]
            print("fuzz {} of {}: {}".format(test_num, num_iterations,
                                             tests[test_idx][5:]))
            getattr(self, test)()

if __name__ == "__main__":
    # to do: write tidy up method!
    unittest.main()
```

Now works without errors. Thanks!

closed 2019-09-11 by hartytp