sinara-hw/Booster #284: CHANnel:DIAGnostics? still references input interlock

hartytp, 2019-09-11:
should this bit field just be labelled as `reserved (1)` or something?

hartytp, 2019-09-11:
Sorry, these aren't bit fields so we should just delete that from the returned array and from the wiki...

hartytp, 2019-09-11:
maybe better to just replace it with input power.

Can you also add fan speed while you're at it?

wizath, 2019-09-11:
I would wait with removing input interlock signal till hardware will be removed. Just in case.

hartytp, 2019-09-11:
Okay, fine, I'll just leave it out of my python driver and add a note on the wiki. Can we have the fan and input power measurements for completeness here please?

wizath, 2019-09-11:
done

closed 2019-09-11 by hartytp

hartytp, 2019-09-11:
> done

fw please?

wizath, 2019-09-11:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3601483/eth-scpi.dfu.zip)

hartytp, 2019-09-11:
Nope, that build still doesn't (d52b929) seem to have the fan speed in `CHAN:DIAG?`