sinara-hw/Booster #285: I2C bus errors

hartytp, 2019-09-11:
cf https://github.com/sinara-hw/Booster/issues/282

NB since the DAC bias voltage is set by the I2C bus, this could easily have been the cause of all of those dead FETs...

First check is whether the pull-ups are high enough...

hartytp, 2019-09-11:
Maybe the switch locking up? Didn't we have some problem like that on Kasli?

wizath, 2019-09-11:
Switch is being resettes every channel switch

hartytp, 2019-09-11:
@dnadlinger which hw revision did you observe this issue with? So far I’ve only tested on v1.3 (just want to be sure which hw rev we’re dealing with as there were a few changes between revisions)

hartytp, 2019-09-11:
@dnadlinger confirmed he's using a TS v1.4 so this affects both hw revisions.

So, the summary is that:
- we see infrequent complete lockups of the I2C bus
- the switch is reset each time the channel is changed
- we do not see this even after a week or more continuous operation if the SCPI interface is not used
- my guess is that this is **not** a SI issue (e.g. a glitch when the channels are enable/disabled). IIRC the regulators already have SS enabled, so the transients should be too slow to cause this kind of issue (but correct me if I'm wrong @gkasprow)
- I suspect it must be a timing issue or maybe a race between the standard polling look and the commands triggered by the SCPI interface or something like that

gkasprow, 2019-09-11:
we can connect a scope with I2C detect capability and observe the last I2C transfer before it dies, and also I2C line levels during lockup.

hartytp, 2019-09-11:
In principle, yes. However, it took me quite a while to make it lock up last time. The lockups have generally (maybe always? not sure) been after I was fiddling with code, so it's possible that they are only in a response to malformed instructions. Maybe not, hard to tell without better statistics.

Anyway, given how hard it seems to reproduce this, I'd say it might be worth doing some general testing on the line first. Look at voltage levels, noise, timings etc while running some random commands on the I2C bus. See if anything looks suspicious.

hartytp, 2019-09-11:
I also wonder if you can tell what locked up from the code? Might be easier to debug if we knew what kind of fault we were dealing with.

wizath, 2019-09-13:
Every lock gets released after transaction, so that's not an issue.

hartytp, 2019-09-13:
One final question: have you reviewed the PCBs and checked the timings for *every* component and made sure that none of them are marginal? The traces are quite long (=relatively high capacitive loads), so I don't want to push the max clock frequencies of anything.

gkasprow, 2019-09-13:
1MHz was definitely too high clock rate. For such frequency, one needs to use much lower pullups, even 330Ohm.

ADS7924IRTET works up to 400kHz
DAC7571DBV: 3.5MHz but only in Fast mode (push-pull), standard mode up to 400kHz
TCA9548ARGER: 400kHz
MAX6639AEE+: 100kHz. It sits on the bus all the time.
MAX6642: 100kHz
AD5627RBCPZ: 3.4MHz but only in Fast mode (push-pull), standard mode up to 400kHz
MCP3221A5T-I/OT: 400kHz
24AA02E48T-I/OT: 400kHz
We should lower the frequency to 100kHz, at least for access to some ICS

hartytp, 2019-09-13:
ouch! No wonder we got bus errors when we over-locked the temperature sensor by an order of magnitude! oops...

What's the load capacitance/pull-up requirement for the 100kHz chips? We have quite long traces. I'd prefer things to be slow than to risk data corruption.

hartytp, 2019-09-13:
@gkasprow can you recommend a bus speed for @wizath to use. Once that's implemented, I'll continue testing.

gkasprow, 2019-09-13:
The traces capacitance are not higher than 100pF so I think we can easily go with 100kHz

hartytp, 2019-09-13:
@wizath unless you see any issues with using a 100kHz clock can you post a binary with that please? I’d like to finish my randomised testing tomorrow morning if I can

wizath, 2019-09-13:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3611468/eth-scpi.dfu.zip)

100k version

wizath, 2019-09-13:
Check if the error count are not excessive. There may be a need to increase timeout counters, since transmission speed is like 10x lower.

wizath, 2019-09-13:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3611506/eth-scpi.dfu.zip)

Just in case, timeout count is 10x larger

hartytp, 2019-09-13:
Is the error counter a timeout you added to the fw in case of stalled i2c transactions? How do I check it? Chan:diag?

wizath, 2019-09-13:
Each error which is listed in `i2cerr` command or parts of `chan:diag?` command is result of timeout counter being exceeded while performing various steps of I2C transaction

hartytp, 2019-09-13:
Ok thanks. I’ll use the new fw and watch for errors on the chan:diag.

hartytp, 2019-09-14:
Thinking about this more, I don't see any reason why an I2C buss error shouldn't trigger the error flag for a channel in the same way that ove-rtemperature does. This won't catch everything, but it would be a good start!

Assuming that's easy to implement, can do you it please?

hartytp, 2019-09-14:
Started fuzzing with: https://github.com/sinara-hw/Booster/issues/285#issuecomment-531358577 (no increased timeout). I wasn't able to produce any I2C bus errors, but my testing wasn't too extensive since the SCPI interfaces hangs too often. The same with https://github.com/sinara-hw/Booster/files/3611506/eth-scpi.dfu.zip

So, provisionally, both firmware revisions you posted seem fine. If there is no harm having slightly longer timeouts then I suggest we keep the longer timeouts.

closed 2019-09-14 by hartytp