sinara-hw/Booster #289: booster fails to resond to commands

hartytp, 2019-09-11:
Much rarer than it used to be, but I've definitely seen it happen ~a few times today.

hartytp, 2019-09-14:
@wizath this is definitely still happening and is a pain. e.g. makes my randomized testing hang while waiting for a response to commands that never comes...

wizath, 2019-09-14:
Yep, I know working on it. You can set up timeout for now and reconnect if it happens.

hartytp, 2019-09-16:
any luck with this?

wizath, 2019-09-16:
https://forum.wiznet.io/t/topic/2609
It's not new issue with w5500, looking for possible fixes

hartytp, 2019-09-16:
uuurgh...can you have a background poll loop that picks up any missed interrupts?

hartytp, 2019-09-16:
(by checking the registers to see if there is any data remaining?)

wizath, 2019-09-16:
I've added additional checking if the w5500 has any RX data to process inside task, polling will loop in 100 ms intervals while waiting for interrupt. I'm currently running scripts to test the interface. So far on 100kish packets no failed response. attaching modified firmware if you want to test

[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3615530/eth-scpi.dfu.zip)

hartytp, 2019-09-16:
Okay. To be clear, the downside of this is that it means that sometimes ethernet packets will take ~100ms to be received instead of being received quickly? That's a shame but if you can't see a better way around this issue then we can live with it...

wizath, 2019-09-16:
That's the issue with the w5500, can't do anything more without sacrificing excessive CPU usage for watching at RX queue.

hartytp, 2019-09-16:
Okay, understood. So the conclusion is that if we want to make this faster then we should consider replacing the Wiznet chip with a small PCB with a STM32 microprocessor and implement this ourselves? That's something that can be done easily later if we ever need to since these PCBs are nice and cheap.

wizath, 2019-09-16:
We can hook up just ETH PHY to existing CPU and it'll be fine. Just need to reorder some pins to make room for connections.

Another thing - I've moved delay between clearing the interrupt and checking it again. That little bit of time could help, since as for now I've got 0 packets with access time larger than 50 ms.

hartytp, 2019-09-16:
To be clear, your suggestion is that we could replace the Wiznet chip with a standard phy and route the output of that to the microprocessor via the ribbon cable? Would that work in terms of SI?

If so, let's make this an option on the next hw rev...

wizath, 2019-09-16:
Firstly, lets test this thoroughly. As my testing goes, still no response above 50 ms and it's been running for like half an hour now

hartytp, 2019-09-16:
okay, nice!

hartytp, 2019-09-16:
still getting freeze ups on `Testing on: Version(fw_rev='v1.4.1', fw_hash='0d48c19', fw_build_date=datetime.datetime(2019, 9, 16, 11, 34, 56), device_id='34364708', hw_rev='hw rev 1.3')
`

wizath, 2019-09-16:
Post me your script please

hartytp, 2019-09-16:
Driver is: https://github.com/sinara-hw/Booster/issues/299#issue-493612868

Test script is:
```python
import argparse
import unittest
import time
import random
import sys

from oxart.devices.scpi_synth.driver import Synth
from oxart.devices.booster.driver import Booster

parser = argparse.ArgumentParser(description="Booster hardware in the loop "
                                 "test suite. Connect one channel of Booster "
                                 "to a synth and RF power meter")
parser.add_argument("--read_only", help="if set, no tests are run which would "
                                         "change the sate of any hardware. "
                                         "This mode allows testing multiple "
                                         "SCPI connections simultaneously.")
parser.add_argument("--booster", help="IP address of the Booster to test")
parser.add_argument("--synth", help="Address of the synth to use for tests")
parser.add_argument("--meter", help="Address of the power meter to use for "
                                    "tests")
parser.add_argument("--gain", help="Power out of booster = (power set on synth"
                                   " + gain)")
parser.add_argument("--chan", help="Booster channel connected to synth + "
                                    "power meter")
args = None

class TestBooster(unittest.TestCase):

    @classmethod
    def setUpClass(cls):
        cls.dev = Booster(args.booster)

        cls.synth = Synth(args.synth)
        cls.synth.set_freq(200e6)
        # self.meter.set_freq(200)

        cls.I29V = [0.]*8
        cls.I6V = [0.]*8

        for channel in range(8):
            cls.dev.set_enabled(channel)
        time.sleep(1)
        for channel in range(8):
            status = cls.dev.get_status(channel)
            cls.I29V[channel] = status.I29V
            cls.I6V[channel] = status.I6V

        print("Testing on: {}".format(cls.dev.get_version()))
        print("29V currents: {}".format(cls.I29V))
        print("6V currents: {}".format(cls.I6V))

    def assertRange(self, val, min, max):
        self.assertTrue(val >= min and val <= max)

    def assertApproxEq(self, a, b, eps):
        if abs(a - b) >= eps:
            print("fail!", a, b)
        self.assertTrue(abs(a - b) < eps)

    def _cmd(self, cmd):
        self.dev.dev.write((cmd+'\n').encode())
        resp = self.dev.dev.readline().decode().strip().lower()
        if "?" in cmd:
            return resp
        self.assertEqual(resp, "ok")

[... 138 lines cut ...]
    #             if self.dev.get_interlock_status(hitl_channel):
    #                 break
    #         else:
    #             self.assertTrue(0)  # more idiomatic way of doing this?

    def test_fuzz(self):
        skip = ["test_fuzz"]
        tests = [method_name for method_name in dir(self)
                  if callable(getattr(self, method_name)) \
                  and method_name.startswith("test_") \
                  and method_name not in skip]
        num_iterations = 100
        for test_num in range(num_iterations):
            test_idx = random.randint(0, len(tests)-1)
            test = tests[test_idx]
            print("fuzz {} of {}: {}".format(test_num, num_iterations,
                                             tests[test_idx][5:]))
            getattr(self, test)()

if __name__ == "__main__":
    args = parser.parse_args()
    print(sys.argv[0])
    unittest.main(argv=sys.argv[0:1])

```

hartytp, 2019-09-16:
@wizath NB I'm also seeing occasional failures in this test bench where the measured 29V current is quite a long way from where it should be (I saw 100mA on one channel once).

hartytp, 2019-09-16:
Is this definitely the right fw version to test? https://github.com/sinara-hw/Booster/issues/289#issuecomment-531731262

hartytp, 2019-09-16:
@wizath 

```
logstash
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
```

hartytp, 2019-09-16:
```
ethdbg
getSn_SR 20
getSn_MR 1
getSn_RX_RSR 0
getSn_CR 0
getSn_IR 0
getSn_RXBUF_SIZE 2
> 
> 
```

hartytp, 2019-09-16:
again
```
ethdbg
getSn_SR 20
getSn_MR 1
getSn_RX_RSR 0
getSn_CR 0
getSn_IR 0
getSn_RXBUF_SIZE 2
> 
> 
```

wizath, 2019-09-16:
Got one on my own, but by looking at `logstash` I've got `[WARN] watchdog reset occurred!`. HMM it seems like something is freezing up. Gonna disable it and investigate further. Could you check logstash for watchdog events?

gkasprow, 2019-09-16:
The easiest way of doing this is to connect another PHY like ENC28J60 with just SPI.
I'm not sure if we can easily connect the entire RMII interface to already routed board.

gkasprow, 2019-09-16:
There is also 100Mbit version, ENC424J60. One can easily buy modules with such MAC/PHY.

wizath, 2019-09-16:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3616651/eth-scpi.dfu.zip)

Ok, the SPI DMA's transfers timeouts are lowered and each timeout should post an error to stash. Also lowered the SPI clock since that could cause timeouts like I2C was.

I'm at 554/999 your fuzzing tests and still going

wizath, 2019-09-16:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3616707/eth-scpi.dfu.zip)
* Changed SPI speed, decreased TIMEOUT counters, errors logged to logstash
* I2C proper 100k speed, errors logged to logstash

In any case of TX error, there should now be an appropriate error logged

hartytp, 2019-09-16:
@wizath This is interesting...

I was fuzzing and the fuzzer froze. While fuzzing I had a VCP connection. At the same time the fuzzer froze, the VCP connection died with `serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)`

Logstash gives `logstash
[INFO] hardware revision 3
[WARN] watchdog reset occurred!
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1980 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] MACADDR 36:34:39:35:38:32
[INFO] RF Channel at 0 detected, id = B2:13
[INFO] RF Channel at 1 detected, id = B1:CF
[INFO] RF Channel at 2 detected, id = 90:A2
[INFO] RF Channel at 3 detected, id = 69:E0
[INFO] RF Channel at 4 detected, id = 6D:76
[INFO] RF Channel at 5 detected, id = 56:81
[INFO] RF Channel at 6 detected, id = 3C:9C
[INFO] RF Channel at 7 detected, id = 39:6B
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.233
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
[INFO] [tempmgt] Temp 109.000000 (d 100) -> Fan 100
[INFO] [tempmgt] Temp -54.500000 (d 80) -> Fan 20
[INFO] [tempmgt] Temp 54.500004 (d 52) -> Fan 72
[INFO] [tempmgt] Temp 27.250000 (d 52) -> Fan 20
> 
`

`
> ethdbg
getSn_SR 20
getSn_MR 1
getSn_RX_RSR 0
getSn_CR 0
getSn_IR 0
getSn_RXBUF_SIZE 2
> 
`

wizath, 2019-09-16:
You've got `[WARN] watchdog reset occurred!`

hartytp, 2019-09-16:
Yes. So, the issue here is the watchdog triggering. Also, notice that I got another negative temperature reading, but no error.

hartytp, 2019-09-16:
Okay, fuzzing with `Testing on: Version(fw_rev='v1.4.1', fw_hash='218aca0', fw_build_date=datetime.datetime(2019, 9, 16, 16, 23, 32), device_id='34364708', hw_rev='hw rev 1.3')```

hartytp, 2019-09-16:
Still more non-physical temperatures:
```
logstash
[INFO] hardware revision 3
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1980 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] MACADDR 36:34:39:35:38:32
[INFO] RF Channel at 0 detected, id = B2:13
[INFO] RF Channel at 1 detected, id = B1:CF
[INFO] RF Channel at 2 detected, id = 90:A2
[INFO] RF Channel at 3 detected, id = 69:E0
[INFO] RF Channel at 4 detected, id = 6D:76
[INFO] RF Channel at 5 detected, id = 56:81
[INFO] RF Channel at 6 detected, id = 3C:9C
[INFO] RF Channel at 7 detected, id = 39:6B
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.233
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
[INFO] [tempmgt] Temp 109.000000 (d 100) -> Fan 100
[INFO] [tempmgt] Temp -54.500000 (d 80) -> Fan 20
[INFO] [tempmgt] Temp 54.500004 (d 52) -> Fan 72
[INFO] [tempmgt] Temp 27.250000 (d 52) -> Fan 20
[INFO] network client 10.255.6.130 connected
[INFO] [tempmgt] Temp 27.322464 (d 20) -> Fan 0
[INFO] [tempmgt] Temp 27.682981 (d 20) -> Fan 20
[INFO] network client disconnected
> 
```

wizath, 2019-09-16:
Oh, that print `[INFO] [tempmgt] Temp -54.500000 (d 80) -> Fan 20` is from temperature management. That don't display real temperature but an averaged value. That's why no error is produced since this is not a real one.

hartytp, 2019-09-16:
Okay. How does an average go negative however??

hartytp, 2019-09-16:
Same again I'm afraid: VCP disconnected, ethernet connection lost. BTW I've been mainly testing with the VCP `start` task running in the BG. Not sure if that makes a difference?

wizath, 2019-09-16:
Oh, fixed small issue. I've gathered temperature from all channels (even broken). Changed it to `enabled` only. That could cause that error.

hartytp, 2019-09-16:
sorry, ethernet connection lost means no response to query.

hartytp, 2019-09-16:
as expected:
```
logstash
[INFO] hardware revision 3
[WARN] watchdog reset occurred!
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1981 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] MACADDR 36:34:39:35:38:32
[INFO] RF Channel at 0 detected, id = B2:13
[INFO] RF Channel at 1 detected, id = B1:CF
[INFO] RF Channel at 2 detected, id = 90:A2
[INFO] RF Channel at 3 detected, id = 69:E0
[INFO] RF Channel at 4 detected, id = 6D:76
[INFO] RF Channel at 5 detected, id = 56:81
[INFO] RF Channel at 6 detected, id = 3C:9C
[INFO] RF Channel at 7 detected, id = 39:6B
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.233
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
[INFO] [tempmgt] Temp 109.000000 (d 100) -> Fan 100
[INFO] [tempmgt] Temp -54.500000 (d 80) -> Fan 20
[INFO] [tempmgt] Temp 54.500004 (d 52) -> Fan 72
[INFO] [tempmgt] Temp 27.250000 (d 52) -> Fan 20
> 
> 
```

hartytp, 2019-09-16:
any other diagnostics you want?

wizath, 2019-09-16:
Nope, it's the watchdog - happened to me like a second ago. I'll disable it and try to catch again. Maybe it's reset interval is too low.

wizath, 2019-09-16:
But now I know how to reproduce error - just use `start` with fuzzing

hartytp, 2019-09-16:
yes, I saw that.

hartytp, 2019-09-16:
NB @dnadlinger also reported that the `macconfig default` command sometimes caused his VCP connection to die. I believe that was also connected to the watchdog (although I didn't get a log).

wizath, 2019-09-16:
I've got potential issue. System faults while using simultaneously VCP and SCPI at high intervals. Now need to resolve this issue.

Anyway that has no connection to losing interrupts with w5500, no need to double check received packets.

hartytp, 2019-09-16:
> Anyway that has no connection to losing interrupts with w5500, no need to double check received packets

Cool, so do we revert some of those changes?

wizath, 2019-09-16:
I'll check after fixing this issue

wizath, 2019-09-16:
As I started debugging this problem and it seems that multiple access to `sprintf` (which both VCP and SCPI for `*IDN?` use) could possibly lead to HardFault for dynamic memory allocation :scream:. At least it leads somewhere.

hartytp, 2019-09-16:
fun

wizath, 2019-09-16:
so much

hartytp, 2019-09-16:
This is why @sbourdeauducq et al are so keen on Rust. I'm starting to agree with them, although still fighting the language quite a bit

wizath, 2019-09-16:
With removal of printing floats in `start` command it seems to work fine.

hartytp, 2019-09-16:
.....why is that? Is it possible that this bug will manifest in different ways? e.g. two SCPI connections both querying the power etc. Is this a bug we actually understand now?

wizath, 2019-09-16:
I think so, we'll see tomorrow

hartytp, 2019-09-16:
good, thanks! So, what was the underlying issue then?

wizath, 2019-09-16:
Thread-safe dynamic memory allocation

hartytp, 2019-09-16:
Been a while since I wrote embedded C, but isn't it generally best practice to only use snprintf with static buffers? If so, a quick find and replace for sprints should guarantee that this issue doesn't come back in some form later on.

hartytp, 2019-09-16:
(basically, can you do a code review and try to make sure there are no more possible dynamic memory allocation multi-threading issues plz?)

wizath, 2019-09-16:
It's problem with `printf` which causes this issue. I use static buffers where I can. Anyway I'll review it after I'll fix this issue

hartytp, 2019-09-16:
ok, cool.

hartytp, 2019-09-17:
@wizath once you've finished reviewing the code to check for threading issues like this, can you have a go at implementing:
- https://github.com/sinara-hw/Booster/issues/305
- https://github.com/sinara-hw/Booster/issues/304
- https://github.com/sinara-hw/Booster/issues/300

(i.e. make sure that all serious error conditions result in channels being shut down and a red ERROR LED).

Once that's done, I should be able to finish testing the SCPI interface...

[35 more replies]

closed 2019-10-01 by wizath