sinara-hw/Booster #294: Not killing flash...

hartytp, 2019-09-13:
@wizath a question for you: I'm going to do more randomized testing soon. However, I'm conscious that, the interlock is automatically saved in flash on each update. So, if I set hundreds of different interlock values and check they trip, that's actually quite a lot of flash writes.

My questions are:
1. How many times can one safely update the interlock before hitting flash write limits?
2. Is there anything other than interlock threshold that is accessible via SCPI and is saved in flash? (I won't fuzz the VCP interface since it's only a debug port)

wizath, 2019-09-13:
> • More than 1 Million Erase/Write Cycles

hartytp, 2019-09-13:
Okay, that's great then.

closed 2019-09-13 by hartytp

wizath, 2019-09-13:
> Is there anything other than interlock threshold that is accessible via SCPI and is saved in flash? (I won't fuzz the VCP interface since it's only a debug port)

Nope, only interlock level is saved to EEPROM