sinara-hw/Booster #295: detect bus errors

hartytp, 2019-09-13:
@wizath how do you detect I2C bus errors? Is there an easy way to make it obvious to the user when a bus error occurs?

wizath, 2019-09-13:
Based on timeouts on several I2C peripherial flags. I can add it to logs if it helps

hartytp, 2019-09-13:
I2c errors are potentially really serious and could cause catastrophic damage (wrong bias current, wrong interlock threshold). I wonder if all channels should shut down (red leds etc) when they occur. Users won’t notice something in the logs if there isn’t an obvious symptom.

hartytp, 2019-09-13:
Actually unless there is a way we can be sure that no data corruption has occurred on the dac (short of measuring the output voltage) there isn’t any point

closed 2019-09-13 by hartytp

wizath, 2019-09-13:
We can just connect DAC output to unused ADC input to make 100% sure that voltage is correct

hartytp, 2019-09-13:
Not enough. We’d also have to check interlock thresholds etc

This should just work so we shouldn’t need diagnostics. Let me test it properly