sinara-hw/Booster #297: fanlevel doesn't work

hartytp, 2019-09-14:
```
> start
␛[2J
PGOOD: 1
> 
> FAN SPEED: 0 %
AVG TEMP: 25.50 CURRENT: 25.50
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		B2:13	B1:CF	90:A2	69:E0	6D:76	56:81	3C:9C	39:6B	
INPWR [V]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
TXPWR [V]	0.01	0.03	0.01	0.01	0.01	0.01	0.01	0.01	
RFLPWR [V]	0.01	0.00	0.01	0.01	0.01	0.02	0.02	0.02	
INPWR [dBm]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
TXPWR [dBm]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
RFLPWR [dBm]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
I30V [A]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	
I6V0 [A]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	
5V0MP [V]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	
ON		0	0	0	0	0	0	0	0	
SON		0	0	0	0	0	0	0	0	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		9	55	10	10	22	18	11	9	
ADC2		8	8	8	9	8	36	39	31	
INTSET [dBm]	30.70	30.00	30.49	30.00	33.00	33.00	33.00	31.50	
DAC1		4095	4095	4095	4095	4095	4095	4095	4095	
DAC2		3319	3444	3328	3259	3566	3600	3608	3451	
SCALE1		83	87	87	85	85	87	86	87	
OFFSET1		613	693	562	600	656	635	681	594	
BIASCAL		1811	1960	1690	1700	1712	1687	1657	1756	
HWIS		86.00	90.00	89.00	87.00	87.00	87.00	86.00	88.00	
HWIO		679.00	744.00	614.00	649.00	695.00	729.00	770.00	679.00	
LTEMP		25.50	24.00	25.00	24.00	25.00	25.00	25.50	24.00	
RTEMP		23.00	25.00	25.50	25.00	25.50	25.50	24.00	25.25	
==============================================================================
stop
> 
> fanlevel
[fanlevel] 20
> 
> version
RFPA v1.4.1 cea4585, built Sep 13 2019 21:14:28, hv rev 3
```

Shouldn't the fan never go below 20%?

hartytp, 2019-09-14:
Nope, never mind. fanlevel only matters when channels are enabled. sorry!

hartytp, 2019-09-14:
it works perfectly! Thanks!

hartytp, 2019-09-14:
@wizath one question: do fan levels below ~20% just get clipped to zero?

wizath, 2019-09-16:
> do fan levels below ~20% just get clipped to zero?

`fanlevel` states at which minimal speed fans should work. If you set them to 20%, they won't go below that, even with channels being cool. We discussed that  - with any channel enabled, fans should work constantly to reduce temperature drift

hartytp, 2019-09-16:
>fanlevel states at which minimal speed fans should work. If you set them to 20%, they won't go below that, even with channels being cool. We discussed that - with any channel enabled, fans should work constantly to reduce temperature drift

@wizath sorry, I'm not explaining myself correctly. `fanspeed 20` does exactly what I expect and works well. However, `fanspeed 10` allows the fans to go to 0 %. So, it seems that any time the fans go below 20% or so, they are set to 0. That's fine, but should be documented to avoid confusion. If you clarify how this works for low fan speeds then I'm happy to document.

wizath, 2019-09-16:
Oww, sorry. There is general threshold that below 15% they just will be clipped to 0% to avoid that annoying clicking. Yep that should be documented.

hartytp, 2019-09-16:
Thanks. That's fine, I'll document it later.

hartytp, 2019-09-17:
Wiki updated

closed 2019-09-17 by hartytp