sinara-hw/Booster #301: dead channel

hartytp, 2019-09-16:
@gkasprow one of our older TechnoSystem v1.3 boosters has a channel where the 6V current is 500mA. I've never seen that issue before (it was always the 29V current where we had issues). I think we should understand what went wrong before producing v1.5. If I send you this booster can you have a look at it and do some debugging?

gkasprow, 2019-09-16:
sure, please send it

hartytp, 2019-09-16:
Thanks, will do once I'm done fuzzing it.

hartytp, 2019-09-25:
@gkasprow did you start looking at this yet?

hartytp, 2019-09-27:
@gkasprow ping...

gkasprow, 2019-09-27:
We were with @wizath out of the office last week (in field SDR measurements). WIll have a look at it next week.

hartytp, 2019-09-27:
Aah, okay. I'd forgotten you were gone all week. No problem, sorry for being impatient :)

hartytp, 2019-10-09:
@gkasprow any luck with this?

wizath, 2019-10-09:
I've got this channel, I'll break it apart and look

hartytp, 2019-10-09:
Fine, unless the new Booster starts to exhibit odd behaviour, I think we can write this off as being due to a (now fixed) hw flaw in the older design

hartytp, 2019-10-17:
@gkasprow please can you personally look at this next week?

I have a strong suspicion that this is also a mechanical issue like https://github.com/sinara-hw/Booster/issues/321 (e.g. a via on the underside of the PCB shorting to the chassis).

It may be that we need to tweak the mounting slightly to avoid shorts like this for the next revision...

gkasprow, 2019-10-19:
I finished an intensive period of travels - since June I was away almost every week. No plans for travel until the end of the year. So I will finally have time to dig deeper into several unsolved mysteries in our designs.

jordens, 2019-10-19:
I thought we'd meet at psi next week

hartytp, 2019-10-22:
@gkasprow ping

hartytp, 2019-10-23:
@gkasprow ....

gkasprow, 2019-10-23:
Checking this is the plan for tomorrow.

gkasprow, 2019-10-24:
I assume it was an issue with channel 2
now it shows 0, the voltage is right and there is 20mV drop on the sense resistor

hartytp, 2019-10-24:
You can see the issue here: https://github.com/sinara-hw/Booster/issues/282#issuecomment-530039255

It was channel 3 (compare the hw Id) the new fw disables this channel I believe

gkasprow, 2019-10-24:
Well, that's interesting because it says it is channel 2. I replaced channel 1 with 2 and nothing changed. Then I unplugged channel 3 and nothing changed. So it seems that it reports the current from channel 3 as a channel 2...

hartytp, 2019-10-24:
> Well, that's interesting because it says it is channel 2

Odd. Do you mean that the firmware no longer prints the same thing I posted here? https://github.com/sinara-hw/Booster/issues/282#issuecomment-530039255

hartytp, 2019-10-24:
> I replaced channel 1 with 2 and nothing changed. Then I unplugged channel 3 and nothing changed. So it seems that it reports the current from channel 3 as a channel 2...

I don't follow what you mean here...

gkasprow, 2019-10-24:
It looks like the current of channel 3 is reported as a current of channel 2.

gkasprow, 2019-10-24:
This is consistent with the measurements I did.

gkasprow, 2019-10-24:
The firmware prints the status when I enter "start" command

hartytp, 2019-10-24:
> The firmware prints the status when I enter "start" command

What does it print, and what do you measure?

gkasprow, 2019-10-24:
The channel 3 is removed. And it prints
[1B][2J
PGOOD: 1
FAN SPEED: 0 %
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	0	1	1	1	1	
HWID		B2:13	B1:CF	90:A2	00:00	6D:76	56:81	3C:9C	39:6B	
INPWR [V]	0.31	0.18	0.56	0.00	0.03	0.20	0.00	0.00	
TXPWR [V]	0.01	0.04	0.01	0.41	0.05	0.01	0.02	0.01	
RFLPWR [V]	0.01	0.01	0.01	0.62	0.01	0.07	0.05	0.02	
INPWR [dBm]	nan	nan	nan	0.00	nan	nan	nan	nan	
TXPWR [dBm]	5.00	5.00	5.00	0.00	5.00	5.00	5.00	5.00	
RFLPWR [dBm]	-2.78	-2.64	-3.49	0.00	-4.44	-3.89	-4.96	-5.53	
I30V [A]	0.055	0.050	0.050	0.000	0.055	0.051	0.050	0.038	
I5V0 [A]	0.257	0.255	0.279	0.000	0.263	0.260	0.253	0.271	
5V0MP [V]	4.989	4.980	4.971	0.000	4.963	4.968	4.954	4.976	
ON		1	1	1	0	1	1	1	1	
SON		1	1	1	0	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		11	59	15	664	83	22	26	10	
ADC2		11	11	15	1018	12	122	82	30	
INTSET [dBm]	26.71	30.00	30.49	0.00	33.00	33.00	33.00	31.50	
DAC1		4095	4095	4095	0	4095	4095	4095	4095	
DAC2		2976	3444	3328	0	3566	3600	3608	3451	
SCALE1		83	87	87	0	85	87	86	87	
OFFSET1		613	693	562	0	656	635	681	594	
BIASCAL		1811	1960	1690	0	1712	1687	1657	1756	
HWIS		86.00	90.00	89.00	0.00	87.00	87.00	86.00	88.00	
HWIO		679.00	744.00	614.00	0.00	695.00	729.00	770.00	679.00	
LTEMP		31.50	31.50	32.00	0.00	31.00	32.25	32.00	32.50	
RTEMP		29.00	29.00	32.50	0.00	32.25	32.50	32.50	31.00	
==============================================================================

gkasprow, 2019-10-24:
OK, I realized what happened when I copied and placed the test here. On my terminal it looks like this
[image]

gkasprow, 2019-10-24:
Which was misleading due to shifted column...

hartytp, 2019-10-24:
seems sensible. I don't see any issues there. 

The data I had before was 
```
s␛[2J
PGOOD: 1
FAN SPEED: 30 %
AVG TEMP: 0.00 CURRENT: 0.00
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		B2:13	B1:CF	90:A2	69:E0	6D:76	56:81	3C:9C	39:6B	
TXPWR [V]	0.01	0.04	0.01	0.01	0.01	0.01	0.01	0.01	
RFLPWR [V]	0.01	0.01	0.01	0.01	0.01	0.03	0.03	0.02	
TXPWR [dBm]	-7.25	-7.30	-6.33	-6.92	-7.48	-7.06	-7.77	-6.71	
RFLPWR [dBm]	-2.78	-2.09	-3.54	-4.30	-4.33	-5.11	-5.82	-5.77	
I30V [A]	0.057	0.053	0.054	0.056	0.059	0.053	0.054	0.032	
I6V0 [A]	0.254	0.253	0.278	0.500	0.260	0.257	0.250	0.268	
5V0MP [V]	4.991	4.980	4.971	4.979	4.963	4.968	4.956	4.977	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		11	58	11	12	20	21	13	10	
ADC2		12	10	13	11	12	46	42	36	
DAC1		4095	4095	4095	4095	4095	4095	4095	4095	
DAC2		3319	3444	3328	3259	3566	3600	3608	3451	
SCALE1		83	87	87	85	85	87	86	87	
OFFSET1		613	693	562	600	656	635	681	594	
BIASCAL		1811	1960	1690	1700	1712	1687	1657	1756	
HWIS		86.00	90.00	89.00	87.00	87.00	87.00	86.00	88.00	
HWIO		679.00	744.00	614.00	649.00	695.00	729.00	770.00	679.00	
LTEMP		27.00	27.00	27.25	26.00	27.00	27.25	26.00	26.00	
RTEMP		26.00	27.00	27.25	26.00	28.50	27.00	26.00	27.00	
==============================================================================
```

hartytp, 2019-10-24:
So, that all looks fine. However, channel 3 is currently shut down, so there is no 6V current. The next step is to enable it in fw and see what the current is. Previously it was 500mA (see above)

gkasprow, 2019-10-24:
OK, Kuba is away and I cannot do much without him. Anyway, tomorrow I will play with another Booster that suffers a mechanical issue.

hartytp, 2019-10-24:
@gkasprow if you have a few minutes the easiest thing to do is just to short the enable line on the main board to enable the channel and then measure the 6V current on the sense resistor. Assuming that still reads 500mA, it's worth checking to see if there is a short to GND like on the other board.

gkasprow, 2019-10-24:
True, i will do that.

hartytp, 2019-10-27:
@gkasprow finishing debugging Booster is my absolute top priority. It would be great to get the final few issues debugged this week.

gkasprow, 2019-10-28:
well, the voltage drop on the 5V sense resistor is 26mV, so the current drawn by the RF module is fine.
Hoewer I observe 3.1V on the amplifier output which is not correct. It should be 1.3V

hartytp, 2019-10-28:
> well, the voltage drop on the 5V sense resistor is 26mV, so the current drawn by the RF module is fine.

That's interesting. So, does that mean that the ADC/current sense amp/firmware was giving the wrong reading for this channel?

> Hoewer I observe 3.1V on the amplifier output which is not correct. It should be 1.3V

Are you talking about one of the pre-amps here? 

Side note: I was beginning to wonder whether some of the bias current fluctuations/hysteresis I see could be an oscillation in the pre-amp or something. Do you think there could be an issue there?

gkasprow, 2019-10-28:
Yes, it is the INA194AIDBVT which generates too high voltage.

gkasprow, 2019-10-28:
It gives 3.27V and the 330R series resistor limits it to 3.1V

gkasprow, 2019-10-28:
I will replace it with new one.

hartytp, 2019-10-28:
> Yes, it is the INA194AIDBVT which generates too high voltage.

Aah, ok, so it's a diagnostic issue which is already fixed with the new hw revision. Good!

hartytp, 2019-10-28:
So, to check, the pre-amps and RF assembly are all fine, this issue is purely those InAmps? If so, that's great and we can close.

gkasprow, 2019-10-28:
I replaced it with a new one and it works.
I'm curious why it failed.

gkasprow, 2019-10-28:
It was partially working. Without a load it gave some 0.2V output. So it is not a power stage that was shorted.

hartytp, 2019-10-28:
I'm not totally sure if it did fail...if you look at the spec sheet, we're operating them way out of the regime where they are properly specified. So, it's possible this is a component tolerance issue.

Since the current didn't used to trigger a software error, it's quite possible that this was an issue from the beginning.

hartytp, 2019-10-28:
> It was partially working. Without a load it gave some 0.2V output. So it is not a power stage that was shorted.

? What's working? I thought there was no RF issue on this channel?

hartytp, 2019-10-28:
AFAICT, there is no issue here apart from the choice of InAmp, which we've already fixed, so we can close this.

Thanks!

Can you have a look at https://github.com/sinara-hw/Booster/issues/321 next?

closed 2019-10-28 by hartytp

gkasprow, 2019-10-28:
[image]
It could be caused by broken PNP transistor. But I have no idea how this could happen. The amplifier accepts so high voltages that it is hard to break it.

gkasprow, 2019-10-28:
> 
>     It was partially working. Without a load it gave some 0.2V output. So it is not a power stage that was shorted.
> 
> ? What's working? I thought there was no RF issue on this channel?

the amplifier was working:)