sinara-hw/Booster #303: I2C errors

hartytp, 2019-09-16:
Using `Testing on: Version(fw_rev='v1.4.1', fw_hash='0d48c19', fw_build_date=datetime.datetime(2019, 9, 16, 11, 34, 56), device_id='34364708', hw_rev='hw rev 1.3') `

I just got an I2C error. Interestingly, the red LED is on channel 6, but I see I2C errors on ch5 in the logs...

Opening up the VCP interface I'm greeted by
```
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
dac status 1 4 4
```

other diagnostics:
```
> logstash
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
[ERROR] Unhandled network socket event 20
> 
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	0	207903	0	0	
> 
> i2cdetect
Wrong number of parameters
> 
> 
(5/5) i2cerr 6
Wrong number of parameters
> 
> 
(5/5) i2cdetect 6
[i2c_scan] start
[i2c_scan] end
> 
> 
(6/6) i2cdetect 5
[i2c_scan] start
[i2c_scan] end
> 
> status 5
[status] e=1 s=1 r1=20 r2=50 tx=5.000 rf=-4.756 curr=0.180 t=27.25 i=1.03 ip=-nan
> 
> status 6
[status] e=0 s=0 r1=12 r2=41 tx=0.000 rf=0.000 curr=0.000 t=63.00 i=1.24 ip=-nan
> 
> start
␛[2J
PGOO> 
> D: 1
FAN SPEED: 98 %
AVG TEMP: 62.85 CURRENT: 63.00
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		B2:13	B1:CF	90:A2	69:E0	6D:76	56:81	3C:9C	39:6B	
INPWR [V]	0.00	0.21	0.41	0.62	0.83	1.03	1.24	1.44	
TXPWR [V]	0.01	0.04	0.01	0.01	0.01	0.01	0.01	0.01	
RFLPWR [V]	0.01	0.01	0.01	0.01	0.01	0.03	0.03	0.02	
INPWR [dBm]	-nan	-nan	-nan	-nan	-nan	-nan	-nan	-nan	
TXPWR [dBm]	5.00	0.00	0.00	5.00	5.00	5.00	0.00	5.00	
RFLPWR [dBm]	-2.79	0.00	0.00	-4.32	-4.44	-4.77	0.00	-5.51	
I30V [A]	0.180	0.000	0.000	0.180	0.180	0.180	0.000	0.180	
I6V0 [A]	0.164	0.000	0.000	0.164	0.164	0.164	0.000	0.164	
5V0MP [V]	2.051	0.000	0.000	2.051	2.051	2.051	0.000	2.051	
ON		1	0	0	1	1	1	0	1	
SON		1	0	0	1	1	1	0	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		11	57	11	11	22	20	12	9	
ADC2		10	9	11	11	11	50	41	35	
INTSET [dBm]	30.10	30.00	30.49	30.00	33.00	33.00	33.00	31.50	
DAC1		4095	4095	4095	4095	4095	4095	4095	4095	
DAC2		3268	3444	3328	3259	3566	3600	3608	3451	
SCALE1		83	87	87	85	85	87	86	87	
OFFSET1		613	693	562	600	656	635	681	594	
BIASCAL		1811	1960	1690	1700	1712	1687	1657	1756	
HWIS		86.00	90.00	89.00	87.00	87.00	87.00	86.00	88.00	
HWIO		679.00	744.00	614.00	649.00	695.00	729.00	770.00	679.00	
LTEMP		27.50	63.00	63.00	27.50	27.50	27.00	63.00	28.50	
RTEMP		25.00	63.00	63.00	26.00	26.00	27.25	63.00	26.00	
==============================================================================
stop
> 
> 
```

wizath, 2019-09-16:
* Sorry, forgot to push that weekend's I2C speed changes to 100 kHz from my home PC, it was at 400k.  
* I2C error occured while initializing I2C transaction  
* `[ERROR] Unhandled network socket event 20` is just socket returning listening status due to out of interrupt polling, nothing to worry, added handling it now

[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3616370/eth-scpi.dfu.zip)

Since I'm unable (for now) to reproduce ETH server locking up - can you post `ethdbg` command if this happens next time?

hartytp, 2019-09-16:
@wizath look at these temperature measurements. Indicative of more issues?
```
> logstash
[INFO] hardware revision 3
[WARN] watchdog reset occurred!
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1981 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] MACADDR 36:34:39:35:38:32
[INFO] RF Channel at 0 detected, id = B2:13
[INFO] RF Channel at 1 detected, id = B1:CF
[INFO] RF Channel at 2 detected, id = 90:A2
[INFO] RF Channel at 3 detected, id = 69:E0
[INFO] RF Channel at 4 detected, id = 6D:76
[INFO] RF Channel at 5 detected, id = 56:81
[INFO] RF Channel at 6 detected, id = 3C:9C
[INFO] RF Channel at 7 detected, id = 39:6B
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.233
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
[INFO] [tempmgt] Temp 110.000000 (d 100) -> Fan 100
[INFO] [tempmgt] Temp -55.000000 (d 80) -> Fan 20
[INFO] [tempmgt] Temp 55.000004 (d 55) -> Fan 75
[INFO] [tempmgt] Temp 27.500000 (d 55) -> Fan 20
> 
```

hartytp, 2019-09-16:
```
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	0	0	0	0	
```

but how else do we get such silly temperatures?

hartytp, 2019-09-16:
also, why don't those temperatures result in an error condition?

wizath, 2019-09-16:
Interesting, since the error conditions are as follows
`if (channels[i].enabled && 
(channels[i].measure.remote_temp > 60.0f || channels[i].measure.remote_temp < 5.0f))`

hartytp, 2019-09-16:
maybe the interlock task crashed? Or maybe there was a race where a second (valid) reading came in before the interlock task ran?

Either way, we definitely still seem to have I2C errors

wizath, 2019-09-16:
BTW. You used that FW with 400k I2C speed? https://github.com/sinara-hw/Booster/issues/303#issuecomment-531770410

hartytp, 2019-09-16:
I posted the version info on the top post of this thread

hartytp, 2019-09-17:
Seems to be fixed with the new fw, thanks!

closed 2019-09-17 by hartytp