sinara-hw/Booster #306: I2C errors

hartytp, 2019-09-17:
Haven't upgraded fw to latest yet (will do now). Using `Testing on: Version(fw_rev='v1.4.1', fw_hash='218aca0', fw_build_date=datetime.datetime(2019, 9, 16, 16, 23, 32), device_id='34364708', hw_rev='hw rev 1.3')`. Does this version have the correct clock freq?

Seeing 
```
logstash
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_TIMEOUT error
[ERROR] I2C_TRANSMIT_TIMEOUT error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_TIMEOUT error
[ERROR] I2C_TRANSMIT_TIMEOUT error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
[ERROR] I2C_TRANSMIT_BUSY error
> 
> 
```

Are there any other tests you want me to do before power cycling and updating fw?

hartytp, 2019-09-17:
```
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	287498	0	0	0	0	
```

wizath, 2019-09-17:
Nope, is that 3rd channel dead or something? Red LED occurred?

hartytp, 2019-09-17:
hmm...no red light, but this channel was previously drawing 500mA 6V0. So, yes, maybe it is just dead :(

Okay, I'll make my tests skip this one channel for now.

hartytp, 2019-09-17:
any idea when the next batch of TS Boosters are due to arrive, as it's not totally straight forward for me to exclude one channel form my tests?

hartytp, 2019-09-17:
Okay, can't reproduce after excluding that channel from tests...

hartytp, 2019-09-17:
Just got another I2C error but no corresponding red LED on the channel. This means that:
- I2C errors still occur
- they don't always trigger ERROR LEDs yet

```
logstash
[INFO] network client 10.255.6.130 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
[ERROR] I2C_TRANSMIT_BUSY error
[INFO] network client disconnected
[INFO] network client 10.255.6.130 connected
> 
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	1	0	0	0	
> 
> version
RFPA v1.4.1 ed443ff, built Sep 17 2019 13:15:33, hv rev 3
> 
> start
␛[2J
PGOOD: 1
FAN SPEED: 20 %
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		B2:13	B1:CF	90:A2	69:E0	6D:76	56:81	3C:9C	39:6B	
INPWR [V]	0.75	0.01	0.55	0.00	0.03	0.17	0.00	0.00	
> 
> TXPWR [V]	0.01	0.04	0.01	0.01	0.01	0.01	0.01	0.01	
RFLPWR [V]	0.01	0.01	0.01	0.01	0.01	0.03	0.03	0.02	
INPWR [dBm]	-nan	-nan	-nan	-nan	-nan	-nan	-nan	-nan	
TXPWR [dBm]	25.42	5.00	5.00	0.00	5.00	5.00	5.00	5.00	
RFLPWR [dBm]	-0.52	-2.63	-3.52	0.00	-4.43	-4.84	-5.39	-5.48	
I30V [A]	0.111	0.052	0.053	0.000	0.059	0.053	0.054	0.029	
I6V0 [A]	0.254	0.252	0.278	0.000	0.260	0.257	0.250	0.268	
5V0MP [V]	4.989	4.980	4.971	0.000	4.965	4.968	4.956	4.977	
ON		1	1	1	0	0	1	1	1	
SON		1	1	1	0	0	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		11	57	11	11	11	21	13	9	
ADC2		10	10	11	11	11	46	41	34	
INTSET [dBm]	37.00	30.00	30.49	30.00	33.00	33.00	33.00	31.50	
DAC1		4095	4095	4095	4095	4095	4095	4095	4095	
DAC2		3861	3444	3328	3259	3566	3600	3608	3451	
SCALE1		83	87	87	85	85	87	86	87	
OFFSET1		613	693	562	600	656	635	681	594	
BIASCAL		1811	1960	1690	1700	1712	1687	1657	1756	
HWIS		86.00	90.00	89.00	87.00	87.00	87.00	86.00	88.00	
HWIO		679.00	744.00	614.00	649.00	695.00	729.00	770.00	679.00	
LTEMP		26.25	27.25	26.00	26.25	26.00	26.00	27.50	27.50	
RTEMP		26.50	25.00	27.00	24.00	27.25	27.00	26.00	26.00	
==============================================================================
stop
```

wizath, 2019-09-17:
Issue could be that `rf_channel_disable_on_i2c_err` checks if channels is enabled. If error happened in enable procedure, it could not work (since `enabled = 0`). I'll change to disable all detected channels on I2C error, not only enabled.

hartytp, 2019-09-17:
aah, good catch.

However, it does seem that we still get I2C errors, so there is still an issue there :(

hartytp, 2019-09-18:
@gkasprow @wizath how is the design review going?

hartytp, 2019-09-18:
ping

also @wizath do you know when the next TS v1.4 Boosters will be ready? I'd like to check this all on them...

gkasprow, 2019-09-18:
There could be a potential issue with multiple pullup resistors - we have 2k2 before and after the switch.
And then we have 5k pullup on every RF moodule. But on the scope it looks like we have correct logic low level.
The switch resistance is max 30Ohm

hartytp, 2019-09-18:
Tested again with the old firmware. FWIW, the I2C error happened again during the `CHAN:ENAB` command. The last few errors have all occurred during that command, so that's the thing to check.

Swapping to new FW now.

hartytp, 2019-09-18:
@gkasprow @wizath unless you can think of any more tests I should run, I suggest I just ship this Booster back to you so you can test more directly. Now I have a fuzzer that reliably throws I2C errors (but no red LEDs) after 15min or so, hopefully you can track down.

hartytp, 2019-09-18:
Let me know either way, so I can post tomorrow am if that's the plan

gkasprow, 2019-09-18:
Since we cannot recreate such errors in our box, this seems to be the only way out.
Send it please.

hartytp, 2019-09-19:
okay, @gkasprow that Booster is in the post to you. Should be with you tomorrow/monday.

I can post the latest fuzzing code I'm using if that helps. Since the errors seem to occur during enables/disables it might be worth you triggering your scope off that, and getting it to stop when an I2C timeout occurs.

@wizath can you give TS a hand getting our next two Booster v1.4s back asap please? Let me know when you have a shipping estimate for that.

Once that's done, can you have a look at the final outstanding small issues: 
- [ ] https://github.com/sinara-hw/Booster/issues/309
- [ ] https://github.com/sinara-hw/Booster/issues/307
- [ ] https://github.com/sinara-hw/Booster/issues/291

hartytp, 2019-10-01:
@gkasprow mentioned that you'd been able to reproduce this now. How is the debugging going? This and #270 are now the only remaining fw issues.

wizath, 2019-10-01:
I'll need to confirm that, but I'm pretty sure that temperature sensor is quite easy to kill

hartytp, 2019-10-01:
Okay, good. That should be an easy fix then?

wizath, 2019-10-01:
I hope so

hartytp, 2019-10-04:
@wizath any progress on this? I'm keen to finish testing as soon as my new Booster arrive :)

wizath, 2019-10-04:
Expect some results on monday

hartytp, 2019-10-04:
Perfect, thanks!

wizath, 2019-10-07:
I'm struggling to recreate any I2C error with your device... I've been running fuzzing script that turns channels on/off for like 4~ hours now and everything works perfectly

hartytp, 2019-10-07:
Which firmware revision? What fuzzing script? Which hw?

wizath, 2019-10-07:
* v1.3 HW
* recent 100 kHz I2C FW
* Fuzzing script from https://github.com/sinara-hw/Booster/issues/289#issuecomment-531743209 modified to only perform ON/OFF tests

hartytp, 2019-10-07:
Okay, interesting. Is this the same (malfunctioning) v1.3 I sent back? If so, then either you're using different fw or it's the changes to the fuzzing test that made this issue go away

wizath, 2019-10-07:
Also I've taken off the lid and removed screws from channels

hartytp, 2019-10-07:
Interesting...what's the plan then? Put it back together, enable the other tests and see if you can reproduce it?

When the new v1.4 hw arrives, I'll fuzz again with the newest firmware and see if I can still reproduce this issue.

wizath, 2019-10-07:
If my testing goes well, I think the issue will be solved

hartytp, 2019-10-07:
Cool. I should be able to start testing on new HW today.

hartytp, 2019-10-08:
So far can't reproduce on new hw

hartytp, 2019-10-09:
Have you reproduced this?

wizath, 2019-10-09:
Yep, it's ADS7924 on mainboard which causes problems.

hartytp, 2019-10-09:
Aah good so we know the cause. Any idea why that chip causes problems?

wizath, 2019-10-09:
This occurs when we're performing sequential readouts from the chip. I've tried converting it to auto-address-increment readouts etc. but IMHO adding delays between each readout gives best results. It was running yesterday for like 6h with fuzzing script and no error yet.

hartytp, 2019-10-09:
Cool! That's very good. In that case, shall I'll optimistically close this and reopen if I see the problem again?

Right now, I'm not seeing this, but am seeing watchdog resets and scpi hangs

hartytp, 2019-10-09:
Haven't seen any I2C errors on the recent firmware (assuming they're not related to the watchdog issues)

closed 2019-10-09 by hartytp