sinara-hw/Booster #308: Bias current fluctuations

hartytp, 2019-09-17:
@wizath I'm seeing something I don't understand. When my fuzz script starts, it queries the bias current for each channel using the `chan:diag` command. Then, during the course of my fuzzing I check regularly that the current is still what I expect.

However, on rare occasions, the bias current is ~10mA lower than expected when I first query it. At first I though this was something to do with averaging, but delays don't seem to help. I wonder if it's related to https://github.com/sinara-hw/Booster/issues/270

Anyway, I now can't reproduce it :( so not sure what to make of this...

hartytp, 2019-09-17:
Once I can use the VCP interface again, I'll cross-check the diagnostics I'm doing over SCPI with the VCP results.

hartytp, 2019-09-17:
Anyway, other than that, I haven't been able to find any (!!) issues with my fuzzing.

I'd like to retry this with v1.4 hw (currently using v1.3) since we improved quite a few things around the current measurements, so this could be some glitch that we already fixed...

hartytp, 2019-09-17:
I've seen channel 7 on this device vary between about 30mA and 40mA. I wonder if this is another dead FET? Maybe due to the I2C issue messing up the bias (NB the max DC current consumption is lower than the max current consumption with RF applied as no power is dissipated in the load).

hartytp, 2019-09-17:
hmm...logging on the VCP while leaving Booster alone (no RF applied, not enabling/disabling channels, no SCPI transactions) the currents look very stable. Almost all points lie within +-1mA of their means.

[image]

hartytp, 2019-09-17:
Enabling and disabling channels I do see some very large spikes that take a while to settle. So, maybe this is just me being impatient and not waiting long enough between enabling channels and taking data...

[image]

hartytp, 2019-09-17:
That and the fact that I'm taking vastly more data on the SCPI, so infrequent larger deviations from the mean (few mA) do occur very rarely

closed 2019-09-17 by hartytp

hartytp, 2019-09-17:
@wizath how long do you expect the current readings to take to update after the current changes (including any averaging you're doing)? I found that a 2s delay in my code between enabling the channel (or turning input RF off) and measuring the current wasn't always sufficient.

hartytp, 2019-09-17:
Here is something interesting:
- while Booster is running the current seems stable (reported by SCPI/VCP interfaces)
- the current does not seem to change when I enable/disable channels (e.g. via FP button)
- power-cycling doesn't seem to affect it
- yet, I've seen two very different values for the currents on channel 7...

```
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		B2:13	B1:CF	90:A2	69:E0	6D:76	56:81	3C:9C	39:6B	
INPWR [V]	0.45	0.01	0.01	0.00	0.03	0.17	0.00	0.00	
TXPWR [V]	0.01	0.04	0.01	0.01	0.01	0.01	0.01	0.01	
RFLPWR [V]	0.01	0.01	0.01	0.01	0.01	0.03	0.03	0.02	
INPWR [dBm]	-nan	-nan	-nan	-nan	-nan	-nan	-nan	-nan	
TXPWR [dBm]	5.00	5.00	5.00	0.00	5.00	5.00	5.00	5.00	
RFLPWR [dBm]	-2.78	-2.63	-3.52	0.00	-4.44	-4.84	-5.48	-5.47	
I30V [A]	0.056	0.052	0.053	0.000	0.059	0.053	0.054	0.055	
I6V0 [A]	0.254	0.253	0.278	0.000	0.260	0.257	0.250	0.268	
5V0MP [V]	4.989	4.980	4.971	0.000	4.963	4.966	4.954	4.979	
ON		1	1	1	0	1	1	1	1	
SON		1	1	1	0	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		11	58	12	12	15	21	13	10	
ADC2		11	11	12	12	11	43	42	34	
INTSET [dBm]	37.00	30.00	30.49	30.00	33.00	33.00	33.00	31.50	
DAC1		4095	4095	4095	4095	4095	4095	4095	4095	
DAC2		3861	3444	3328	3259	3566	3600	3608	3451	
SCALE1		83	87	87	85	85	87	86	87	
OFFSET1		613	693	562	600	656	635	681	594	
BIASCAL		1811	1960	1690	1700	1712	1687	1657	1756	
HWIS		86.00	90.00	89.00	87.00	87.00	87.00	86.00	88.00	
HWIO		679.00	744.00	614.00	649.00	695.00	729.00	770.00	679.00	
LTEMP		27.50	27.50	27.00	25.25	27.50	26.00	27.25	26.00	
RTEMP		25.00	26.00	27.50	25.50	25.00	27.00	26.00	27.00	
==============================================================================
```

when if you look in the log I posted above it was on 30mA...

hartytp, 2019-09-17:
Now that it's gone to the higher current, I cannot reproduce the lower current. This is really odd. I do wonder if it's a dead FET after all (maybe related to the I2C errors).

hartytp, 2019-09-17:
I'll leave it enabled and logging overnight and see if I can catch any meta-stability.

hartytp, 2019-09-17:
cf https://github.com/sinara-hw/Booster/issues/306#issuecomment-532389163

wizath, 2019-09-17:
Channel measurements run in 50 ms intervals

hartytp, 2019-09-17:
> Channel measurements run in 50 ms intervals

hmm...so it's odd that I need to leave such a long delay then.

hartytp, 2019-09-17:
Okay, I've run out of patience testing on v1.3 hw, will resume once I have more v1.4 hw to test, since this could be related to a hw bug we already fixed...

hartytp, 2019-09-17:
How does this function work? https://github.com/sinara-hw/booster-firmware/blob/0d48c19db08ea1ca4e9fac1782fc18dea86a688d/Source/board/channels.c#L245

What does this do? https://github.com/sinara-hw/booster-firmware/blob/0d48c19db08ea1ca4e9fac1782fc18dea86a688d/Source/board/channels.c#L264

What do these lines do? https://github.com/sinara-hw/booster-firmware/blob/0d48c19db08ea1ca4e9fac1782fc18dea86a688d/Source/board/channels.c#L288-L293

Am I misunderstanding things, or does this not implement the recommended biasing sequence... 
[image]

(it looks like you set VGS first, then enable the 29V voltage, unless I'm misunderstanding the code).

Also, what happens if the lock_take call fails e.g. due to a bus conflict? Doesn't that leave the DAC improperly initialized without giving an error?

hartytp, 2019-09-17:
I also don't understand why you need this line https://github.com/sinara-hw/booster-firmware/blob/0d48c19db08ea1ca4e9fac1782fc18dea86a688d/Source/board/channels.c#L296 why doesn't `rf_channels_sigon` update `channels[channel].enabled = true;`?

wizath, 2019-09-17:
`I also don't understand why you need this line https://github.com/sinara-hw/booster-firmware/blob/0d48c19db08ea1ca4e9fac1782fc18dea86a688d/Source/board/channels.c#L296 why doesn't rf_channels_sigon update channels[channel].enabled = true;?`

https://github.com/sinara-hw/Booster/issues/283

hartytp, 2019-09-17:
That's not my point...my point is that https://github.com/sinara-hw/booster-firmware/blob/0d48c19db08ea1ca4e9fac1782fc18dea86a688d/Source/board/channels.c#L206 should update `channels.enabled`. Separating the code that updates the GPIO pins from the code that updates the status flags is a recipe for synchronization issues/bugs; they should be in the same function call.

wizath, 2019-09-17:
https://github.com/sinara-hw/booster-firmware/blob/0d48c19db08ea1ca4e9fac1782fc18dea86a688d/Source/board/channels.c#L264

That command switches the GPIO which controls hardware sequencing circuit

https://github.com/sinara-hw/booster-firmware/blob/0d48c19db08ea1ca4e9fac1782fc18dea86a688d/Source/board/channels.c#L288-L293

This is an old workaround for interlock tripping on detector spikes during enable procedure

`Am I misunderstanding things, or does this not implement the recommended biasing sequence...`
Sequencing is done via hardware IC

hartytp, 2019-09-17:
I see `#define portMAX_DELAY ( TickType_t ) 0xffffffffUL`, so we essentially block forever.

hartytp, 2019-09-17:
> This is an old workaround for interlock tripping on detector spikes during enable procedure

Can we delete then?

hartytp, 2019-09-17:
> Sequencing is done via hardware IC

Aah, yes, thank you. I'd forgotten. hmm..some comments in the code would help here.

hartytp, 2019-09-17:
Right, each channel has `SIG_ON` which is the RF switch and `PWR_ON` which does the sequencing IC (which pulls the bias line to the max neg voltage when the power is disabled)...

hartytp, 2019-09-17:
aah...one thing: the 30mA current is probably actually 0mA current as this is a v1.3 Booster where the InAmp doesn't work at low currents! So, I suspect that the real issue here is that channel 7 was not getting enabled sometimes.

hartytp, 2019-09-17:
By power cycling the Booster a few times, I've been able to make channel 7 start up with a different bias current:

```
**PGOOD: 1
FAN SPEED: 20 %
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		B2:13	B1:CF	90:A2	69:E0	6D:76	56:81	3C:9C	39:6B	
INPWR [V]	0.46	0.01	0.55	0.00	0.03	0.17	0.00	0.00	
TXPWR [V]	0.01	0.04	0.01	0.01	0.01	0.01	0.01	0.01	
RFLPWR [V]	0.01	0.01	0.01	0.01	0.01	0.02	0.03	0.02	
INPWR [dBm]	-nan	-nan	-nan	-nan	-nan	-nan	-nan	-nan	
TXPWR [dBm]	0.00	5.00	5.00	0.00	5.00	5.00	5.00	5.00	
RFLPWR [dBm]	0.00	-2.63	-3.54	0.00	-4.44	-4.87	-5.49	-5.43	
I30V [A]	0.000	0.052	0.053	0.000	0.059	0.053	0.054	0.041	
I6V0 [A]	0.000	0.252	0.278	0.000	0.260	0.257	0.250	0.268	
5V0MP [V]	0.000	4.980	4.969	0.000	4.963	4.968	4.956	4.976	
ON		0	1	1	0	1	1	1	1	
SON		0	1	1	0	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		11	57	11	11	18	21	13	9	
ADC2		10	10	11	11	11	39	41	35	
INTSET [dBm]	37.00	30.00	30.49	30.00	33.00	33.00	33.00	31.50	
DAC1		4095	4095	4095	4095	4095	4095	4095	4095	
DAC2		3861	3444	3328	3259	3566	3600	3608	3451	
SCALE1		83	87	87	85	85	87	86	87	
OFFSET1		613	693	562	600	656	635	681	594	
BIASCAL		1811	1960	1690	1700	1712	1687	1657	1756	
HWIS		86.00	90.00	89.00	87.00	87.00	87.00	86.00	88.00	
HWIO		679.00	744.00	614.00	649.00	695.00	729.00	770.00	679.00	
LTEMP		26.25	27.50	26.00	25.25	27.25	27.00	26.00	27.00	
RTEMP		26.50	26.00	27.00	25.50	27.50	27.00	27.00	27.25	
==============================================================================
```

This seems v stable once it's started up (as with the trace above).

hartytp, 2019-09-17:
- enabling and disabling does not seem to do anything to the channel 7 current. It stays at 43mA.
- Now power cycle booster and we're back to 53mA.

Measuring on the sup ctrl board with the current at 42mA:
- out bias: -1.461
- P5V0: 4.097
- MP: 4.975
- 29V: 27.61

hartytp, 2019-09-17:
Restarting to get 54mA back. Don't see any difference in any rails. So, that's odd...

hartytp, 2019-09-17:
So either there is something V strange going on with the FET or, probably more likely, this really is just another symptom of those damn InAmps. Let me solder some probes onto the sense resistor to check...

hartytp, 2019-09-17:
- when Booster reads 44mA on ch7, the sense voltage is 3.94mV so 44mA. 
- cycle and try again. 46mA read. 4.15mV => 46mA

hm...can't get it to boot up to 53mA now...

hartytp, 2019-09-17:
okay, can't reproduce, probably a v1.3 hw issue...give up.

Let's fix the remaining I2C bugs and then I'll finish testing on v1.4