sinara-hw/Booster #311: hysteresis

hartytp, 2019-09-18:
A curious observation:
- measuring the 29V current with a high-resolution DMM, I find excellent reproducibility when I enable/disable the channels with the FP button  (after enabling/disabling the current returns to exactly where it started)
- if I power cycle the chassis, I find the current changes by a few percent each time
- I don't see any explanation for this in terms of the bias voltage or Vds.

Any idea what could cause this @gkasprow ?

hartytp, 2019-09-18:
In fact, it's more than that, I've seen the bias current vary by almost 20% between power cycles of the chassis, but the bias voltage seems rock solid. What kind of effect in the FET could cause this??

hartytp, 2019-09-18:
I feel like this must be a dead FET symptom, but I don't understand the physics

hartytp, 2019-09-18:
@gkasprow I do not understand this at all.
- I verified that taking the channel in/out of standby mode gives very reproducible currents
- I verified that the current diagnostics do read correctly
- I verified that when the channel is enabled, the 29V current drops to close to zero
- I verified that the 29V and OUT_BIAS voltages are very reproducible across power cycles and disabling/enabling channels
- Yet, despite all that, I still sometimes see fluctuations of 20% in the bias current when I power cycle the chassis!

So, it seems that there is some hysteretic behaviour that happens when the bias is turned to 0V when the power is cut (or maybe it happens when the 29V drops to exactly zero, rather than just a low voltage).

How can that be? Can it be FET damage (e.g. due to an I2C error)? Looking at the FET datasheet, it should be pretty robust, so it should be hard to damage and, in any case, what kind of damage mechanism could lead to this behaviour?

hartytp, 2019-09-18:
I'll make a note to fuzz some new v1.4 hw aggressively for a prolonged period of time and look for any such behaviour.

hartytp, 2019-09-18:
But, @gkasprow if you know of any causes of FET hysteresis then do tell me...

gkasprow, 2019-09-18:
The only hysteresis that comes to my mind is related to temperature. But it is not the case here.
These FETs used to survive serious overcurrent and continuous work with much higher bias than allowed.

hartytp, 2019-09-18:
Indeed, so how do you explain these symptoms? I know the inductors died on some v1.3 boards, but again I don't see how that would be relevant here. All in all this is a mystery to me.

hartytp, 2019-09-18:
hmm...having said that, this channel is now at 74 mA
```
PGOOD: 1
FAN SPEED: 20 %
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		B2:13	B1:CF	90:A2	69:E0	6D:76	56:81	3C:9C	39:6B	
INPWR [V]	0.83	0.16	0.55	0.00	0.03	0.17	0.00	0.00	
TXPWR [V]	1.82	0.04	0.01	0.01	0.01	0.01	0.01	0.01	
RFLPWR [V]	0.28	0.01	0.01	0.01	0.01	0.03	0.03	0.02	
INPWR [dBm]	-nan	-nan	-nan	-nan	-nan	-nan	-nan	-nan	
TXPWR [dBm]	28.53	5.00	5.00	0.00	5.00	5.00	5.00	5.00	
RFLPWR [dBm]	2.56	-2.64	-3.52	0.00	-4.43	-4.85	-5.40	-5.47	
I30V [A]	0.153	0.053	0.054	0.000	0.059	0.054	0.054	0.074	
I6V0 [A]	0.254	0.252	0.278	0.000	0.260	0.257	0.250	0.270	
5V0MP [V]	4.988	4.980	4.969	0.000	4.963	4.966	4.954	4.977	
ON		1	1	1	0	1	1	1	1	
SON		1	1	1	0	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		2981	58	12	12	15	21	13	11	
ADC2		464	11	11	11	12	42	48	34	
INTSET [dBm]	37.00	30.00	30.49	30.00	33.00	33.00	33.00	31.50	
DAC1		4095	4095	4095	4095	4095	4095	4095	4095	
DAC2		3904	3444	3328	3259	3566	3600	3608	3451	
SCALE1		83	87	87	85	85	87	86	87	
OFFSET1		613	693	562	600	656	635	681	594	
BIASCAL		1811	1960	1690	1700	1712	1687	1657	1756	
HWIS		86.00	90.00	89.00	87.00	87.00	87.00	86.00	88.00	
HWIO		679.00	744.00	614.00	649.00	695.00	729.00	770.00	679.00	
LTEMP		25.50	24.50	24.00	24.25	23.00	24.00	24.25	24.25	
RTEMP		25.50	23.00	24.25	24.50	24.25	24.25	22.00	23.00	
==============================================================================
```

but when I first powered the unit on, it was 44mA...

hartytp, 2019-09-18:
nope, scratch that, mistake in my setup...

hartytp, 2019-09-18:
since we have no convincing theory for how this could have been damaged, I think the best thing is if I send you the entire Booster chassis back (1 ch with dead 5V rail and 1 ch with hysteresis on the FET) and you can investigate....

When I get the new v1.4 boards I'll see if any such issues appear after a long run of stress testing...

gkasprow, 2019-09-18:
OK. Next week I will be out of my office, doing RF tests in the middle of nowhere, but once I come back will have look at that. It's really intriguing! 
RF is real wizardry, today I discovered that 2.2nF 50V 0603 capacitor at the output of 20W power stage didn't let me dump more than 5W of power to the load. Finally, I shorted it and it works beautifully. It didn't get damaged, it still shows 2.2nF.
Btw, if you are looking for very cheap 20W (<10EUR) amplifier that works in 400..500MHz region, look at [SHWE1112](https://www.tvsat.com.pl/pdf/M/mhw720a_mot.pdf) or MHW720A2. These are CATV power stages optimized for various ranges.
I designed a PCB with 3D printed frame + lid that one attaches to the heatsink and uses wit 12V, 3A laptop wall wart.
[image]
[image]
[image]

hartytp, 2019-09-18:
Nice! Okay, I'll send it back. It's fine if you look at next week. Let's just get the last couple of bugs out and get this working well.

@wizath in the meantime, can you look at the other small issues (inc VCP docs) and help TS get the final Booster v1.4s shipped to us so I have more HW to test on please?

hartytp, 2019-10-07:
@gkasprow are you able to look at this today? I really need to get this fixed and move onto the next hw revision

hartytp, 2019-10-08:
@gkasprow how are you getting on with this??

gkasprow, 2019-10-08:
I had no chance to touch it today. Too much urgent paperwork. Hopefully will be able to work on it tomorrow.

hartytp, 2019-10-08:
understood! On the plus side, the new booster I have seems to work beautifully so far (apart from the small ethernet bug)

hartytp, 2019-10-28:
@gkasprow can you have a look at this now please? I observed this on the booster we sent you that had the "high 6V current" on one channel.

The observation was that if I power cycled the Booster, then enabled all channels and looked at the bias current in software, the currents could vary significantly (20% sometimes) between power cycles. I would have expected the currents to be very stable (up to thermal affects, which wouldn't be significant over a quick power cycle). So, this is quite concerning. See the comments above for further details...

gkasprow, 2019-10-28:
Does it happen on a particular channel?

gkasprow, 2019-10-28:
now I have
[image]

after power cycle
[image]

gkasprow, 2019-10-28:
after another power cycle
[image]

gkasprow, 2019-10-28:
I will try in 10 mins or so.

gkasprow, 2019-10-28:
[image]

hartytp, 2019-10-28:
I tried power cycling it 20 times or so back to back and only saw the issue once or twice...

hartytp, 2019-10-28:
IIRC it was mainly channel 6 or 7 (can't remember which).

gkasprow, 2019-10-28:
OK, I will continue testing tomorrow.

hartytp, 2019-10-28:
Thanks!

While you're looking, the other issue on my list is https://github.com/sinara-hw/Booster/issues/325

Basically, looking at the SCPI interface, I sometimes see periods where the bias current on one channel fluctuates by a few mA even without RF applied. Then it goes back to being stable to better than 1mA. So it seems that there is some instability. However...I was having trouble tracking this down because my loggers kept crashing...

dnadlinger, 2019-10-28:
This could be related to #329, at least in appearance, although I sadly don't have bias current logs due to the crashing issue.

hartytp, 2019-10-31:
@gkasprow any progress with this?

gkasprow, 2019-10-31:
I tested it for a day a few times but did not observe anything suspicious.

hartytp, 2019-10-31:
@gkasprow hm....can you make sure you've power-cycled this Booster a decent number of times (say, 20) and looked for variations in the bias current? It's best to use the SCPI interface for this (I can send you a python driver or you can just use miniterm) as that prints out with much better resolution than the VCP.

If you can't reproduce any significant hysteresis, can you reassemble and ship this back to me please?

gkasprow, 2019-10-31:
OK, I will try with python script. Send it please.

hartytp, 2019-11-01:
@gkasprow the driver is https://paste.debian.net/1112566/

You can use this with a script like:
```python
from driver import Booster

def main():
    dev = Booster("xx.xx.xx.xx")
    currents = ["{:.2f}".format(1e3*dev.get_current(chan)) for chan in range(8)]
    print(dev.get_version)
    print(" ".join(currents))
if __name__ == "__main__":
    main()
```

(That's pseudocode so you might need to fix typos, but it should be close)

hartytp, 2019-11-01:
The test I'd like you to do is:
- turn Booster power on
- enable all channels
- wait 1s or so
- run python script to print out all channel currents
- turn Booster power off
- repeat 20 times

If all channel currents are stable to 1mA or so then there is no issue. However, previously I saw changes of several mA on some Boots.

hartytp, 2019-12-18:
Assuming this is lack of TVS diodes. Closing for now.

closed 2019-12-18 by hartytp