sinara-hw/Booster #313: Input detector ADC not working on some channels

wizath, 2019-09-30:
As I implemented automated input power calibration, I've noticed that some of the input ADC's return constant value, no matter on what level input signal is.

hartytp, 2019-09-30:
...that's odd. Can you stick a DMM on the supplemental control board and see if there is a signal there?

wizath, 2019-09-30:
DMM shows exactly the same output as ADC like < 0.01 V

hartytp, 2019-09-30:
hmm...oops...did we remove them from the design in this revision?

[image]

That might explain why they don't work any more, so this might have to become one for the v1.5 milestone.

hartytp, 2019-09-30:
> DMM shows exactly the same output as ADC like < 0.01 V

Which pin on the sup control board are you probing, since I don't see anything on the schematics with the input power connected any more.

hartytp, 2019-09-30:
Also, does it work on any channels? I can't see how it would, since there is no connection AFAICT. Are you seeing low-level cross-talk or something??

wizath, 2019-09-30:
Input power is measured via ADC on SUP_CTRL board. That's an J6 pin 6

hartytp, 2019-09-30:
Aah, apologies, I'd forgotten that only the forward and reverse powers go to the CPU ADC. So, this is U1 we're looking at.

Well, if there is no voltage on J6 pin 6 I guess there is some problem with the pre-amp board. If there is less than 10mV on that pin then it's probably a power supply issue or a dodgy connection, since IIRC the power detector offset is larger than that!

If you can power the module without the supplemental control board then worth checking U5 pin 4 (the detector and OpAmp 3V3 supply), as well as looking at the detector output on R33

hartytp, 2019-09-30:
> That's an J6 pin 6

Wait...this doesn't seem right.

[image]

[image]

Am I going crazy, or is there a mismatch between the pinout on the preamp board and the one on the suplemental control board?

hartytp, 2019-09-30:
On the supplemental board INPWR is J6 pin 6, while it's pin 5 on the pre-amp board. Or is this just a mirroring thing on the connectors due to the board-board mating?

hartytp, 2019-09-30:
I guess it's just a mating thing since all the pins are inverted, never mind.

hartytp, 2019-09-30:
Okay, I don't see an obvious reason, so probably needs to probe the PCB.

wizath, 2019-10-01:
Found the issue, nevermind. U7 is just not mounted on PCB not any error

hartytp, 2019-10-01:
@wizath Is it just U7, or are there other components not mounted as well? e.g. is the power detector itself (U6) mounted?

If the other components are all there, would it be possible to mount the OpAmp on at least one channel so that we can test the input detector firmware code? That way we can sign off on all code working in this design revision.

gkasprow, 2019-10-01:
U7 should be mounted. It is present in documentation, PnP files and BOM

gkasprow, 2019-10-01:
Is it absent in other modules as well ?

hartytp, 2019-10-01:
@wizath can you check this with TS. Did they make a mistake in the BOM or something??

gkasprow, 2019-10-01:
I already asked them about it.

jmilczarek, 2019-10-01:
It was a simple mistake. Someone took wrong boards from the shelf and mounted it in devices. Our crew member is on his way to the laboratory to fix it.

closed 2019-10-01 by hartytp

gkasprow, 2019-10-01:
It looks like it could be an issue with component library. At the moment, some components are from CERN library, some are from Maciek private one, so the resulting BOM and PnP file may cause confusion. I already got all high quality components that meet the compliance so will be able to update the schematics and PCB. These RF modules are the only ones that do not keep compliance with CERN rules

hartytp, 2019-10-01:
Okay, let's aim to do that for the next revision to avoid these problems in future.