sinara-hw/Booster #314: Input detector components not mounted

hartytp, 2019-10-01:
Currently, U7 on the pre-amp PCB is not mounted so the input power detectors don't work. I think this was a decision we made at some point. However, the PDF schematics do not show the component as being DNP, so there is clearly a mistake.

For v1.5 let's add the input detector back in, since it's actually a useful diagnostic and it's now more work to remove it than to keep it.

hartytp, 2019-10-01:
cf https://github.com/sinara-hw/Booster/issues/313

closed 2019-10-01 by hartytp

gkasprow, 2019-10-01:
it looks like the issue with component library. At the moment, some components are from CERN library, some are from Maciek private one, so the resulting BOM and PnP file may cause confusion. I already got all high quality components that meet the compliance so will be able to update the schematics and PCB. These RF modules are the only ones that do not keep compliance with CERN rules