sinara-hw/Booster #315: v1.4 testing

hartytp, 2019-10-08:
Issue to track testing on a new v1.4 Booster using recent firmware (including fixes to bias current calibration).

Fresh from receiving, the log looks like:
```
PGOOD: 1
FAN SPEED: 20 %
AVG TEMP: 26.50 CURRENT: 26.50
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		66:1A	C9:D3	E3:F7	50:07	91:95	E9:D9	E9:F8	1F:A2	
INPWR [V]	0.07	0.00	0.06	0.08	0.02	0.00	0.28	0.01	
TXPWR [V]	0.14	0.07	0.08	0.03	0.14	0.01	0.02	0.01	
RFLPWR [V]	0.14	0.06	0.06	0.01	0.05	0.26	0.02	0.06	
INPWR [dBm]	-19.00	-19.00	-19.00	-19.00	-19.00	-19.00	-19.00	-19.00	
TXPWR [dBm]	5.00	5.00	5.00	5.00	5.00	5.00	5.00	5.00	
RFLPWR [dBm]	-3.68	-4.73	-4.57	-4.70	-4.66	-3.64	-0.01	-4.63	
I30V [A]	0.050	0.051	0.052	0.051	0.052	0.051	0.052	0.051	
I6V0 [A]	0.252	0.247	0.249	0.242	0.238	0.249	0.240	0.254	
5V0MP [V]	4.926	4.934	4.914	4.938	4.928	4.940	4.954	4.944	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		225	122	132	49	231	13	40	10	
ADC2		227	99	94	13	73	427	27	101	
INTSET [dBm]	35.00	35.00	34.99	35.00	35.00	34.99	34.99	35.00	
DAC1		4095	4095	4095	4095	4095	4095	4095	4095	
DAC2		3819	2823	3753	3989	3802	3795	3848	3779	
SCALE1		78	56	80	88	80	85	84	88	
OFFSET1		652	468	619	558	658	485	538	382	
BIASCAL		1519	1789	1327	1531	1645	1749	1623	1883	
HWIS		83.83	57.42	81.92	86.75	82.08	84.17	82.92	83.25	
HWIO		884.83	813.42	886.92	952.75	929.08	850.17	946.92	865.25	
LTEMP		26.25	26.25	27.25	27.25	27.25	26.00	26.00	26.00	
RTEMP		26.50	26.50	25.00	25.00	25.00	26.25	26.50	26.25	
==============================================================================
```

Very low spread on the bias currents, which is excellent!

hartytp, 2019-10-08:
Here are some statistics (from the newer HITL test code) on the currents from this new model. Looks like lower noise than the last unit I was testing, which is interesting. Will see if it still performs this nicely after a few tens of thousands of power cycles etc 

```
Testing Booster: Version(fw_rev='v1.4.1', fw_hash='254615d', fw_build_date=datetime.datetime(2019, 10, 2, 11, 17, 10), device_id='30374705', hw_rev='hw rev 1.4')
Channel 0 (from 1000 samples)...
29V current: mean 51.163 mA, min 51.163, max 51.163 mA, std 0.000 mA
6V current: mean 251.912 mA, min 251.767, max 252.416 mA, std 0.076 mA
Channel 1 (from 1000 samples)...
29V current: mean 52.201 mA, min 52.100, max 52.287 mA, std 0.093 mA
6V current: mean 247.076 mA, min 245.398, max 247.477 mA, std 0.149 mA
Channel 2 (from 1000 samples)...
29V current: mean 53.218 mA, min 53.037, max 53.224 mA, std 0.035 mA
6V current: mean 249.161 mA, min 247.607, max 249.687 mA, std 0.183 mA
Channel 3 (from 1000 samples)...
29V current: mean 51.628 mA, min 51.538, max 51.725 mA, std 0.094 mA
6V current: mean 241.684 mA, min 241.628, max 242.148 mA, std 0.105 mA
Channel 4 (from 1000 samples)...
29V current: mean 51.973 mA, min 51.912, max 52.100 mA, std 0.088 mA
6V current: mean 238.525 mA, min 238.379, max 239.029 mA, std 0.079 mA
Channel 5 (from 1000 samples)...
29V current: mean 51.947 mA, min 51.912, max 52.100 mA, std 0.073 mA
6V current: mean 249.417 mA, min 247.737, max 249.947 mA, std 0.191 mA
Channel 6 (from 1000 samples)...
29V current: mean 52.542 mA, min 52.475, max 52.662 mA, std 0.090 mA
6V current: mean 240.607 mA, min 240.458, max 241.108 mA, std 0.082 mA
Channel 7 (from 1000 samples)...
29V current: mean 51.403 mA, min 51.350, max 51.538 mA, std 0.084 mA
6V current: mean 253.867 mA, min 253.846, max 254.366 mA, std 0.091 mA
```

hartytp, 2019-10-08:
Adding some global statistics (e.g. mean/std across all channels):

```
Channel 0 (from 1000 samples)...
29V current: mean 51.162 mA, min 50.975, max 51.163 mA, std 0.014 mA
6V current: mean 251.890 mA, min 251.767, max 252.416 mA, std 0.090 mA
Channel 1 (from 1000 samples)...
29V current: mean 52.197 mA, min 51.912, max 52.287 mA, std 0.094 mA
6V current: mean 246.987 mA, min 245.398, max 247.477 mA, std 0.127 mA
Channel 2 (from 1000 samples)...
29V current: mean 53.138 mA, min 53.037, max 53.224 mA, std 0.093 mA
6V current: mean 249.152 mA, min 247.607, max 249.687 mA, std 0.203 mA
Channel 3 (from 1000 samples)...
29V current: mean 51.545 mA, min 51.538, max 51.725 mA, std 0.037 mA
6V current: mean 241.650 mA, min 241.628, max 242.148 mA, std 0.092 mA
Channel 4 (from 1000 samples)...
29V current: mean 51.915 mA, min 51.912, max 52.100 mA, std 0.020 mA
6V current: mean 238.520 mA, min 238.379, max 239.029 mA, std 0.086 mA
Channel 5 (from 1000 samples)...
29V current: mean 51.930 mA, min 51.912, max 52.100 mA, std 0.055 mA
6V current: mean 249.415 mA, min 247.737, max 249.947 mA, std 0.182 mA
Channel 6 (from 1000 samples)...
29V current: mean 52.477 mA, min 52.475, max 52.662 mA, std 0.020 mA
6V current: mean 240.610 mA, min 240.588, max 241.108 mA, std 0.086 mA
Channel 7 (from 1000 samples)...
29V current: mean 51.353 mA, min 51.163, max 51.538 mA, std 0.026 mA
6V current: mean 253.861 mA, min 253.586, max 254.366 mA, std 0.079 mA
Global statistics:
29V current: mean 51.965 mA, min 51.162, max 53.138 mA, std 0.601 mA
6V current: mean 246.511 mA, min 238.520, max 253.861 mA, std 5.255 mA
```

Next job is to power cycle a few times to check for hysteresis before I start testing. If that looks good, we can conclude that all worked well when the device was first received...

hartytp, 2019-10-08:
Hysteresis test:
- Power booster on
- enable all channels via SCPI interface
- wait 4s
- query bias current via SCPI interface
- power down
- wait a few seconds
- repeat 25 times

Here is the data, first line is the (mean) data from above for reference, rest is new data...

```
51.2 52.2 53.1 51.6 51.9 51.9 52.5 51.4
51.4 52.3 53.4 52.1 52.7 52.7 53.2 52.1
51.5 52.5 53.4 51.9 52.7 52.7 53.2 52.1
51.5 52.3 53.4 51.9 52.5 52.5 53.2 52.1
51.4 52.3 53.4 51.9 52.5 52.5 53.2 52.1
51.4 52.3 53.4 51.9 52.5 52.5 53.  52.1
51.4 52.3 53.4 51.9 52.7 52.5 53.2 52.1
51.4 52.5 53.4 51.9 52.5 52.5 53.  52.1
51.4 52.5 53.4 51.9 52.5 52.5 53.2 52.1
51.4 52.1 53.4 51.9 52.5 52.5 53.  51.9
51.4 52.3 53.4 51.9 52.5 52.5 53.  52.1
51.4 52.3 53.4 51.9 52.5 52.3 53.  51.9
51.4 52.3 53.4 51.9 52.5 52.5 53.  52.1
51.4 52.3 53.4 51.9 52.5 52.3 53.  51.9
51.4 52.3 53.4 51.9 52.5 52.5 53.  51.9
51.4 52.3 53.4 51.9 52.3 52.3 53.  51.9
51.4 52.1 53.4 51.9 52.3 52.3 53.  51.7
51.4 52.1 53.2 51.7 52.3 52.3 52.8 51.7
51.4 52.3 53.2 51.9 52.3 52.3 53.  51.9
51.4 52.3 53.4 51.9 52.3 52.3 53.  51.9
51.4 52.3 53.4 51.9 52.3 52.3 53.  51.9
51.4 52.1 53.4 51.9 52.3 52.3 53.  51.9
51.4 52.3 53.4 51.9 52.3 52.3 53.  51.9
51.4 52.3 53.4 51.9 52.3 52.3 53.  51.9
51.4 52.1 53.4 51.9 52.3 52.3 53.  51.9
51.4 52.1 53.4 51.7 52.3 52.3 53.  51.7

251.9 247. 249.2 241.7 238.5 249.4 240.6 253.9
251.4 246.6 248.6 241.1 238.2 248.9 240.1 253.2
251.2 246.2 248.4 240.7 237.5 248.6 239.7 253.1
251.1 246.  248.4 240.6 237.3 248.5 239.5 252.9
251.1 246.  248.4 240.6 237.3 248.4 239.5 252.9
251.  246.  248.6 240.6 237.2 248.4 239.5 252.9
251.  246.  248.4 240.6 237.3 248.4 239.5 252.9
251.1 246.2 248.4 240.6 237.3 248.5 239.5 252.9
251.1 246.2 248.4 240.6 237.9 248.5 239.5 253.1
251.1 246.  248.4 240.6 237.7 248.4 239.5 252.9
251.1 246.2 248.4 240.6 237.3 248.5 239.5 253.1
251.  246.  248.4 240.6 237.6 248.4 239.5 252.9
251.1 246.2 248.4 240.6 237.3 248.5 239.5 253.1
251.2 246.  248.4 240.6 237.3 248.5 239.5 253.1
251.2 246.2 248.4 240.6 237.5 248.6 239.7 253.1
251.1 246.2 248.4 240.7 237.5 248.5 239.5 253.1
251.1 246.  248.4 240.6 237.3 248.5 239.7 253.1
251.  246.  248.3 240.5 237.2 248.4 239.5 252.9
251.2 246.3 248.5 240.8 237.6 248.6 239.8 253.2
251.  246.  248.4 240.6 237.3 248.6 239.5 252.9
251.  246.  248.4 240.5 237.3 248.4 239.5 253.1
251.  246.  248.4 240.5 237.2 248.4 239.5 252.9
251.  246.  248.4 240.5 237.2 248.4 239.5 252.9
251.  246.  248.4 240.5 237.3 248.4 239.5 253.1
251.  246.  248.4 240.5 237.2 248.4 239.5 252.9
251.  246.  248.3 240.5 237.2 248.4 239.5 253.1
```

hartytp, 2019-10-08:
Small change between the new and old data, but that could be due to warm up time in-between the data sets. No evidence of anything untoward in the data...good!

hartytp, 2019-10-08:
Very happy with that so far. Let's see how it withstands some abuse!

hartytp, 2019-10-08:
Looking again for any funny business with the currents after fuzzing for a while. As before, first line is the currents I measured originally (when I first received the Booster):

```
51.2 52.2 53.1 51.6 51.9 51.9 52.5 51.4
50.8 52.8 53.  51.7 51.9 52.5 52.5 51.4
50.8 52.8 53.  51.7 51.9 52.5 52.7 51.4
50.8 52.8 53.2 51.7 51.9 52.5 52.5 51.4
50.8 52.7 53.2 51.7 51.7 52.5 52.5 51.4
50.8 52.7 53.  51.5 51.7 52.3 52.5 51.4
50.8 52.7 53.  51.5 51.9 52.3 52.5 51.4
50.8 52.7 53.  51.5 51.9 52.3 52.5 51.4
50.8 52.7 53.  51.5 51.7 52.3 52.5 51.4

251.9 247. 249.2 241.7 238.5 249.4 240.6 253.9
251.6 246.7 248.6 240.6 237.9 248.6 240.1 253.3
251.4 246.3 248.4 240.5 237.6 248.4 239.8 253.1
251.2 246.3 248.4 240.3 237.5 248.4 239.7 253.1
251.2 246.2 248.5 240.3 237.5 248.4 239.5 252.9
251.2 246.2 248.4 240.3 237.5 248.4 239.5 252.9
251.1 246.2 248.3 240.3 237.3 248.4 239.5 252.8
249.6 246.  248.3 240.3 237.3 248.3 239.5 252.9
251.  246.  248.3 240.2 237.3 248.3 239.5 252.8
```

NB channel 0 is the channel I've been applying RF to so I expect it to be a little warm (hence lower bias current)

hartytp, 2019-10-08:
All looks good there (all variation can plausibly be explained by thermal effects)

hartytp, 2019-10-08:
Leaving for a few minutes with no RF and fans on to stabilize temperature

hartytp, 2019-10-08:
As before after some time to thermalize:
```
51.2 52.2 53.1 51.6 51.9 51.9 52.5 51.4
51.2 53.  53.2 51.5 51.9 52.5 52.7 51.5
51.2 52.7 53.2 51.5 51.9 52.3 52.7 51.5
51.  52.8 53.2 51.5 51.9 52.3 52.7 51.4
51.  52.7 53.  51.5 51.7 52.3 52.5 51.4
51.  52.7 53.  51.5 51.7 52.3 52.5 51.4

251.9 247. 249.2 241.7 238.5 249.4 240.6 253.9
251.2 246.3 248.4 241.1 237.7 248.9 239.8 253.1
251.2 246.3 248.4 241.  237.6 248.4 239.7 253.1
251.2 246.2 248.4 240.5 237.5 248.4 239.7 252.9
251.1 246.  248.3 240.3 237.3 248.4 239.5 252.9
251.  245.8 248.1 240.1 237.2 248.5 239.5 252.8
```

hartytp, 2019-10-08:
@gkasprow @wizath this is excellent! Apart from the SCPI interface hanging occasionally, this all looks very good. I haven't been able to spot any issues whatsoever in half a day of stress testing. All looks highly reproducible (as one would hope!).

I'll set a logger running overnight using the SCPI interface. If it doesn't crash we'll have some useful data by morning...

hartytp, 2019-10-08:
It already crashed...

I'm pretty close to finishing testing now. @wizath if you can fix the SCPI interface tomorrow we might be able to wrap this whole project up...

hartytp, 2019-10-09:
One interesting thing is that my fuzzing code has occasionally complained that the DAC bias currents can vary by nearly 3 ma (in one case, ch1 varied from `0.0509753781030548` to `0.05378652023963437`). For now I've slackened the tolerance of my checks. But it will be interesting to see what it looks like if we run the logger with/without fuzzing...

hartytp, 2019-10-09:
@wizath still using the same fw, this unit was still running happily after 1000 iterations of the fuzzing with the VCP constantly logging. Not sure why the SCPI interface didn't die this time.

Power-cycled the unit another 10 times to see if the bias currents are still reproducible after many hours of testing. As before, the first line is the numbers I measured when this unit first arrived
```
51.2 52.2 53.1 51.6 51.9 51.9 52.5 51.4
50.8 54.  53.6 52.1 52.5 52.8 52.8 51.7
50.8 54.  53.6 52.1 52.5 52.8 52.8 51.7
50.8 53.8 53.6 52.1 52.3 52.8 52.7 51.7
50.8 53.8 53.6 51.9 52.3 52.7 52.8 51.7
50.8 53.8 53.6 51.9 52.3 52.7 52.7 51.5
50.8 53.8 53.6 51.9 52.3 52.7 52.7 51.7
50.8 53.6 53.6 51.9 52.3 52.7 52.7 51.5
50.8 53.8 53.6 51.9 52.3 52.8 52.8 51.7
50.8 53.8 53.6 52.1 52.3 52.8 52.8 51.7

251.9 247. 249.2 241.7 238.5 249.4 240.6 253.9
251.2 246.3 248.1 240.8 237.7 248.4 239.8 253.1
251.2 246.3 248.  240.6 237.5 248.4 239.5 252.9
251.  246.  248.  240.5 237.2 248.1 239.5 252.5
250.9 246.  248.  240.3 237.2 248.1 239.3 252.5
250.9 246.2 247.9 240.3 237.2 248.1 239.3 252.5
250.9 245.9 247.9 240.2 237.1 248.1 239.3 252.5
250.7 245.8 247.7 240.1 236.9 248.  239.2 252.4
250.7 245.8 247.9 240.2 237.1 248.  239.3 252.5
250.9 245.9 248.1 240.1 236.9 248.  239.3 252.5
`

hartytp, 2019-10-09:
So, we see some variation. Maybe some channels colder because of higher fan speed, and channel 0 a bit hotter due to recent RF application? Anyway, nothing obviously suspicious there so good!!

hartytp, 2019-10-09:
The fuzzing is clearly showing some small software issues (SCPI hands, watchdog resets) but no sign of any I2C errors or anything else suspicious. Certainly no evidence of a hw bug so far which is very relieving!

Final test I want to do on the hw is leave it logging for a few days to check for glitches. Running ch0 at 35dBm, all other channels with no input RF. Logging via VCP because SCPI interface is not sufficiently reliable yet

hartytp, 2019-10-09:
@wizath After nearly 1 hour of logging under the above conditions, the only slightly odd thing I see is the 29V current on channel 1

[image]

All other channels are perfectly flat, but that one has some large ish spikes. Is that just the resolution of the ADC? It seems quite coarse...

[image]

hartytp, 2019-10-09:
For reference, here are the largest variations I saw on channel 0

[image]

hartytp, 2019-10-09:
Much much smaller (and, NB that channel has RF on it)

hartytp, 2019-10-09:
Here is an hour of testing, now driving channel 1 at just over 1W

[image]

The infrequent spikes in the bias current don't seem to correlate with anything on the RF at a noticeable level.

hartytp, 2019-10-09:
Will keep gathering data a while longer, but so far I'm pretty happy with this all modulo the minor firmware bugs.

hartytp, 2019-10-09:
- [ ] to do: hook up V3500A power meter to check there are no spikes missed by the internal power detector

hartytp, 2019-10-15:
Results from my logging on a v1.4 Booster before the watchdog reset kicked in. RF is applied to channel 1 only...

Ch0: 
[image]

ch1:
[image]

ch 2:
[image]

ch3
[image]

ch4
[image]

ch5
[image]

ch6
[image]

ch7
[image]

(there are a few grafana rendering issues in the above data...)

hartytp, 2019-10-15:
Aah, that looks a bit odd because I had grafana plotting means. Replotting with distinct points on a few channels only, we see:

ch1:
[image]

ch2:
[image]

@wizath That makes more sense, but I still don't get it. I assume those steps are quantization e.g. at the ADC, right? But then why are the steps 3mA for the lower bias current, but only 1mA for the higher current when the RF is on?

hartytp, 2019-10-15:
This is a v1.4 booster. Current sense gain is 43V/V*0.1A=4.3V/A=4.3mV/mA

ADC is 12-bit, 3.3V ref, so 1LSB is 0.8mV. So, I would have expected to see quantization noise at around the 200uA level. Any idea why the steps I see on the VCP output are so large, and are different for the different bias currents? Is this something to do with software processing/rounding?

hartytp, 2019-10-16:
All testing I can think of doing is now done, just need to figure out https://github.com/sinara-hw/Booster/issues/325 As part of the investigation into that I'll post more logs of bias current etc, which will help to rule out other so-far undetected issues.

Just need reliable data acquisition now!!!

closed 2019-10-16 by hartytp

gkasprow, 2019-10-17:
This ADC causes so many problems, we should consider replacing it with the other one.