sinara-hw/Booster #316: scpi fails to respond to command

hartytp, 2019-10-08:
@wizath my fuzzing just stalled while waiting for a response from Booster.

Looking at the logstash I see nothing:
```
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	0	0	0	0	
> 
> i2cdetect
Wrong number of parameters
> 
> logstash
[INFO] hardware revision 4
[WARN] watchdog reset occurred!
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1978 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] MACADDR 80:1F:12:5D:E7:F0
[INFO] RF Channel at 0 detected, id = 66:1A
[INFO] RF Channel at 1 detected, id = C9:D3
[INFO] RF Channel at 2 detected, id = E3:F7
[INFO] RF Channel at 3 detected, id = 50:07
[INFO] RF Channel at 4 detected, id = 91:95
[INFO] RF Channel at 5 detected, id = E9:D9
[INFO] RF Channel at 6 detected, id = E9:F8
[INFO] RF Channel at 7 detected, id = 1F:A2
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.79
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
> 
```

Any idea how I can debug this?

hartytp, 2019-10-08:
```
fuzz 155 of 10000: power
fuzz 156 of 10000: status
fuzz 157 of 10000: status
^CTraceback (most recent call last):
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ion/scratch/oxart/oxart/devices/booster/tests.py", line 337, in <module>
    print(sys.argv[0])
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/main.py", line 95, in __init__
    self.runTests()
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/main.py", line 256, in runTests
    self.result = testRunner.run(self.test)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/runner.py", line 176, in run
    test(result)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/suite.py", line 84, in __call__
    return self.run(*args, **kwds)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/suite.py", line 122, in run
    test(result)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/suite.py", line 84, in __call__
    return self.run(*args, **kwds)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/suite.py", line 122, in run
    test(result)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/case.py", line 653, in __call__
    return self.run(*args, **kwds)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/case.py", line 605, in run
    testMethod()
  File "/home/ion/scratch/oxart/oxart/devices/booster/tests.py", line 329, in test_fuzz
    tests[test_idx][5:]))
  File "/home/ion/scratch/oxart/oxart/devices/booster/tests.py", line 150, in wrapped
    def wrapped(self, *args, **kwargs):
  File "/home/ion/scratch/oxart/oxart/devices/booster/tests.py", line 245, in test_status
    self.assertApproxEq(status.temp, dev.get_temperature(chan), 2)
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 184, in get_fan_speed
    return self._query_float("MEAS:FAN?", None)
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 56, in _query_float
    resp = self._cmd(cmd, channel, arg)
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 35, in _cmd
    response = self.dev.readline().decode().lower().strip()
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/site-packages/serial/urlhandler/protocol_socket.py", line 158, in read
    ready, _, _ = select.select([self._socket], [], [], timeout.time_left())
KeyboardInterrupt
```

hartytp, 2019-10-08:
@wizath apologies, I'm being dumb! There is a watchdog reset in the logstash!

hartytp, 2019-10-08:
Didn't we agree that after a watchdog reset Booster would boot up with all red LEDs? Is this not implemented, or am I running the wrong fw rev? Any idea how to debug the cause of the watchdog reset?

hartytp, 2019-10-08:
Got another crash while fuzzing. This time I had the VCP running in `start`  mode (last time I did have a VCP connection, but not in `start` mode). This time nothing in the logstash:

```
> version
RFPA v1.4.1 254615d, built Oct  2 2019 12:23:57, hv rev 4
> 
> i2cerr 
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	0	0	0	0	
> 
> logstash
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
> 
```

hartytp, 2019-10-08:
I'll close the VCP terminal and see if I still get these freezes...

hartytp, 2019-10-08:
I don't see that hash anywhere in the booster-firmware logs. What firmware revision is that? Is it the right one to test??

hartytp, 2019-10-08:
The good news is that so far I don't see any I2C errors or any other sign of issue with the hw other than ethernet freezes, so that's a big step forward!!

hartytp, 2019-10-08:
Okay, still see this even with the VCP closed. :( Still nothing in logstash
```
logstash
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
[ERROR] Interlock tripped on channel 0, i=0 o=1
> 
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	0	0	0	0	
> 
> 
```

```
fuzz 84 of 10000: power
^CTraceback (most recent call last):
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ion/scratch/oxart/oxart/devices/booster/tests.py", line 338, in <module>
    print("fuzz {} of {}: {}".format(test_num, num_iterations,
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/main.py", line 95, in __init__
    self.runTests()
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/main.py", line 256, in runTests
    self.result = testRunner.run(self.test)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/runner.py", line 176, in run
    test(result)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/suite.py", line 84, in __call__
    return self.run(*args, **kwds)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/suite.py", line 122, in run
    test(result)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/suite.py", line 84, in __call__
    return self.run(*args, **kwds)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/suite.py", line 122, in run
    test(result)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/case.py", line 653, in __call__
    return self.run(*args, **kwds)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/unittest/case.py", line 605, in run
    testMethod()
  File "/home/ion/scratch/oxart/oxart/devices/booster/tests.py", line 330, in test_fuzz
    tests = [method_name for method_name in dir(self) if (
  File "/home/ion/scratch/oxart/oxart/devices/booster/tests.py", line 155, in wrapped
    if "?" in cmd:
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 132, in get_status
    resp = self._cmd("CHAN:DIAG?", channel).split(',')
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 35, in _cmd
    response = self.dev.readline().decode().lower().strip()
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/site-packages/serial/urlhandler/protocol_socket.py", line 158, in read
    ready, _, _ = select.select([self._socket], [], [], timeout.time_left())
KeyboardInterrupt
```

hartytp, 2019-10-08:
will try power cycling booster...

hartytp, 2019-10-08:
After a connection hangs, I can kick start it by opening a new connection and requesting data (although, the new connection then hangs)...

hartytp, 2019-10-08:
Running two SCPI sessions at once (e.g. a fuzzer and a logger) really does not work at all.

hartytp, 2019-10-08:
```
> ethdbg
getSn_SR 20
getSn_MR 1
getSn_RX_RSR 0
getSn_CR 0
getSn_IR 0
getSn_RXBUF_SIZE 2
> 
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	0	0	0	0	
> 
> logstash
[INFO] hardware revision 4
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1977 | VrefInt 1.21 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] MACADDR 80:1F:12:5D:E7:F0
[INFO] RF Channel at 0 detected, id = 66:1A
[INFO] RF Channel at 1 detected, id = C9:D3
[INFO] RF Channel at 2 detected, id = E3:F7
[INFO] RF Channel at 3 detected, id = 50:07
[INFO] RF Channel at 4 detected, id = 91:95
[INFO] RF Channel at 5 detected, id = E9:D9
[INFO] RF Channel at 6 detected, id = E9:F8
[INFO] RF Channel at 7 detected, id = 1F:A2
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.79
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
[INFO] network client 10.255.6.219 connected
[INFO] [tempmgt] Temp 25.500000 (d 20) -> Fan 20
[INFO] network client disconnected
[INFO] network client 10.255.6.219 connected
[INFO] network client disconnected
> 
> version
RFPA v1.4.1 254615d, built Oct  2 2019 12:23:57, hv rev 4
> 
> 

```

hartytp, 2019-10-09:
ping @wizath are you able to look at this today? Or, can you give me any advice on further diagnostics to run?

wizath, 2019-10-09:
First of all, sorry because that v1.4's got some strange magic firmware that I was using to diagnose I2C errors.

[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3706484/eth-scpi.dfu.zip)

Flash this, it contains all recent fixes (along with experimental ADC I2C delays to eliminate these stalls in I2C transactions)

hartytp, 2019-10-09:
Thanks will recheck.

Is the firmware repo up to date?

hartytp, 2019-10-14:
@wizath what's the status of this? Are you able to reproduce on your hw?

wizath, 2019-10-15:
Got minor issue, while spamming one socket for prolonged time, DHCP thread doesn't get enough time and it can fail. I'll ramp it's priority

wizath, 2019-10-15:
Apart from that it works very well, no issue after a day of (continuous) use

closed 2019-10-15 by wizath

hartytp, 2019-10-15:
Can you post new fw and I’ll test again

wizath, 2019-10-15:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3729824/eth-scpi.dfu.zip)

I'll push rc2 when I'll fix that watchdog

hartytp, 2019-10-15:
Thanks! I’ll do some more tests later today