sinara-hw/Booster #319: watchdog reset

hartytp, 2019-10-09:
This just happened:
```
logstash
[INFO] hardware revision 4
[WARN] watchdog reset occurred!
> 
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	0	0	0	0	
> 
> version
RFPA v1.4.1 33f0541, built Oct  9 2019 10:58:53, hv rev 4
> 
> 
```

All red LEDs came on, which is good. Any idea what the cause is?

wizath, 2019-10-09:
It won't show any I2C error after reset. Interesting - Mine v1.3 is running flawlessly for like 5h now

wizath, 2019-10-09:
It's expected for SCPI to hang after watchdog reset

hartytp, 2019-10-09:
I've seen several watchdog errors on this unit (but only while fuzzing). Maybe it's a command that I use in my fuzzer that you've taken out?

wizath, 2019-10-09:
Post me the script, I'll try to recreate

hartytp, 2019-10-09:
Okay, will do. Are there any more diagnostics we can add?

hartytp, 2019-10-09:
```python
import numpy as np
import argparse
import unittest
import time
import random
import sys

from artiq.protocols.pc_rpc import Client

from oxart.devices.scpi_synth.driver import Synth
from oxart.devices.booster.driver import Booster

parser = argparse.ArgumentParser(description="Booster hardware in the loop "
                                 "test suite. Connect one channel of Booster "
                                 "to a synth and RF power meter. Assumes there"
                                 " is a 30dB attenuator between Booster and "
                                 "the power meter.")
parser.add_argument("--booster", help="IP address of the Booster to test")
parser.add_argument("--synth", help="Address of the synth to use for tests")
parser.add_argument("--meter", help="IP address of the power meter RPC server")
parser.add_argument("--p_min",
                    help="Minimum synth power to use, set to give ~25dBm "
                    "output power",
                    type=float)
parser.add_argument("--p_max",
                    help="Maximum synth power to use, set to give ~35dBm "
                    "output power",
                    type=float)
parser.add_argument("--chan",
                    help="Booster channel connected to synth + power meter",
                    type=int,
                    default=0)
args = None

class TestBooster(unittest.TestCase):

    @classmethod
    def setUpClass(cls):
        cls.dev = Booster(args.booster)
        print("Testing Booster: {}".format(cls.dev.get_version()))

        # level of stability we expect for the current measurements
        # we allow a wider margin for the channel under test (cut) because
        # of thermal hysteresis (as it gets hotter, the bias current decreases
        # a little)
        cls.I29V_tol = 2.5e-3
        cls.I29V_cut_tol = 3e-3
        cls.I6V_tol = 2e-3
        cls.I6V_cut_tol = 3e-3

        cls.t_settle = 4  # delay (s) between enabling channels and taking data

        for channel in range(8):
            cls.dev.set_enabled(channel)

        cls.cut = args.chan
        cls.synth = Synth(args.synth)
        cls.synth.set_freq(200e6)
        cls.synth.set_rf_on(False)

        cls.meter = Client(args.meter, 4300)
        cls.meter.set_freq(200)

        time.sleep(cls.t_settle)

        # get a baseline measurement of the currents on each channel
        num_measurements = 100
        cls.I29V = np.zeros((8, num_measurements))
        cls.I6V = np.zeros((8, num_measurements))
[... 242 lines cut ...]
            self.assertApproxEq(detector_err, self.detector_err, 0.5)

        self.synth.set_rf_on(False)
        time.sleep(0.1)
        self.dev.clear_interlock(args.chan)
        time.sleep(self.t_settle)

    def test_fuzz(self):
        skip = ["test_fuzz"]
        tests = [method_name for method_name in dir(self) if (
            callable(getattr(self, method_name))
            and method_name.startswith("test_")
            and method_name not in skip)]
        num_iterations = 10000
        for test_num in range(num_iterations):
            test_idx = random.randint(0, len(tests)-1)
            test = tests[test_idx]
            print("fuzz {} of {}: {}".format(test_num, num_iterations,
                                             tests[test_idx][5:]))
            getattr(self, test)()

if __name__ == "__main__":
    args = parser.parse_args()
    unittest.main(argv=sys.argv[0:1])
```

wizath, 2019-10-09:
One second

wizath, 2019-10-09:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3707240/eth-scpi.dfu.zip)

Commented out "disabling channel on I2C error" - that's the only thing I can think of that causes the issue with watchdog

hartytp, 2019-10-09:
Can we add more diagnostics to the log so that we can see more about the cause of the watchdog timeout?

FWIW, the code has now been fuzzing for half an hour without issue. Previously, my code had crashed a few times (e.g. having the thresholds for current measurements set too conservatively, so failed assertions) so I was opening/closing network connections etc. Not sure if this is part of the cause. Anyway, I'll leave it fuzzing and see how it gets on

hartytp, 2019-10-09:
> Commented out "disabling channel on I2C error" - that's the only thing I can think of that causes the issue with watchdog

I don't think we're going to track this down like that. The watchdog resets are really quite infrequent, and I don't have any good way of reproducing them (even on the same firmware, my fuzz script that gave a watchdog reset quickly before has been fuzzing for an hour now without issue). So, it's very hard to tell if individual changes make a difference.

I think we need to add more diagnostics to tell what caused the reset.

wizath, 2019-10-09:
I'll think about more diagnostic way to handle watchdog resets.

hartytp, 2019-10-10:
@wizath this happened again, this time without the SCPI interface (I was just logging over VCP using the `start` command). We need more debug info, because this is a serious problem atm.

hartytp, 2019-10-14:
@wizath how are you getting on with this? ATM both remote interfaces aren't much use as they kill booster after a day or so...

wizath, 2019-10-14:
I can't really find the cause with the v1.3 I'm using ATM since it's working fine. You can post me the v1.4 that triggers watchdog and that broken channel and I'll send you working v1.3 unit - I think that would be the fastest way to resolve this problem

hartytp, 2019-10-14:
Ok. Let's do that

hartytp, 2019-10-14:
@wizath Before sending this back, I'll leave the VCP log running overnight on a couple of our other Boosters and see if I can recreate the issue.

gkasprow, 2019-10-14:
Maybe they used the screws that have diameter bigger than the pad. They can short with the component leads.

hartytp, 2019-10-14:
No, it's not that. I put the screw on a lathe to make the head smaller and the problem persisted. It appears to be related to the screw pressure, which is why I suspect an issue on the rear side of the PCB

hartytp, 2019-10-16:
@wizath I just reproduced this quickly three times in a row as follows:
- running with a new Technosystem v1.4
- `version` gives `RFPA v1.4.1 ade44e9, built Oct 15 2019 16:28:01, hv rev 4 `
- turn all channels off standby
- run the VCP logger and the SCPI loggers below simultaneously
- wait for a minute or two and all red lights are on and both the VCP and SCPI ports become non-responsive
- reboot booster and see the following logstash. NB I don't see any mention of the watchdog reset

```
logstash
[INFO] hardware revision 4
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1969 | VrefInt 1.20 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] MACADDR 80:1F:12:5E:33:BF
[INFO] RF Channel at 0 detected, id = 91:E5
[INFO] RF Channel at 1 detected, id = 7B:26
[INFO] RF Channel at 2 detected, id = C6:B0
[INFO] RF Channel at 3 detected, id = E7:8D
[INFO] RF Channel at 4 detected, id = 2B:2E
[INFO] RF Channel at 5 detected, id = E5:92
[INFO] RF Channel at 6 detected, id = 91:DF
[INFO] RF Channel at 7 detected, id = AE:20
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.77
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
> 

```

```python
import argparse
from influxdb import InfluxDBClient
from influxdb.exceptions import InfluxDBClientError
from requests.exceptions import ConnectionError

from oxart.devices.booster.driver import Booster

def get_argparser():
    parser = argparse.ArgumentParser()
    parser.add_argument("--device", required=True,
                        help="Booster IP address")
    parser.add_argument("--name", required=True,
                        help="logical Booster name, defines measurement name")
    parser.add_argument("--influx-server", default="x.x.x.x",
                        help="Influx server address")
    parser.add_argument("--database", default="junk",
                        help="Influx database name")
    return parser

def write(client, booster_name, measurement, data):
    point = {
        "measurement": booster_name,
        "tags": {
            "name": measurement,
            },
        "fields": {"ch{}".format(idx): value for idx, value in enumerate(data)}
    }
    try:
        client.write_points([point])
    except ConnectionError:
        print("ConnectionError: Influxdb down?")
    print(point)
[... 136 lines cut ...]

        def write_channels(name, values):
            fields = {}
            for i, v in enumerate(values):
                fields["ch{}".format(i)] = v
            write_point(args, client, name, fields)

        try:
            write_channels("temp", [(x+y)/2 for x, y in zip(status["LTEMP"],
                                                            status["RTEMP"])])
            write_channels("i_30V", status["I30V [A]"])
            write_channels("i_6V", status["I5V0 [A]"])
            write_channels("5V0MP", status["5V0MP [V]"])
            write_channels("on", status["ON"])
            write_channels("son", status["SON"])
            write_channels("pwr_tx", status["TXPWR [dBm]"])
            write_channels("pwr_rfl", status["RFLPWR [dBm]"])
        except KeyError as e:
            print("Missing key: {}".format(e))
        except InfluxDBClientError as e:
            print("Data error: {}".format(e))

if __name__ == "__main__":
    main()
```

hartytp, 2019-10-16:
NB the watchdog only seems to trigger when the channels are enabled (probably not surprising).

Please can you check ASAP if you can reproduce this? If not, can you either post new firmware with more diagnostics to help track this down, or I can post the booster back to you for investigation...

hartytp, 2019-10-17:
@wizath ping...any luck reproducing this???!

wizath, 2019-10-18:
Send me over the v1.4 hardware, I've been running exact script like yours (spamming chan:diag commands while gathering status messages from VCP) for several hours and everything works fine

hartytp, 2019-10-18:
@wizath I cannot understand why you can't reproduce this. I just ran the VCP only logger below. After a few minutes it stopped and all red leds were on. I power cycled the booster and now I have a mixture of red and green.

Ethernet isn't even plugged in this time.

This is on v1.3 hardware (Creotech), with one channel removed. logstash doesn't have anything useful since it's just lots of errors about the non connected channel

```
logstash
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
> 
> version
RFPA v1.4.1 ade44e9, built Oct 15 2019 16:28:01, hv rev 3
> 
```

```python
import serial
import argparse

def parse_status_table(lines):
    """Parse the status table into a dictionary with parameter names as keys,
    with values being vectors of parameter values over the channels.
    An example line of the table is:
    IN8V0 [mA]\t0.039\t0.018\t0.013\t0.021\t0.008\t0.109\t0.008\t0.010
    """
    # print(lines)

    status = {}
    for line in lines:
        # The description can contain spaces, so we special case this
        _DESC_LEN = 16
        desc = line[:_DESC_LEN].strip()

        parts = line.strip().split("\t")
        desc = parts[0]
        ch_parts = parts[1:]

        # Skip the channel number template
        if len(desc) == 0:
            continue
        chs = [p for p in ch_parts if len(p) > 0]

        try:
            vals = [float(x) for x in chs]
        except ValueError:
            vals = chs
        status[desc] = vals
    return status

def get_messages(h):
    in_block = False
    lines = []

    while True:
        line = h.readline().decode()

        contains_marker = line.startswith("="*70)

        if in_block:
            if contains_marker:
                # If the marked block is long enough to contain a valid table..
                if len(lines) > 20:
                    in_block = False
                    status = parse_status_table(lines)
                    yield status
                else:
                    # Treat this as a start marker rather than an end marker
                    in_block = True
                lines = []
            else:
                lines.append(line)

        if not in_block and contains_marker:
            in_block = True

def get_argparser():
    parser = argparse.ArgumentParser()
    parser.add_argument("--device", required=True,
                        help="Device serial port address")
    return parser

def print_data(name, fields):
    print("{}: {}".format(name, fields))

def main():
    args = get_argparser().parse_args()

    h = serial.Serial(args.device)

    # Ensure the logging task is running
    h.write("start\r\n".encode())

    for status in get_messages(h):

        def write_channels(name, values):
            fields = {}
            for i, v in enumerate(values):
                fields["ch{}".format(i)] = v
            print_data(name, fields)

        write_channels("temp", [(x+y)/2 for x, y in zip(status["LTEMP"],
                                                        status["RTEMP"])])
        write_channels("i_30V", status["I30V [A]"])
        write_channels("i_6V", status["I5V0 [A]"])
        write_channels("5V0MP", status["5V0MP [V]"])
        write_channels("on", status["ON"])
        write_channels("son", status["SON"])
        write_channels("pwr_tx", status["TXPWR [dBm]"])
        write_channels("pwr_rfl", status["RFLPWR [dBm]"])

if __name__ == "__main__":
    main()
```

Please can you run the VCP logger exactly as it is on that same firmware revision and check you don't see errors?

hartytp, 2019-10-18:
Ok...now whenever I boot that booster up I get a mixture of green and red LEDs on...what's up? Is this some issue with the watchdog reset feature going crazy if a channel is missing? Can I clear the flash memory to remove it?

hartytp, 2019-10-18:
so...now all channels have red LEDs, but channels 1 and 5 also have green LEDs...

```
> start
␛[2J
PGOOD: 1
FAN SPEED: 0 %
CHANNELS INFO
====> 
> ==========================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	0	0	
HWID		8E:95	6D:4F	91:68	21:53	8F:BF	3D:90	00:00	00:00	
INPWR [V]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
TXPWR [V]	0.16	0.12	0.13	0.13	0.13	0.13	0.56	0.13	
RFLPWR [V]	0.10	0.04	0.01	0.01	0.01	0.01	0.40	0.01	
INPWR [dBm]	nan	0.00	nan	nan	nan	0.00	0.00	0.00	
TXPWR [dBm]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
RFLPWR [dBm]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00	
I30V [A]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	
I5V0 [A]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	
5V0MP [V]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000	
ON		0	0	0	0	0	0	0	0	
SON		0	1	0	0	0	1	0	0	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		256	203	208	205	210	206	915	205	
ADC2		157	67	15	9	24	9	661	8	
INTSET [dBm]	33.00	32.00	33.97	32.00	32.93	36.49	0.00	0.00	
DAC1		4095	4095	4095	4095	4095	4095	0	0	
DAC2		3382	3336	3463	3267	3356	3775	0	0	
SCALE1		80	81	79	82	88	87	0	0	
OFFSET1		584	576	538	569	315	458	0	0	
BIASCAL		1417	1629	1523	1209	1317	1341	0	0	
HWIS		85.50	84.00	89.25	83.67	87.92	87.67	0.00	0.00	
HWIO		560.50	648.00	431.25	589.67	460.92	575.67	0.00	0.00	
LTEMP		63.00	63.00	63.00	63.00	63.00	63.00	0.00	0.00	
RTEMP		63.00	63.00	63.00	63.00	63.00	63.00	0.00	0.00	
==============================================================================
stop
> 
> status 0
[status] e=0 s=0 r1=254 r2=163 tx=0.000 rf=0.000 curr=0.000 t=63.00 i=0.00 ip=nan
> 
> status 1
[status] e=0 s=1 r1=202 r2=65 tx=0.000 rf=0.000 curr=0.000 t=63.00 i=0.00 ip=0.00
> 
> status 2
[status] e=0 s=0 r1=210 r2=15 tx=0.000 rf=0.000 curr=0.000 t=63.00 i=0.00 ip=nan
> 
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	0	0	22396	0	
> 
> logstash
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 6
```

so, why all the red LEDs?

hartytp, 2019-10-18:
```
> i2cdetect 1
[i2c_scan] start
[i2c_scan] end
> 
> i2cdetect 0
[i2c_scan] start
[i2c_scan] end
> 
> i2cdetect 1
[i2c_scan] start
[i2c_scan] end
> 
> i2cdetect 2
[i2c_scan] start
[i2c_scan] end
> 
> 
```

hartytp, 2019-10-18:
Going back to the v1.4 that was crashing reliably earlier on. I've run my scripts a few times now and sometimes they ran for a long time without killing the booster, while sometimes it dies quickly.

Is there nothing you can do to give us better debugging info about what Booster is doing when the watchdog trips? Otherwise figuring out these kind of intermittent issues is going to be a nightmare.

hartytp, 2019-10-18:
@wizath we need to fix this urgently as we can't make any progress with booster while the interfaces are this unreliable. We need better diagnostics on this...

hartytp, 2019-10-18:
NB with the v1.4 the times when all red LEDs come on, the logstash after reebot does not contain a `watchdog` reset. So, maybe this isn't a watchdog issue?

Somehow, I wonder if this is related to the ethernet freezing sometimes?

hartytp, 2019-10-18:
Isn't it easy to use a an early warning interrupt to record what code is being executed when the watchdog timer fires and then save that to flash?

hartytp, 2019-10-18:
@wizath since I these errors don't seem to produce watchdog reset messages in the log, are they actually hard faults or something? do you have proper panic handlers for those?

Can we get info dumped to the VCP when this kind of thing happens??

hartytp, 2019-10-18:
For whatever reason, I'm finding this issue hard to reproduce this evening. Don't know why. For whatever reason, I haven't been able to reproduce the errors when replacing my VCP logger script with just using `miniterm`. No idea why. So, I don't think that just shipping this booster back to you will help. We really need better diagnostics so I can create a proper crash report

hartytp, 2019-10-18:
@wizath okay, I have finally managed to reproduce this using `miniterm` instead of the VCP logger. For some reason it took me ages to reproduce the bug in this way. Before I saw it, I'd power cycled booster many times, closed and opened connections, etc etc

hartytp, 2019-10-21:
@wizath ping....any progress adding better diagnostics to help track this down?

hartytp, 2019-10-22:
@wizath ping

hartytp, 2019-10-23:
@wizath ping...ping...ping...

hartytp, 2019-10-24:
With the latest firmware my Boosters go to all RED LEDs reliably every day or so, even without any logging etc.

hartytp, 2019-10-24:
I plan to send this booster to Creotech for debugging. Before I do that, I want to make sure that I know how to reproduce the issue. So far, I have observed that with the USB + ethernet plugged in (although, no active connections) the Booster always goes into an error state after some time (a day or so). I want to verify that this isn't related to ethernet/VCP (e.g. if some traffic on our network upsets Booster).

To confirm, I've left a Booster v1.4 powered, with all channels enabled, but with the USB and ethernet cables unplugged. If this goes into an error state then I'll send it to Creotech immediately. If it's still all green on Monday, I'll have to think further about what to do.

hartytp, 2019-10-28:
Okay, this is interesting. I've left this running over the weekend with the USB and Ethernet unplugged and no red LEDs...I've just plugged the Ethernet back in. Will see what happens. My suspicion is that some ethernet traffic on our network is causing this issue. If that's the case it may be hard to determine the exact cause without more diagnostics...

hartytp, 2019-10-29:
Plugged the ethernet back in, but still no red LEDs. I'll plug USB back in and see if that makes any difference...

dnadlinger, 2019-11-09:
@wizath Any progress on this? I've still got ~2 Booster failures a day due to this.

It also appears that the likelihood of this problem occurring is vastly increased if both Ethernet and USB (VCP) are connected. All the 6 Boosters in my lab are connected to the (same) network, and I think I've only seen one or two failures on devices not connected to VCP as well. 

(Even on those connected via USB, the console is idle; there are neither commands being sent nor is the watch thread running. All are running version v1.4.1 33f0541, with the exception of one v1.4.1 ade44e9.)

gkasprow, 2019-11-10:
Two colleagues from CTI are assigned to solve the issue. They started on Wednesday.

dnadlinger, 2019-11-10:
Great to hear!

gkasprow, 2019-11-26:
My coleagues are fighting with the issues. One observation is that with debugger connected they did not observe any issues. Without debugger it hangs after a few hours of testing. The statistics is not high but we have at least some idea

hartytp, 2019-11-26:
Okay, really glad to hear there is some progress! These embedded issues can be horrible to debug.

gkasprow, 2019-11-28:
We are just curious if hanging VBAT may cause some issues. I don't think so but we will short it to 3V3.
Guys did a careful review of the code and found a few possible but not very probable places where issues can arrive. They are testing one by one but since the errors occur very rarely it takes time.

hartytp, 2019-12-04:
Creotech have done some great work on the firmware and identified/fixed some I2C issues. The hope is that this issue is now resolved.

We have a few Boosters running on the new firmware, including one that is constantly logging on the VCP+SCPI ports. Let's see what happens...

hartytp, 2019-12-04:
If these units are still running happily on Monday with no red LEDs we'll close the issue.

hartytp, 2019-12-04:
I left my VCP + SCPI loggers running simultaneously. After a few hours the loggers died and Booster had all RED leds.

```
version
RFPA v1.4.1 ade44e9, built Nov 20 2019 15:34:57, hv rev 3
> 
> logstash
[INFO] hardware revision 3
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1974 | VrefInt 1.20 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] MACADDR 80:1F:12:56:1F:21
[INFO] RF Channel at 0 detected, id = 21:4B
[INFO] RF Channel at 1 detected, id = DE:0E
[INFO] RF Channel at 2 detected, id = A8:53
[INFO] RF Channel at 3 detected, id = 5B:EA
[INFO] RF Channel at 4 detected, id = FB:68
[INFO] RF Channel at 5 detected, id = 02:41
[INFO] RF Channel at 6 detected, id = B5:46
[INFO] RF Channel at 7 detected, id = 1F:B8
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.249
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
> 
> 
```

As far as I understand it, Creotech have found hard faults which they believe to be due to memory leaks in the SCPI interface. They are still investigating. So, maybe this is not a surprise.

I'll try now with the ethernet still connected, but no SCPI logger running. I'll only run the VCP logger and see what happens.

hartytp, 2019-12-05:
Without SCPI my logger has been running fine since yesterday afternoon.

hartytp, 2019-12-05:
still no crashes when not using the SCPI interface...

hartytp, 2019-12-06:
still going strong

hartytp, 2019-12-09:
Without the SCPI interface I haven't been able to produce any crashes over a weekend of continuous logging on the VCP port (read only by scraping the output of the `start` command).

SCPI interface is still prone to crashing, but that's progress at least.

closed 2019-12-09 by hartytp