sinara-hw/Booster #320: can't run two SCPI connections at once

hartytp, 2019-10-09:
I tried running this logger while fuzzing:

```python

import argparse
from influxdb import InfluxDBClient
from influxdb.exceptions import InfluxDBClientError
from requests.exceptions import ConnectionError

from oxart.devices.booster.driver import Booster

def get_argparser():
    parser = argparse.ArgumentParser()
    parser.add_argument("--device", required=True,
                        help="Booster IP address")
    parser.add_argument("--name", required=True,
                        help="logical Booster name, defines measurement name")
    parser.add_argument("--influx-server", help="Influx server address")
    parser.add_argument("--database", default="junk",
                        help="Influx database name")
    return parser

def write(client, booster_name, measurement, data):
    point = {
        "measurement": booster_name,
        "tags": {
            "name": measurement,
            },
        "fields": {"ch{}".format(idx): value for idx, value in enumerate(data)}
    }
    try:
        client.write_points([point])
    except ConnectionError:
        print("ConnectionError: Influxdb down?")
    print(point)

def main():
    args = get_argparser().parse_args()

    client = InfluxDBClient(
            host=args.influx_server,
            database=args.database)

    dev = Booster(args.device)

    cmds = {"temp": "temp",
            "i_30V": "I29V",
            "i_6V": "I6V",
            "5V0MP": "V5VMP",
            "pwr_tx": "output_power",
            "pwr_rfl": "input_power"}

    while True:
        try:
            for key, value in cmds.items():
                data = [0]*8
                for channel in range(8):
                    status = dev.get_status(channel)
                    data[channel] = getattr(status, value)
                write(client, args.name, key, data)
        except InfluxDBClientError as e:
            print("Data error: {}".format(e))

if __name__ == "__main__":
    main()

```

hartytp, 2019-10-09:
This logger pretty reliably hangs quickly:
```
{'measurement': 'tph_v1.4', 'fields': {'ch7': 5.0, 'ch3': 5.0, 'ch0': 33.3589743589744, 'ch1': 5.0, 'ch5': 5.0, 'ch4': 5.0, 'ch2': 5.0, 'ch6': 5.0}, 'tags': {'name': 'pwr_tx'}}
{'measurement': 'tph_v1.4', 'fields': {'ch7': 0.252936305614717, 'ch3': 0.240588434541302, 'ch0': 0.249426899546465, 'ch1': 0.246567414824237, 'ch5': 0.2483870869202, 'ch4': 0.237468977435203, 'ch2': 0.248127133763634, 'ch6': 0.239678579266016}, 'tags': {'name': 'i_6V'}}
{'measurement': 'tph_v1.4', 'fields': {'ch7': -19.0, 'ch3': -19.0, 'ch0': -6.74757194519043, 'ch1': -19.0, 'ch5': -19.0, 'ch4': -19.0, 'ch2': -19.0, 'ch6': -19.0}, 'tags': {'name': 'pwr_rfl'}}
{'measurement': 'tph_v1.4', 'fields': {'ch7': 25.5, 'ch3': 24.25, 'ch0': 26.0, 'ch1': 24.0, 'ch5': 24.25, 'ch4': 24.25, 'ch2': 23.0, 'ch6': 24.5}, 'tags': {'name': 'temp'}}
{'measurement': 'tph_v1.4', 'fields': {'ch7': 0.0519124254819147, 'ch3': 0.0522872485919222, 'ch0': 0.000937047487153202, 'ch1': 0.0539739317946382, 'ch5': 0.053224295970782, 'ch4': 0.0526620647711571, 'ch2': 0.0537865202396344, 'ch6': 0.0530368844157783}, 'tags': {'name': 'i_30V'}}
^CTraceback (most recent call last):
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ion/scratch/oxart/oxart/devices/booster/logger.py", line 66, in <module>
    main()
  File "/home/ion/scratch/oxart/oxart/devices/booster/logger.py", line 58, in main
    status = dev.get_status(channel)
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 132, in get_status
    resp = self._cmd("CHAN:DIAG?", channel).split(',')
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 35, in _cmd
    response = self.dev.readline().decode().lower().strip()
  File "/home/ion/anaconda3/envs/artiq-tph/lib/python3.5/site-packages/serial/urlhandler/protocol_socket.py", line 158, in read
    ready, _, _ = select.select([self._socket], [], [], timeout.time_left())
KeyboardInterrupt
```

wizath, 2019-10-09:
The case is that if you spam messages on one socket, the second one will take vast amount of time to react. I'll try to change server behavior to handle all clients equally.

hartytp, 2019-10-09:
Or, is the ethernet polling just being choked by my fuzzing?

wizath, 2019-10-09:
With using your fuzzing script, I've waited 0.5 - 1s for response on second socket, so there is at least some traffic

wizath, 2019-10-09:
I can also create 2 threads for 2 sockets each, that would speed things up

hartytp, 2019-10-09:
> With using your fuzzing script, I've waited 0.5 - 1s for response on second socket, so there is at least some traffic

Okay, I got no traffic on the second socket after a minute.

wizath, 2019-10-10:
Added sequential processing for multiple sockets. E.g if socket 0 got handled last time and there are two waiting (0, 1) the 1 will get priority

hartytp, 2019-10-14:
@wizath slow is fine so long as it doesn't completely choke. If that's implemented can you make sure there is a new firmware release posted somewhere and close this issue please?

hartytp, 2019-10-14:
somewhere = new RC on the firmware GH project

hartytp, 2019-10-15:
@wizath ping...can we close this?

closed 2019-10-15 by wizath