sinara-hw/Booster #321: Creotech v1.3 Booster died

dnadlinger, 2019-10-10:
One of our Creotech v1.3 Boosters (devid 002A00463436510A32323838) suddenly died. All channel error LEDs are on, which power-cycling doesn't change.

```
> logstash
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error
[ERROR] RF Channel not detected at 7, detections 0 1 0
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.252
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
[INFO] network client 10.255.6.191 connected
[INFO] network client 10.255.6.149 connected
[INFO] network client disconnected
[INFO] network client 10.255.6.191 connected
```

(Note the **[ERROR] RF Channel not detected at 7, detections 0 1 0.**)

```
> i2cdetect 7
[i2c_scan] start
[i2c_scan] Found I2C device at 8
[i2c_scan] Found I2C device at D
[i2c_scan] Found I2C device at 2C
[i2c_scan] Found I2C device at 2E
[i2c_scan] Found I2C device at 2F
[i2c_scan] Found I2C device at 48
[i2c_scan] Found I2C device at 49
[i2c_scan] Found I2C device at 4C
[i2c_scan] Found I2C device at 4D
[i2c_scan] Found I2C device at 50
[i2c_scan] Found I2C device at 51
[i2c_scan] Found I2C device at 52
[i2c_scan] Found I2C device at 53
[i2c_scan] Found I2C device at 54
[i2c_scan] Found I2C device at 55
[i2c_scan] Found I2C device at 56
[i2c_scan] Found I2C device at 57
[i2c_scan] Found I2C device at 70
[i2c_scan] end
```

```
> i2cerr
                #0      #1      #2      #3      #4      #5      #6      #7
I2C ERR         2       2       2       2       2       2       3       112
```

```
> status 7
[status] e=0 s=0 r1=100 r2=1 tx=0.000 rf=0.000 curr=0.000 t=0.00 i=0.00 ip=0.00
```

```
> version
RFPA v1.4.1 218aca0, built Sep 16 2019 16:23:32, hv rev
```

Any idea what is going on here? This part of our setup has not been modified in a while, so it is unlikely there were any significant changes to the enviroment/signal levels/etc.

hartytp, 2019-10-10:
Swap it with the v1.4 I was testing and I’ll look at it on Monday (there aren’t really any more tests I can run on that one). Let’s look at the voltage rails and see what died. There isn’t much that could take down the i2c bus so should be quick to find/fix

wizath, 2019-10-10:
First thing - try to take out all modules and check if they are detected one by one

hartytp, 2019-10-10:
@wizath yes, that was my plan (although, this could well be some mechanical issue like before, in which case as soon as I move anything it will go away!).

@dnadlinger on the plus side, this is likely not a major issue; dead FETs are hard to understand and debug and generally a big issue, locked I2C busses are generally easier to debug (e.g. a power rail missing on one channel for some reason).

@dnadlinger if this unit is still in the rack, can you run the `start` VCP command please? IIRC one time this happened because the 5V rail got shorted on one channel...

dnadlinger, 2019-10-10:
@hartytp:
```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 0.00 CURRENT: 0.00
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	0	0	0	0	0	0	0	0
HWID		00:00	00:00	00:00	00:00	00:00	00:00	00:00	00:00
INPWR [V]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00
TXPWR [V]	0.06	0.09	0.06	0.12	0.06	0.07	0.00	0.06
RFLPWR [V]	0.00	0.00	0.00	0.00	0.03	0.00	0.00	0.00
INPWR [dBm]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00
TXPWR [dBm]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00
RFLPWR [dBm]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00
I30V [A]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
I6V0 [A]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
5V0MP [V]	0.000	0.000	0.000	0.000	0.000	0.000	0.000	0.000
ON		0	0	0	0	0	0	0	0
SON		0	0	0	0	0	0	0	0
IINT		0	0	0	0	0	0	0	0
OINT		0	0	0	0	0	0	0	0
SINT		0	0	0	0	0	0	0	0
ADC1		100	144	102	202	100	112	5	100
ADC2		2	1	4	1	46	1	0	1
INTSET [dBm]	0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00
DAC1		0	0	0	0	0	0	0	0
DAC2		0	0	0	0	0	0	0	0
SCALE1		0	0	0	0	0	0	0	0
OFFSET1		0	0	0	0	0	0	0	0
BIASCAL		0	0	0	0	0	0	0	0
HWIS		0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00
HWIO		0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00
LTEMP		0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00
RTEMP		0.00	0.00	0.00	0.00	0.00	0.00	0.00	0.00
==============================================================================
```

hartytp, 2019-10-10:
Fair enough! Not much there. So, it's a question of probing supply lines, then removing modules until it starts working... Shouldn't be too hard to debug (unlike some awful RF issue), I'll look on Monday. Thanks again for the bug report....sigh...

hartytp, 2019-10-14:
removed lid from box and removed lid from one RF channel. `5V_MP` is at 1.6V.

Next step, remove all channels and see what happens...

hartytp, 2019-10-14:
Removing channel 6 seems to resolve the issue

hartytp, 2019-10-14:
Permuted channels 6 and 7 to verify that it is really an issue with channel 6 and not e.g. the main board

hartytp, 2019-10-14:
The 3V3 rail on this channel has developed a 1.5Ohm short to ground

hartytp, 2019-10-14:
It's the `P3V3A` line on P3 that has the ~1Ohm short to GND

hartytp, 2019-10-14:
[image]

So, it's either the power detectors or the OpAmps...

hartytp, 2019-10-14:
With the RF part of the module removed (just plug the sup control board into the Dsub without the metal enclosure) the 3V3 rail comes up to 3V3 correctly.

So, it's a short somewhere on the main board `P3V3A` rail. This only has 4 ICs on it, so should be easy to find.

@gkasprow I'm reluctant to get into probing the RF module myself since it's high density and needs to be treated with care. Can I send it back to you for final debugging and repair?

hartytp, 2019-10-14:
Okay, this seems to be a mechanical issue. When I remove one of the screws holding the RF PCB down to the enclosure, the resistance of the short increases significantly. So, I wonder if it's something like the via tenting being broken through over time.

The screws on this unit are done up very tightly FWIW...

hartytp, 2019-10-14:
Yes, it's definitely mechanical since loosening a few screws fixes the issue. I'll send it back for repair. Not sure what the best solution that won't degrade the thermal performance is. @gkasprow can you have a look and see if there is anything we should change for the next revision...

hartytp, 2019-10-14:
@gkasprow hmm...if there are high-impedance shorts to ground on Booster due to mechanical issues then it's just possible that could explain some of the odd behavior we saw before which we thought was due to "dead FETs". If there is a short in one place, it's very possible there are more that have gone undetected so far, which could explain a lot.

Please can you have a careful look at this module and see if you think there could be any more shorts to the enclosure?

gkasprow, 2019-10-14:
Yes, please send it to Creotech. I will be at another conference (QT Flagship meeting) until the end of this week.

hartytp, 2019-10-14:
Will do. Could also be something silly like a burr in the mechanics that's pressing against the PCB and finally wore through. Anyway, I'll send it to CTI and we'll see what they think...

gkasprow, 2019-10-28:
OK, so there is a short circuit to GND on 3.3V rail on the PA board.

gkasprow, 2019-10-28:
It's not related to the design. When I removed the screws, the short circuit disappeared.

hartytp, 2019-10-28:
Indeed. I think it's mechanical. A burr in the enclosure? Breaking solder mask on a bottom-side via?

gkasprow, 2019-10-28:
Yes, it's a blurr on the edge of the cavity under the SMA connector. It protruded through the solder mask and made a short circuit.

hartytp, 2019-10-28:
Do we need to do anything to the circuit design? Or just ask Creotech to be more careful with their manufacturing?

hartytp, 2019-10-28:
@gkasprow shall we close this?

gkasprow, 2019-10-28:
They will add it to the assembly procedures

hartytp, 2019-10-28:
okay, so do we close now?

gkasprow, 2019-10-28:
some photos documenting the issue:
via
[image]
the cavity edge
[image]

closed 2019-10-28 by gkasprow