sinara-hw/Booster #322: VCP 'powercfg' does not validate arguments

dnadlinger, 2019-10-11:
This had me confused for a while:
```
> powercfg on
[powercfg] set 0
```
Would be great if the command could reject invalid parameters, instead of silently defaulting to 0.

This is on:
```
> version
RFPA v1.4.1 33f0541, built Oct  9 2019 10:58:53, hv rev 4
```

(@hartytp: Remember when I mentioend powercfg didn't work a month or so ago, but then it worked for you first try? Pretty sure I wasn't going crazy, and it was just this.)

hartytp, 2019-10-14:
@wizath can you check that all VCP commands have proper input validation, silently attempting to guess what the user intended with malformed commands is a recipe for disaster...

wizath, 2019-10-14:
In `strtol` docs - `If no valid conversion could be performed, a zero value is returned (0L).`
But anyway, I'll review type checking in VCP to provide more detailed errors

dnadlinger, 2019-10-14:
For standard-compliant strtol, couldn't you just check errno/the pointer output parameter?

wizath, 2019-10-14:
Yep, I'll check pointer output since `errno` warns only about overflows, not parsing errors

hartytp, 2019-10-14:
> For standard-compliant strtol, couldn't you just check errno/the pointer output parameter?

Will you have threading issues there?

hartytp, 2019-10-16:
ping @wizath is this done yet?

closed 2020-02-27 by hartytp