sinara-hw/Booster #323: RF module screw clearance

hartytp, 2019-10-14:
The pan head screws on the creotech boosters protrude outside the pads on the PCB and get extremely close to shorting against nearby components. I think that it's possible for them to short if the PCB isn't located in the centers of the holes.

@gkasprow can we increase the clearances around the screws a little bit to ensure this doesn't create problems? Particularly around the PA power detectors...

gkasprow, 2019-10-14:
This was my suspicion.
I plan to release new RF modules, with CERN libraries and as a single PCB to make assembly easier.

gkasprow, 2019-10-25:
I examined the module under the microscope and did not find anything suspicious. These are not the screws that cause the issue.
[image]
[image]
I have a few more ideas to check.

hartytp, 2019-10-25:
@gkasprow did you have any luck figuring out why the 6V current is so high on this channel?

gkasprow, 2019-10-25:
Not yet, I was not in my lab today. I will do that on Monday. I promise.

hartytp, 2019-10-28:
@gkasprow if you think this isn't a problem feel free to close

closed 2019-10-28 by gkasprow