sinara-hw/Booster #325: periods of excessive current noise on some channels

hartytp, 2019-10-16:
I've occasionally seen periods of increased noise on the bias current measurements on some channels like this: 

[image]

Normally, the noise is << 1mA -- even in that dataset, the noise was only that high for a brief period.

This is the only remotely suspicious behaviour I've seen on the new Boosters and is the only non-software thing I'd like to look into before approving the next production run.

However...I can't currently do that as I can't reliably get data from the VCP/SCPI ports due to watchdog resets etc. Once I can log reliably for extended periods:
- [ ] log a couple of boosters on all channels and look for noisy periods
- [ ] try to figure out if this is also on the bias current or just a measurement error (e.g. hook up external DMM)?
- [ ] does it correlate with more noise/oscillations on the output rf?

May be related to https://github.com/sinara-hw/Booster/issues/311

hartytp, 2019-12-04:
Using a Creotech v1.3 and the latest fw from Creotech I see occasional spikes in the measured currents on one channel.

[image]

[image]

hartytp, 2019-12-04:
Ignore the time when all channels drop out -- I shut the unit down. So, here you can see what working and non-working channels look like! I think we can also see damage to both the input and output stages...

hartytp, 2020-02-18:
@gkasprow are you planning to look at this and https://github.com/sinara-hw/Booster/issues/338 today?

gkasprow, 2020-02-18:
I prepared the setup and started debugging today. I will keep you up to date. It's not that easy to trigger the scope on such glitch!

gkasprow, 2020-02-18:
I plan to modify the code so it toggles IO when the value is much higher than the previous one...

hartytp, 2020-02-18:
Great!

gkasprow, 2020-02-19:
I modified the code, setup the scope with I2C analytic plugins. I added a few lines

```
  i30_temp1 = ads7924_get_channel_voltage(0);
  channels[i].measure.i30 = (i30_temp1  / dev->p30_gain) / dev->p30_current_sense;
  i60_temp1 = ads7924_get_channel_voltage(1);
  channels[i].measure.i60 = (i60_temp1 / dev->p6_gain) / 0.1f;
  channels[i].measure.p5v0mp = ads7924_get_channel_voltage(3) * 2.5f;
	if (i30_temp1 > (i30_temp2+2)) 
        {
		GPIO_SetBits(BOARD_LED1);
		GPIO_ResetBits(BOARD_LED1);
        }
		i30_temp2 = i30_temp1;
```

and so far did not manage to capture any of such events.
When I compare the ADC value with the previous one, the code triggers the scope.

gkasprow, 2020-02-19:
It works without the load currently

hartytp, 2020-02-19:
> It works without the load currently

I saw this both with and without loads, so that shouldn't matter. Also on both v1.3 and v1.4.

```
if (i30_temp1 > (i30_temp2+0.05)) 
```

Are you checking for a 50mA step there? That's larger than I've seen. The max glitch I see on the current is a couple of mA. See https://github.com/sinara-hw/Booster/issues/350#issuecomment-583849667

The distribution of glitches between channels seems to change over time. At any given moment, I often don't see any glitches on some channels. See e.g. https://user-images.githubusercontent.com/21218399/74103892-bcc4d980-4b47-11ea-8a86-4b623f666b2b.png for a Booster where the glitches are infrequent.

Probably the best next step is to see if you can reproduce my measurements: set up a SCIP logger and plot the currents. Do you have code to do that or do you want me to send you some?

hartytp, 2020-02-19:
I guess it's also possible there is a race condition or something so these glitches only appear when the scpi logger is active. Anyway, let's see if you can reproduce my logs first...

gkasprow, 2020-02-19:
I'm checking for 50mV difference at the ADC output.

gkasprow, 2020-02-19:
OK, I will leave it overnight with RF applied and see what comes out of it. Each time the current is exceeded it will print the numbers on the UART and trigger the scope with I2C readout.

hartytp, 2020-02-20:
@gkasprow let's check you're doing exactly the same thing as me:
- please use the latest firmware version CTI sent me `Version(fw_rev='v1.4.1', fw_hash='d0f3d02', fw_build_date=datetime.datetime(2019, 12, 24, 12, 49, 18), device_id='30374705', hw_rev='hw rev 1.4')`
- don't apply RF to any channels
- run the python driver + SCIP logger I emailed you
- use the plotting tool in the logger to plot the data
- you should see some glitches (on some units the glitches are infrequent and come and go, so you may need to leave it overnight).

Once we confirm that in an identical setup to us you see the glitches we can think about how to identify the source.

hartytp, 2020-02-20:
FWIW I don't really understand the code snippet you posted [here](https://github.com/sinara-hw/Booster/issues/325#issuecomment-588345004). The currents are stored as a vector like ` channels[i].measure.i30` but you're only doing your glitch checks with a single number for all channels `i30_temp2 = i30_temp1;` (also the white space there is really confusing but C doesn't care about that). Anyway, I can't see all the code so maybe this is fine, I'm just flagging that I don't understand what you're doing yet.

Anyway, let's see if you can reproduce this using **exactly** the same firmware and code as me.

hartytp, 2020-02-24:
@gkasprow were you able to reproduce this using the scripts I sent you? Also, any update on the refurbished Booster from CTI? That's been outstanding for a while now...

gkasprow, 2020-02-24:
I assigned Bartek to this task and gave him all the info. He was running this script on Friday for a few hours on the Booster produced by TS and so far did not observe any excessive noise. I also asked @wizath to run this script on the Booster which we have at the WUT. We used the dfu file you sent us.

hartytp, 2020-02-24:
Interesting! Did he use the same fw rev and the exact same script (SCPI logger) as me? How was the hw configured?

gkasprow, 2020-02-24:
Yes, he did. You'd be able to see it in my lab tomorrow :)

gkasprow, 2020-02-27:
Bartek is observing similar issues in his HW. I mean that the current is oscillating +/- 3mA.

hartytp, 2020-02-27:
> Bartek is observing similar issues in his HW. I mean that the current is oscillating +/- 3mA.

🎆 Awesome! Now you can reproduce it we can fix it :)

So, the question is now where the noise comes from. Hardware issue or firmware issue?

gkasprow, 2020-02-27:
It seems that we have two issues. One is HW noise, another is probably rounding that generates some extra noise. This is what Bartek found today. I'm just trying to see if I'm able to recreate with my HW

gkasprow, 2020-02-27:
OK, this is what I got after turning the power on
[image]

gkasprow, 2020-02-27:
the funny thing is that the firmware does not trigger the scope...

gkasprow, 2020-02-27:
don't look at the slope, I used channel that I broke before with transients (no TVS)

gkasprow, 2020-02-27:
But at least I managed to recreate the thing that you see. And, indeed, this happens periodically which makes debugging moredifficult

gkasprow, 2020-02-27:
every measurement is taken every 150ms so the entire plot above is 22mins

gkasprow, 2020-02-27:
I think I know why my code did not trigger the scope. The range is 43V/V which gives 580mA range. I was looking for 50mV voltage jumps which correspond to 1.1mV on the sense resistor, which is 11mA.
That's why I didn't see anything. I should look for 2mA jumps which is 0.2mV on the sense register and 8.6mV at the ADC input.

hartytp, 2020-02-27:
I did wonder about hw noise (meaning EMI) but it seems unlikely. Why isn’t it removed by the low pass filters? Why does it normally only affect one channel at a time? Why is it so intermittent?

hartytp, 2020-02-27:
Anyway really glad you can recreate the issue!

gkasprow, 2020-02-27:
I modified the FW and now it triggers the scope.

gkasprow, 2020-02-27:
I'll connect the scope input to the ADC input and trigger with glitch in the data

gkasprow, 2020-02-27:
I set the scope sensitivity to 10mV/div and AC coupling. Let's see what we get

gkasprow, 2020-02-27:
this is the plot showing I2C readout with problematic value. The signal is varying less than 2mV
[image]

gkasprow, 2020-02-27:
the glitches occur periodically, T=455us
[image]
It is a probe pickup. I'm observing the voltage on 100nF capacitor

gkasprow, 2020-02-27:
The glitch oscillation period is 20ns

hartytp, 2020-02-27:
@gkasprow the peak-peak size of the glitch we're looking for is ~4mA (approx +-2mA). If I understand your calculations, that's approx 17mA at the ADC input. The fast glitches you have there are only 8mA pk-pk, so probably too small to be responsible for this.

In any case, there is an RC filter between the mux and ADC, so presumably this should be filtered out?

So, do you think those glitches are relevant to the problem, or just scope pick up that we can ignore? Out of curiosity, is the I2C word consistent with the glitch size?

hartytp, 2020-02-27:
What's going on with the SDA line here?

[image]

hartytp, 2020-02-27:
Aah, that's an ACK

hartytp, 2020-02-27:
I think this might be a race condition in the I2C code.

Look [here](https://github.com/sinara-hw/booster-firmware/blob/b181bbd2f2b7b34caf66365ed9fa945c136c9415/Source/devices/ads7924.c#L107-L121)

```C
void ads7924_init(void)
{
	uint8_t id = 0;
	i2c_read(I2C1, ADS7924_ADDRESS, 0x16, &id);

	ads7924_reset();
	for (int i = 0; i < 84000; i++) {}; // wait for power-up sequence to end

	ads7924_set_mode(MODE_AUTO_SCAN_SLEEP);
	i2c_write(I2C1, ADS7924_ADDRESS, 0x12, 0b11100010);

	// sleep to 10ms per measurement
	i2c_write(I2C1, ADS7924_ADDRESS, 0x13, 0b00000010);
	i2c_write(I2C1, ADS7924_ADDRESS, 0x14, 0b00011111);
}
```

The ADC is set to convert once per 10ms. With the ethernet the CPU seems to get quite busy and I suspect this delay can be exceeded. In that case, is it possible that a conversion occurs while we are reading out the registers, leading to an error?

As a quick test, can we switch the ADC to single measurement mode. Then put conversion (with suitable delay) in the ADC get voltage code. That way we are guaranteed that there will not be an issue?

gkasprow, 2020-02-27:
I was thinking that this could be an issue of acquisition time violation. It could be easily the effect of reading ADC during conversion. I plan to look at it closer tomorrow morning. I have a few more ideas to check. I did not observe the effects of CPU load on the I2C communication.
The things to check:
- route DONE line to INT signal and observe it with scope. I believe we find some correlation.
- switch the ADC to on-request conversion mode. I2C communication during acquisition may cause errors due to ground bouncing. We can fire all channels ADCs once, then read the results.
- connect the scope probe to GND to check if the glitches disappear. Anyway I think they are induced by some external source
- make sure we do not violate acquisition time by too high RC filter time constant
- measure ADC supply voltage during conversion
- replace the ADC value with the fixed constant number and check if there are some issues with SCPI

hartytp, 2020-02-27:
@gkasprow good ideas! I'm pretty sure one of those will give the answer.

> make sure we do not violate acquisition time by too high RC filter time constant

I've been thinking about this. The acquisition time is currently at max (~50us). The RC time constant is 100R * 220pF, which seems fine.

In any case, all the ADC inputs are roughly static and the duty cycle is fixed (automatic sampling) so I think any effects due to the RF filter should be constant and not cause glitches (although, we should still double check to make sure the ADC doesn't become in accurate/have high cross-talk between the measurements).

hartytp, 2020-02-27:
Anyway, thanks for digging into this!

If the new Booster with TVS don't break and we can fix this issue, I think we can say that the hardware is fine.

Looking over the code, it's a bit of a mess at the moment. So, I think we'll need a long term plan for it, but we can sort that once we're confident the HW works...

gkasprow, 2020-02-28:
I routed the BUSY signal to the INT pin. The free run conversion period is 8.7ms.
And indeed, the faulty measurements occur during I2C transmission
[image]

gkasprow, 2020-02-28:
Bartek modified the firmware that it triggers the acquisition every time on request and it seems it helped
[image]

gkasprow, 2020-02-28:
[image]

gkasprow, 2020-02-28:
After opening the window
[image]

hartytp, 2020-02-28:
Cool! That looks really good.

Could you ask your student to merge this change with the Creotech changes and upload to the GitHub please? Can you also post a firmware binary as an RC on the firmware git hub? Thanks!

hartytp, 2020-02-28:
Once that's done please close this issue

gkasprow, 2020-02-28:
Bartek is from Creotech :)

hartytp, 2020-02-28:
aah, sorry!

hartytp, 2020-02-28:
PS thanks Bartek!!!

hartytp, 2020-03-02:
Fixed by Bartek at Creotech (but please do push the changes to git hub)

closed 2020-03-02 by hartytp