sinara-hw/Booster #333: Capacitor ratings

hartytp, 2019-11-29:
@gkasprow can you do me a favour and please double check the ratings of all capacitors used on Booster.

At least one issue is that C6 on the power amp board (100nF 28V decoupling capacitor) is only rated to 16V.

hartytp, 2019-11-29:
PA stage schematic looks OK apart from that one cap. Haven't checked main board yet.

hartytp, 2019-11-29:
What's the voltage on main board U1 compensation/shut-down pin? Does C77 need to be 50V as well?

hartytp, 2019-11-29:
C87 is 35V which feels a little marginal for a 30V rail

hartytp, 2019-11-29:
Checked the main board and didn't find any other critical issues (all caps other caps are rated to at least 16V and the highest supply is 12V).

gkasprow, 2019-11-29:
OK, I replaced all parts in RF boards with CERN symbols so could make some error.

hartytp, 2019-12-19:
- [ ] To do: double check all matching capacitors (particularly on the output stage) and make sure they can withstand the peak surges during glitches/no load operation. Should be fine but worth double checking

gkasprow, 2020-02-27:
fixed, but Bartek is doing a detailed design review right now.

hartytp, 2020-02-27:
> fixed, but Bartek is doing a detailed design review right now.

Awesome! Thanks so much!

That's really great news. Can you close this once you're happy? Thanks!

closed 2022-02-07 by gkasprow