sinara-hw/Booster #334: Booster doesn't boot without all channels present

dnadlinger, 2019-11-29:
With channel 7 removed, all sorts of LEDs light up on the front panel (@hartytp to fill in details):

```
> version
RFPA v1.4.1 ade44e9, built Oct 15 2019 16:28:01, hv rev 4
> logstash
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_DIRECION_TIMEOUT error ch 7
> i2cerr
                #0      #1      #2      #3      #4      #5      #6      #7
I2C ERR         0       0       0       0       0       0       0       41557
```

hartytp, 2019-11-29:
@wizath please can you look at this asap! This is preventing us from using a Booster with one channel missing and we don't have spare hw any more

hartytp, 2019-11-29:
[IMG_1012.zip](https://github.com/sinara-hw/Booster/files/3905835/IMG_1012.zip)

This is what happens at startup

gkasprow, 2019-11-30:
I observe the same behavior. It does not work with one channel removed. I thought it was a feature, not an issue :)

hartytp, 2019-11-30:
Definitely an issue as t means that a single channel failing (still unfortunately common) makes an entire booster unusable.

Also we agreed that this should not happen and it did not used to, so I think it’s regression in the recent firmware builds.

@wizath

hartytp, 2019-12-04:
Fixed by Creotech. Thanks!

closed 2019-12-04 by hartytp