sinara-hw/Booster #335: scrap PHA-1

hartytp, 2019-12-03:
Not sure if this is worth the effort, but...

Booster's gain is actually a bit too high. In hindsight 30dB is probably plenty (maybe 35dB abs max). Using with Urukul we end up having to run with ~20dB attenuation for a 1W output. Internally booster already has a 5dB attenuator to reduce the gain. That means we end up attenuating by ~25dB which is silly from a noise POV.

The PHA-1+ has something like 16dB of gain. If we scrap the internal 5dB and 6dB attenuators and go for a gain of 35dB then we could scrap it completely. This would simply the design and reduce cost and power consumption.

The matching networks might need tweaking, but I guess those are tuned at production time anyway so may not be an issue?

Maybe it's not worth the effort, but I thought I'd mention it, as it would make a somewhat better final product. What do you think @gkasprow

hartytp, 2019-12-03:
P1dB of the ADL is a couple of dB lower than the PHA-1+. The schematic indicates that we need something like 23dBm to saturate the output stage. So it may be that the ADL isn't quite powerful enough to do the job if we move to a single amplifier.

Anyway, at a high-level the point is that we may be able to improve the design by moving to a suitable single-stage amplifier if we can find something with the right gain and P1dB.

gkasprow, 2019-12-03:
We use strong feedback to limit the gain of the amplifiers. In this way, we achieve quite good matching between channels. No gain tuning is done in production. If we go for max gain, then tuning would be required. So, I'd further increase the feedback gain and improve gain matching.

hartytp, 2019-12-03:
I don't see any feedback on the ADL, only the PHA-1+. Why does that amp need feedback to stabilise the gain and not the ADL?

hartytp, 2019-12-03:
At the very least, we should probably go for the ADL5535, which has 4dB less gain, but is otherwise basically the same as the 36 AFAICT

gkasprow, 2019-12-03:
the amplifier gains is G0 /1-G0B, where B is feedback gain and G0 is open loop gain.
PHA1 gain is 14 do 19dB and drops quickly with frequency. ADL is much more stable.

hartytp, 2019-12-03:
hmm..according to the schematic, the PHA-1+ is used in a G=16 configuration.

If we scrap the ADL5536 completely then we loose ~20dB of gain. If we scrap the two attenuators then we win 11dB of gain. That would put the overall system gain to 29dB, which is probably a bit too low. e.g. phaser doesn't have the pre-amp and would struggle to get a 1W output with Booster (would require a few dBm after filters, attenuators, etc which seems a bit high).

gkasprow, 2019-12-03:
we can also add feedback to ADL to further improve gain matching.

hartytp, 2019-12-03:
OK. So your summary is that it's going to be hard to find a pre-amp with high enough P1dB and high enough gain to do the job in a single-stage -- particularly if we use feedback to stabilise the gain. So, it's worth keeping the two stages.

Is that correct?

gkasprow, 2019-12-03:
yes.

hartytp, 2019-12-03:
@gkasprow what output power do you expect form phaser in BB mode?

closed 2019-12-03 by hartytp

gkasprow, 2019-12-03:
I wouldn't touch something that works. Let's find and solve issues first.

hartytp, 2019-12-03:
FWIW it's always annoyed me that Urukul has those pre-amps. If we scrapped them then the current Booster design would be about perfectly matched to Urukul.

@jordens what's your view on them? Would you be in favour of DNPing them by default in a future revision?

gkasprow, 2019-12-03:
Phaser will have roughly -5dBm

gkasprow, 2019-12-03:
with attenuator set to 0

dnadlinger, 2019-12-03:
Agreed with @gkasprow on not changing a(n almost) running system. If we ever end up changing the channel design, though, a somewhat lower gain would indeed be better matched to our use case.

To illustrate, here are the gain settings for some Urukul channels connected to Boosters (and then AOMs) in our lab:

24.0, 23.5, 20.5, 23.0, 22.5, 23.0, 23.0, 26.5, 17.0, 20.0, 22.5, 22.0, 22.5, 23.5, 19.0, 22.0, 22.5, 23.5, 21.0, 21.0, 25.5, 25.5, 25.5, 25.5, 20.0, 23.0, 21.0, 20.0, 14.5, 18.0, 17.0, 22.0, 21.0

hartytp, 2019-12-03:
Okay, worth keeping the gain at ~40dB then. Just a shame that Urukul has so much more gain than phaser as it means we have to crank the attenuators up to cancel out the pre-amp and needlessly burn off a decent amount of power.

hartytp, 2019-12-03:
@dnadlinger yes, but then do we add a pre-amp to Phaser?

Probably the real answer here is to add a jumper to allow the pre-amp on Urukul to be bypassed and then save power at that end. That would keep the output levels matched between urukul and phaser...

gkasprow, 2019-12-03:
you can always ask for Urukul variant without PAs

dnadlinger, 2019-12-03:
@hartytp: A population option (or easy move-two-0Ω-jumpers tweak) would indeed sounds sensible. It's really not something I'm worried about at this point, though.

dnadlinger, 2019-12-03:
(NB: We'd then need to decrease the attenuator on Urukul as well, since the ERA-3SM+ provides ~23 dB gain.)

gkasprow, 2019-12-03:
one 0805 resistor fits nicely instead of ERA-3SM+ amplifier.

hartytp, 2019-12-03:
@dnadlinger from the numbers you posted, it looks like we do need *a* pre-amp on Urukul. It's just that the 23dB of gain from the ERA-3SM+ is way too much for us.

Maybe the better proposal here is to go for a pre-amp with ~10dB of gain. That would put us in a more comfortable regime. I suspect we can find a pin-compatible one with lower gain/P1dB and lower current consumption.

dnadlinger, 2019-12-03:
@hartytp: MAR-1SM+ might work at first glance. Didn't look very carefully, though.

hartytp, 2019-12-03:
hmm...Urukul currently has ~14dB of gain due to the attenuator/amp combo, but with a fairly poor tolerance. So, scrapping the amplifier completely looks like it wouldn't give us much headroom at all.

However, going for an amplifier/attenuator combo with ~10dB lower gain would probably be ideal.

dnadlinger, 2019-12-03:
Apologies, ignore the 14.5 in (the previous version of) the above list; that channel is apparently unused.

hartytp, 2019-12-03:
Okay, so reducing Urukul's gain by 10-13dB would probably be ideal for us.

hartytp, 2019-12-03:
I *think* that Phaser will be fine without a pre-amp, but we may want to reconsider that for the next design revision.

@gkasprow Is it worth leaving a little space in the layout in case we add one later?

gkasprow, 2019-12-03:
we have space for SOT-89 amplifier (like ADL5534) supplied from 5V. We can add it easily without moving other components.
[image]