sinara-hw/Booster #336: input attenuator

hartytp, 2019-12-03:
The input attenuator needlessly decreases the amplifier NF.

It's worth considering whether we want to change it.
- move to after the first pre-amp to provide a better resistive load?
- add feedback to the first pre-amp to stabilise the gain?
- increase the gain feedback in other stages?

@gkasprow out of curiosity, why do you implement these attenuators with resistors rather than attenuators? What's the cost difference?

gkasprow, 2019-12-03:
Feedback lowers the input impedance and makes matching worse. I would swap the second attenuator with the filter or add a limiting diode. The second is simpler and can me added to existing modules easily.
Resistors were used to, later on, tune the gain which eventually was not necessary. Cost is not an issue here. It was designed by my colleague. We can replace with fixed attenuators but for these frequencies I do not expect any difference.

hartytp, 2019-12-03:
Ok, sounds like a good plan.

closed 2020-03-02 by gkasprow