sinara-hw/Booster #337: out bias isolation resistor

hartytp, 2019-12-04:
The OpAmp driving the FET bias drives 1uF of capacitance with only a 4R7 isolation resistor. Is that enough to ensure stability?

We should definitely measure the FT of this to check it's not massively over/under shooting...

Is there nay reason not to swap R38 for 30-50Ohm in the next revision?

hartytp, 2019-12-04:
AFAICT this can't actually damage the FET since D6 prevents the bias from ever going positive, and the OpAmp only runs from a -5V supply. So, we always stay within the abs max ratings for the gate voltage. I guess it could cause issues with current surges when the unit is powered on, but the fold-back limiting on the 29V LDO should prevent damage.

So, worth fixing but not critical I think.

hartytp, 2019-12-04:
I wonder if this could explain the intermittent periods of high noise on the bias current. I never looked at the bias voltage on a scope, so any AC noise would be easy to miss...

gkasprow, 2020-03-02:
let's use R38= 33R

closed 2020-03-02 by gkasprow