sinara-hw/Booster #338: temperature readings more noisy when all channels disabled

hartytp, 2019-12-04:
Look at this screen shot

[image]

This was taken using a Creotech v1.3 with the latest firmware including Creotech's I2C fixes.

You can see that the temperature diagnostics are more noisy when the channels are disabled. Why is that?

hartytp, 2019-12-04:
I observe this in the logs with both the VCP and SCPI loggers. It is very reproducible.

hartytp, 2019-12-04:
@gkasprow while you're investigating the RF issues can you quickly stick a scope on the 3V3 supply for the thermometer and check that there isn't a DC level shift/excess noise when the unit is in stand by mode?

hartytp, 2019-12-04:
Here is the raw data. 

[image]

You can see the excess noise come and go when the channels are enabled/disabled

hartytp, 2019-12-04:
Here is the raw data. 

[image]

This channel bounces between values separated by over 1C!

You can see the excess noise come and go when the channels are enabled/disabled

hartytp, 2019-12-04:
This channel bounces around by over 1C!

[image]

What's interesting is that it looks bistable: one of the readings is the reading it settles on.

hartytp, 2019-12-04:
Here is another channel 
[image]

hartytp, 2019-12-04:
Notice also the huge hysteresis.

[image]

When the channels are disabled the temperature reading instantly jumps up and becomes noisy. When the channels are re-enabled the noise goes but the temperature reading remains at the high value. This seems non-physical...

hartytp, 2019-12-04:
6V currents showing when Booster was enabled/disabled (forgive the mislabelled y-axis)

[image]

Temperature on ch6

[image]

Something odd is definitely going on...

gkasprow, 2019-12-04:
I will look at that tomorrow.

hartytp, 2019-12-05:
@gkasprow while you're waiting for more amps can you have a look at this please?

hartytp, 2019-12-29:
Creotech suggest that the problem here is that the temperature is not updated at all when the channels are enabled.

@wizath do you have any idea why this is happening?

hartytp, 2019-12-29:
related to #322?

hartytp, 2020-01-02:
@wizath do you think you can have a look at this one?

hartytp, 2020-02-13:
@gkasprow @wizath any idea what the cause of this is?

hartytp, 2020-02-27:
I don't think there is any mystery to this one..

Look in the code base here https://github.com/sinara-hw/booster-firmware/blob/ade44e9ae6dd3dd36f435e10f69d60876c4fdddd/Source/board/channels.c#L529-L530

The temperature measurement functions are only called when the channels are disabled. So, it's a simple oversight.

@gkasprow while you/your students are working on the code, could you push a fix to this please (just copy those two lines in to the if statement that deals with the channels being activated).

hartytp, 2020-02-27:
@gkasprow NB the temperature measurements also free-run and read during acquisition. That's also liable to result in increased noise/glitches.

I'd expect noise at the ~1LSB level, not these 1C glitches. So, most likely this is another case of a silly FW bug (or a timing violation)...

hartytp, 2020-02-27:
We should also check that there is a suitable filter capacitor on the ext diode.

hartytp, 2020-02-27:
My C is a bit rusty, but I'm pretty sure the logic here is not right:
```C
	if (temp_ext & 0x40) total += 0.250;
	if (temp_ext & 0x80) total += 0.500;
	if (temp_ext & 0xC0) total += 0.750;
```

e.g. `0XC0 & 0x40=0x40` which evaluates to `true`. Why is this not the obvious conversion of
`temp = ((float)temp_h) + 0.25*((float)temp_l >> 6` (needs fixing for real C syntax)

hartytp, 2020-02-27:
i.e. with the current code 0xC0 (0.75C) gives 1.5C. So, a 0.25C change from 0.75C to 1C would cause the temperature to decrease by 0.5C. Similarly, a change from 0C to 0.25C would change the calculated temperature from 0C to 1C. That likely explains the crazy statistics I saw...

@gkasprow could you ask your students to fix these simple bugs and merge that with the latest Creotech codebase? (Above all, please can you ask Creotech to **push their changes to git hub** so I actually know what version of the code we're working on!!!)

hartytp, 2020-02-28:
Summarising the work needed:
- [ ] fix the temperature conversion algorithm
- [ ] actually measure the temperature when the channels are enabled
- [ ] decide whether/how to synchronise the readout with the acquisition to avoid glitches
- [ ] push the changes to git hub and create a new RC binary release

hartytp, 2020-03-02:
I believe this is now fixed. I'll re-open if the problem recurs.

closed 2020-03-02 by hartytp