sinara-hw/Booster #339: Testing check list

hartytp, 2019-12-04:
- [ ] can the RF switch open before the pre-amp and power-amp are fully biassed (e.g. on power on/off, standby/enable, interlock trip/reset)? If so, could this cause damage with inputs applied?
- [ ] is the in bias stable or does the OpAmp ring due to capacitive load. I guess the best thing is to look at the step response
- [ ] what do the 30V and 6V current transients look like at standby/enable and interlock trip/reset. Is there an issue here?
- [ ] are there voltage spikes at any of the three amplifier inputs/outputs during turn on/turn off, standby/enable, interlock trip/reset, Urukul swithces. Could these cause damage?

hartytp, 2019-12-04:
I think these tests are already being carried out, but I wanted to have a single place to track which ones have been performed.

@gkasprow are there any other tests you think we should perform?

gkasprow, 2019-12-05:
bias step response
[image]
noise is below the scope capability

hartytp, 2019-12-05:
that looks nice! no issue there.

gkasprow, 2019-12-05:
no spikes during power on or power off. the supply rails rise time is too slow to pass via amplification chain.
[image]

hartytp, 2019-12-05:
are you happy with the sequencing there?

hartytp, 2019-12-05:
okay, so the conclusion is that you've checked all the transitions you can think of and the only issue seems to be the large voltage transients caused by the switch?

closed 2019-12-05 by hartytp

gkasprow, 2019-12-05:
Yes, I didn't find anything suspicious here.

hartytp, 2019-12-05:
Okay. I don't have any other suggestions for things to try. Let's refurbish all channels Booster with new amps (all three stages) and see if the TVSs stop it dying...

gkasprow, 2019-12-05:
I ordered such TVS
https://pl.mouser.com/datasheet/2/240/Littelfuse_tvs_diode_array_AQ4021_datasheet.pdf-1372549.pdf
https://pl.mouser.com/datasheet/2/240/Littelfuse_tvs_diode_array_AQ4020_datasheet.pdf-1372610.pdf
https://pl.mouser.com/datasheet/2/916/PESD6V5C1USF-1599731.pdf
https://pl.mouser.com/datasheet/2/308/NUP1301ML3T1-D-1293028.pdf
It would be very easy to add them to existing design