sinara-hw/Booster #341: fan exhaust to back

jbqubit, 2019-12-18:
Current airflow pattern on my Booster is back-to-front. Any reason why fans can't be reversed to support front-to-back airflow? I propose that this is supported as a build option.

dnadlinger, 2019-12-18:
Front to back would definitely be more expected in rack-mount hardware. I think our rack might even have a mix right now, which of course isn't ideal. (Then again, the fans aren't ever above idle on our loads.)

gkasprow, 2019-12-18:
They can be assembled in either way.

hartytp, 2019-12-19:
Add a schematic annotation?

Will help when CTI post the assembly instructions as agreed and we can make sure this kind of detail is present there.

gkasprow, 2019-12-19:
I think that the fan direction should be specified during ordering.

gkasprow, 2022-02-07:
added schematic annotation.

closed 2022-02-07 by gkasprow