sinara-hw/Booster #342: Testing: symptoms of damage

hartytp, 2019-12-18:
Previously we've seen the following symptoms of damage in Booster:
- **Hysteresis** when the amplifier is power-cycled the PA bias current does not return to the same value. Standby/enable does not exhibit this hysteresis, but that's a little different since the gate voltage does not go to 0V during standby. c.f. https://github.com/sinara-hw/Booster/issues/311 
- **PA current fluctuations over time**
- **Pre-amp supply current fluctuations over time**
- **Gain changes**

c.f. https://github.com/sinara-hw/Booster/issues/332

The plan is to get a set of Boosters repaired and have the TVS diodes fitted. We'll then stress test them over a period of a few weeks while carefully monitoring for all of these symptoms. Hopefully we can also fix the SCPI issues, which will make gathering diagnostics easier.

gkasprow, 2019-12-19:
@hartytp my colleague told me that he found possible place where SCPI crashes. He is fixing it right now.

hartytp, 2019-12-19:
Whoo! What was the issue?

gkasprow, 2019-12-19:
The code references external libraries that dynamically allocate memory and overwrite the stack. So it's tricky because it's not easy to debug someones code.

hartytp, 2019-12-19:
Indeed. Okay, great. Can you ask them to push the fixes to GH when they get a chance?

Thanks so much to all for working on this! If the HW and SW now work reliably for this I will be **very** happy :)

hartytp, 2019-12-19:
BTW on the testing front, my other observation was this:

My randomized hw testing does things like enable/disable channels, trip interlocks, measure powers/gains etc while monitoring all currents. It has assertions and errors out if the currents don't lie in a very tight range.

Last time I ran it on a new Booster I found that initially the current ranges could be set tight (<mA) without issue. However after a while the tests started failing due to mA changes in the currents. I suspect that was channels dying due to switch transients during standby/enable.

Anyway, if Booster can survive a few hundred thousand randomized tests while still maintaining tight tolerance on the currents then it will be a very good sign.

hartytp, 2019-12-19:
@gkasprow CTI currently have two of our Boosters and you have one.

Do CTI need more than one Booster? If not, could you ask them to patch it and return it to us please?

For any CTI Booster v1.3 that get patched could you also ask them to modify the interlock so the hysteresis is AC only (not DC) as done in v1.4, Thanks!

gkasprow, 2019-12-19:
I'm just testing performance and noticed that TVS insertion limits the max power. I managed to get barely 32dBm out of it. The gain is 36dB. I'm working to find what is limiting the gain.

hartytp, 2019-12-19:
oops...okay. Let's fix that and then check that we still can't break any channels with the square waves.

Can you post an updated schematic when you're done showing how the TVSs will be connected?

gkasprow, 2019-12-19:
the gate TVS does not change neither gain nor the max power.

gkasprow, 2019-12-19:
The first stage TVS is limiting the max power by 0.6dB
The input TVS is limiting the max power by 0.1dB

gkasprow, 2019-12-19:
The gain is reduced by 0.2dB after insertion of 2 TVS

gkasprow, 2019-12-19:
The output TVS reduces the low signal gain by another 0.1dB. Max power is the same.
My previous doubt was caused by the wrong interlock setting.

hartytp, 2019-12-19:
The gain changes don't worry me too much since I guess we can compensate by tweaking the other matching capacitances.

What do the TVSs do to the circuit linearity? Do they add much distortion?

hartytp, 2019-12-19:
Anyway, if you're happy that the new circuit protects the amps without degrading the performance then please close the issue.

gkasprow, 2019-12-19:
I didn't notice any distortion. IMHO it's their capacitance that matters.

closed 2019-12-19 by gkasprow

hartytp, 2019-12-19:
Great. I was just worried about it acting like a small varicap, leading to distortion.