sinara-hw/Booster #343: Test check list

hartytp, 2019-12-19:
@gkasprow assuming you're happy that we have understood the issues that caused some channels to die, there are a few final things we need to do before finishing this round of testing:
- [ ] You had one channel where the 3V3 rail was shorted. We should figure out why. Let's remove it from the enclosure and see if the short still persists. If it goes then the short was a mechanical issue which we need to look at. Otherwise it's an issue with the PCB or a component failure
- [ ] https://github.com/sinara-hw/Booster/issues/338 Could be a software issue, but we should sanity check the HW (do the I2C lines still look good when the unit is in standby? Is there excessive noise on the supplies etc when the unit is in standby?)
- [ ] Fix all channels one one unit and then log the 29V bias currents over night. Do you see periods of excessive noise (cf https://github.com/sinara-hw/Booster/issues/325)? If so, let's see if this is a software issue or a hw issue by measuring the output of the inamp carefully (we're looking for mV changes so needs careful probing). If not, let's close that issue and assume it was related to the amplifier death.

Once those tests are done, can you send us a unit with all channels fixed (all three amplifier ICs changed) and TVS diodes added so we can do our own testing.

Thanks!!!

hartytp, 2019-12-19:
So...just these on the HW test list.

gkasprow, 2019-12-19:
I found that 3V3 was shorted during PCB production. It was not an amp I got from you. There were 2 amps with the same number. 
The university is closing for Xmass so I took all the stuff and will deliver to TS tomorrow morning. They will patch all remaining boards with new amps and TVS. I run out of ICs so they need to order them. I made photos of patched board and noted TVS models.
I patched and tested two channels, one from CTI, second from TS and will ship to you tomorrow.
In the meantime TS will add TVS and prepare complete Booster for you. We will test it and ship back to you.

hartytp, 2019-12-19:
> I found that 3V3 was shorted during PCB production

hartytp, 2019-12-19:
> I found that 3V3 was shorted during PCB production

I see. So you mean:
- this fault occurred on a unit that had never been tested/used
- it was present on the PCB before assembly into the RF enclosure
- it's a production fault so either a problem with the PCB or a short during solder

In other words, other than giving it back to TS for them to check their internal QC, you're not worried about it.

Is that all correct?

hartytp, 2019-12-19:
> The university is closing for Xmass so I took all the stuff and will deliver to TS tomorrow morning. They will patch all remaining boards with new amps and TVS. I run out of ICs so they need to order them. I made photos of patched board and noted TVS models.
I patched and tested two channels, one from CTI, second from TS and will ship to you tomorrow.
In the meantime TS will add TVS and prepare complete Booster for you. We will test it and ship back to you.

Great! I think the easiest thing is if you don't ship me individual channels but ship me a complete Booster when you can.

Currently you have one of my Boosters and CTI have two. I guess CTI only need one for software development so you should be able to ship me two back now? As I said, if you can ask them to also fix the hysteresis before sending back that would be amazing!

hartytp, 2019-12-19:
Last question: can you have a quick look at the bias current noise on a new amp (i.e. check if this is a sw issue or a hw issue) and the issue with the temperature not reading correctly when the unit is in standby mode.

hartytp, 2019-12-20:
@gkasprow what's the plan with these final bits of testing? Can you have a quick look at them after Christmas before returning the unit to us to check whether there is anything suspicious in the hw?

gkasprow, 2019-12-20:
I'm leaving for longer holidays outside of Warsaw, but Creotech and Technosystem will work after Xmass. I will be back on 7.01.2020. In the meantime, I will be available online but without access to the lab.

hartytp, 2019-12-20:
No prob! Have a good holiday

closed 2020-01-22 by hartytp