sinara-hw/Booster #344: Add TVSs

hartytp, 2019-12-19:
See https://github.com/sinara-hw/Booster/issues/329

hartytp, 2019-12-19:
@gkasprow can you draw out what the new circuit with TVS should look like so we have a record of what patches are being implemented?

gkasprow, 2019-12-19:
The addition of the TVS is very simple. One does not have to remove the board from the enclosure nor disconnect the RF cable.
The input TVS consists of two diodes in one SOT-23 package. Leads 1 and 2 must be shorted to GND. Lead 3 goes to the RF track.
Alternatively lead 3 goes to GND and leads 1,2 go to the RF track.
One way - simpler but less elegant
[image]
The other way - more complex because one must scratch the solder resist
[image]

gkasprow, 2019-12-19:
The first stage output TVS can be added quickly.
It is 6V8 bidir TVS [AQ4021-01FTG-C ](https://pl.mouser.com/datasheet/2/240/littelfuse_tvs_diode_array_aqxxc-01ftg_datasheet.p-1531773.pdf) Bidir ones have lower capacitance because there are 2 diodes connected in series. In our case, we don't care if it is uni-dir or bi-dir.

[image]

gkasprow, 2019-12-19:
The MOSFET needs the TVS at the gate as well. I used AQ4020-01FTG which is easy to solder and has 2.5pF capacitance and 3.5V breakdown voltage. What is important, it must be uni-dir TVS because we don't want a positive voltage on the gate because just 3V can kill the transistor. One could also use here and at the output of the first stage the same 6V uni-dir TVS. The cathode goes to GND!

One can add it in less elegant way
[image]

or in more elegant way, that needs scratching of the solder resist.
[image]

gkasprow, 2019-12-19:
The TVS protecting the output must protect it against too high voltages that can occur during operation without the load. The transistor sustains up to 100V of the drain voltage so we need some clipping between 45 and 90V. I could not find easy to obtain low C TVS so I used two standard 0.5W BZT55-series 33V Zener diodes (i.e. BZT55C33) in series with low capacitance TVS diode. Not very elegant but works.

[image]
[image]

DO NOT CONNECT the pad 2

hartytp, 2019-12-19:
Nice! Hopefully that'll be the end of the odd behaviour in Booster!

gkasprow, 2019-12-19:
With these components, it's very hard to exceed ratings of the amplifier.
Another aspect is the soldering of the power FET thermal pad which was an issue in 3 cases.

hartytp, 2019-12-19:
> Another aspect is the soldering of the power FET thermal pad which was an issue in 3 cases.

What's the plan for that? Better QC?

gkasprow, 2019-12-19:
We can activate the FET thermal pad surface prior to soldering.

gkasprow, 2019-12-19:
Schematic changes:
Input TVS
[image]
1-st stage TVS
[image]
Gate TVS
[image]
Output TVS
[image]

hartytp, 2019-12-19:
Shouldn't the input TVS be placed after the switch to protect the first stage pre-amp from the glitches generated by the internal switch?

Or are you trying to protect the switch itself?

hartytp, 2019-12-19:
Why not use a bidir TVS on the FET (don't you also want this to protect the second stage output)?

Do you need a TVS on the second stage input (after the AC coupling cap)?

gkasprow, 2019-12-19:
The switch glitches are smaller than 0.7V so TVS won't remove them.
FET has a safe operating range from +3.-10. TVS has a diode in parallel. So it protects the second stage output as well, clipping transients to 0.7V

hartytp, 2019-12-20:
> The switch glitches are smaller than 0.7V so TVS won't remove them.

Okay, I remember now -- the point is that the input glitch is small but it saturates the amp quickly and produces a large glitch at the output.

hartytp, 2019-12-20:
> So it protects the second stage output as well, clipping transients to 0.7V

Okay, I'd need to think about this. Aren't there DC block capacitors there that can charge up so a TVS on the gate won't necessarily protect the output of the 2nd stage.

gkasprow, 2019-12-20:
The output of the second stage is loaded with 680p and then TVS diode to GND. Moreover, it has a feedback loop that reduces the output signal rise time. So the output spike has to charge both C50 and C44 using L5 energy. The first stage has only 39pF of C40.

gkasprow, 2020-02-27:
added in v1.5

closed 2020-02-27 by gkasprow