sinara-hw/Booster #346: VCP echo

hartytp, 2020-01-03:
I'm writing some code to automate testing using the VCP.

One thing that's made this a bit annoying to do is that the VCP always echos commands back. Can we disable that please? IMHO echoing is something that should be left to the terminal client, not something that's done by default by the device.

To make matters worse, there is a minor inconsistency in the response. It seems that the first time it echos it directly echos the command with a trailing `\r\n`, in subsequent responses it prepends a `> ` to the reply.

These kinds of thing can be worked around, but they make it more annoying to interface with the ports using automated tooling, so should be avoided.

@wizath can you just remove the echo?

hartytp, 2020-01-03:
Also, FWIW, the responses from the VCP driver don't make parsing easy. Some are fixed number of lines, some are terminated in '>\n\r' etc. It would be much easier to work with if all lines had a fixed end of response.

wizath, 2020-01-03:
If you really want to automate things, use commands
`conf remote 1` or `conf remote 0` to disable / enable echoing

hartytp, 2020-01-03:
Okay, good to know. Is the `conf` command documented anywhere? I didn't see it on the [wiki](https://github.com/sinara-hw/Booster/wiki/VCP-interface)

hartytp, 2020-01-03:
Is that setting stored in non-volatile memory? If so that solves my problem entirely, so thanks!

wizath, 2020-01-03:
Nope it's not

hartytp, 2020-01-03:
(not documented? not stored in non-volatile memory? or neither?)

closed 2020-02-27 by hartytp