sinara-hw/Booster #348: Amp death

hartytp, 2020-01-22:
Consolidating various issues that we believe deal with different symptoms from the same issue...

The first pre-amp and power amp in Booster have been dying. After investigation, we believe this is due to voltage transients produced by the switch causing breakdown at the amp inputs/outputs. The proposed solution is to scatter some TVS diodes around to absorb the glitches.

Symptoms of amp death include:
- Hysteresis in the 29V current between power cycles (NB putting channels into standby doesn't reset the bias voltage to 0; it keeps it at pinch off)
- Fluctuations in the 29V bias current
- Fluctuations in the gain

We will receive 24 channels of patched Boosters (with all amp chips replaced) in the coming week. Plan for verifying that the TVS does indeed fix these issues:
1. Power cycle repeatedly, recording bias currents to look for hysteresis
2. log currents and output power for ~24 hours to establish a baseline
3. stress test a Booster using our randomized test suite (trips interlock repeatedly, turns unit on/off etc)
4. repeat (1) and (2)
5. if all looks good so far install in our experiments and use for a few weeks while logging to check for issues

This would all be a lot easier if the SCPI issues were sorted out...

jordens, 2020-02-03:
Are the reworked channels back? First impressions?

hartytp, 2020-02-03:
They were due to arrive from TS today after a few delays at their end. I haven't seen them yet. Expect first impressions in the next couple of days. Will be easier to thoroughly stress test once CTI finish fw development.

closed 2020-02-10 by hartytp