sinara-hw/Booster #349: Interlock sensitive to modulating RF power

connorgoham, 2020-01-28:
Switching on RF power to the amp triggers the forward power interlock, even for power levels/interlock settings where the amp remains active in the steady state. For example setting the interlock to its maximum value and producing a forward power of ~36 dBm at 200 MHz to an AOM, the interlock triggers whenever the RF is switched on. Once the RF is on, if the interlock reset is pressed, the amp operates without a problem until the input RF power turns off and on again, upon which the interlock is tripped. Perhaps this is the result of transient voltage spikes?

Hardware: v1.3

Software: RFPA v1.4.19155 340d759

hartytp, 2020-01-28:
Iirc there are some issues with the interlock in v1.3 

Who’s the manufacturer and do you know if they made any mods post production?

jbqubit, 2020-02-04:
This booster was made by Technosystem. They didn't indicate any post production modifications.

hartytp, 2020-02-09:
Duplicate of https://github.com/sinara-hw/Booster/issues/245

hartytp, 2020-02-09:
FWIW Booster v1.3 is still very much a prototype. e.g. it doesn't have the TVS diodes so will die relatively quickly. I'm currently not recommending anyone buys Booster unless they want to participate in testing/bug fixing. Hopefully that will change in a couple of weeks once we've tested some Boosters with the TVS added, but it's important for people to have their expectations set appropriately until we're confident that things work properly.

closed 2020-02-09 by hartytp