sinara-hw/Booster #351: document fan control

hartytp, 2020-02-10:
@wizath how is the fan speed determined? With all channels running and `fanlevel` set to 0, the fans aren't turning on even with all channels at 40C. Is that expected?

hartytp, 2020-02-10:
@dnadlinger I'd probably keep the min fan level at, say 15%-25%. Assuming the temperature diagnostics are remotely accurate, having the fans off won't cause damage (the modules are pretty well heat sunk to the case) but there is no point letting things run hot. 15% fans isn't offensive from a noise perspective.

dnadlinger, 2020-02-10:
@hartytp: So far, I haven't bothered changing any fan-related settings at all – should I be running `fanlevel 20` or something then?

hartytp, 2020-02-11:
As far as I understand it, there is dynamic fan control to adjust the fan speed to maintain a constant temperature at the hottest channel. Or something like that (I'm not aware of the current implementation being documented anywhere). `fanlevel` adjust the minimum fan level, which only has an impact when the chassis is cold. So, it shouldn't really matter.

Empirically, the fans currently never turn on with `fanlevel 0` (perhaps because of heat sinking through the chassis) but the chassis does get a bit warm. `fanlevel 20` will help keep things a little colder without creating much noise, so seems like a good idea.

closed 2020-02-27 by hartytp