sinara-hw/Booster #352: Booster 003B002F3037470535353239 channel 0 death

dnadlinger, 2020-02-11:
Booster 003B002F3037470535353239, just refurbished with TVS diodes, stopped outputting RF on channel 0.

All LEDs are green, and monitoring looks fine. Note non-zero input power (agreeing with reality as per power meter on input), but close to zero output (also in line with reality, didn't measure carefully):
```
PGOOD: 1
FAN SPEED: 50 %
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
HWID            E7:8D   2B:2E   91:DF   91:E5   C6:B0   7B:26   6D:76   E5:92
INPWR [V]       0.78    0.87    0.00    0.56    0.04    0.11    0.03    0.00
TXPWR [V]       0.16    1.24    0.08    0.01    0.13    0.01    0.01    0.08
RFLPWR [V]      0.06    0.95    0.04    0.01    0.13    0.12    0.00    0.15
INPWR [dBm]     -11.30  -9.81   -19.00  -16.25  -19.00  -19.00  nan     -19.00
TXPWR [dBm]     5.00    29.56   5.00    5.00    5.00    5.00    5.00    5.00
RFLPWR [dBm]    -4.84   14.09   -3.69   -3.22   -3.67   -3.95   -4.02   -4.20
I30V [A]        0.052   0.159   0.048   0.048   0.050   0.050   0.052   0.051
I5V0 [A]        0.221   0.229   0.220   0.260   0.229   0.218   0.257   0.225
5V0MP [V]       4.936   4.942   4.930   4.954   4.924   4.936   4.958   4.922
ON              1       1       1       1       1       1       1       1
SON             1       1       1       1       1       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       0       0       0       0
ADC1            261     2026    126     13      207     14      13      130
ADC2            102     1552    71      12      218     191     8       252
INTSET [dBm]    31.38   30.99   30.48   28.97   36.69   35.98   35.48   33.98
DAC1            4095    4095    4095    4095    4095    4095    4095    4095
DAC2            2521    2472    2405    2096    2931    2772    2536    2606
SCALE1          55      57      55      56      56      59      56      54
OFFSET1         424     341     404     314     526     301     374     396
BIASCAL         1453    1309    1475    1201    1357    1383    1391    1247
HWIS            57.67   56.42   57.33   56.75   58.75   59.33   58.17   57.58
HWIO            711.67  723.42  657.33  451.75  775.75  637.33  472.17  649.58
LTEMP           28.00   30.00   29.25   29.25   30.00   29.25   31.00   31.50
RTEMP           29.25   30.00   29.50   29.50   28.00   29.50   29.00   30.25
> logstash
[INFO] hardware revision 4
[INFO] device boot
[INFO] SYSCLK frequency: 168000000
[INFO] PCLK1 frequency: 42000000
[INFO] PCLK2 frequency: 84000000
[INFO] ADC: OK | raw: 1969 | VrefInt 1.20 V
[INFO] PGOOD: OK
[INFO] Found mainboard EEPROM, loading values
[INFO] MACADDR 80:1F:12:5E:33:BF
[INFO] RF Channel at 0 detected, id = E7:8D
[INFO] RF Channel at 1 detected, id = 2B:2E
[INFO] RF Channel at 2 detected, id = 91:DF
[INFO] RF Channel at 3 detected, id = 91:E5
[INFO] RF Channel at 4 detected, id = C6:B0
[INFO] RF Channel at 5 detected, id = 7B:26
[INFO] RF Channel at 6 detected, id = 6D:76
[INFO] RF Channel at 7 detected, id = E5:92
[INFO] waiting for network address..
[INFO] DHCP IP acquired: 10.255.6.77
[INFO] DHCP lease time: 691200 s
[INFO] TCP Server initialization complete
[INFO] network socket state OK
[INFO] network socket (on port 5000) listen OK
[INFO] [tempmgt] Temp 3.054054 (d 20) -> Fan 20
[INFO] network client 10.255.6.202 connected
==============================================================================

> version
RFPA v1.4.1 d0f3d02, built Dec 24 2019 12:49:18, hv rev 4

> devid
[devid] 003B002F3037470535353239
```

I2cdetect seems broken on that firmware version; always get 

```
> i2cdetect 0
[i2c_scan] start
[i2c_scan] end
> i2cerr
                #0      #1      #2      #3      #4      #5      #6      #7
I2C ERR         129     0       0       0       0       0       0       0

no matter the channel.

The channel amplified RF in the experiment briefly, but stopped some hours later (@hartytp: did you ever test it with RF applied as well?)

dnadlinger, 2020-02-12:
On reflection, I'm not sure whether I've ever seen that channel working (unless @hartytp tested it independently) – we might have just been busy looking for the problem elsewhere.

hartytp, 2020-02-12:
@dnadlinger I'm afraid that I never tested this channel, only checked the bias currents. TS told me that they calibrated all channels before shipping, so I'm assuming they tested it prior to shipping.

This is a bit of a mystery to me since the diagnostics all look correct, and I'm surprised to see catastrophic failure without significant changes in the current draw.

I'll follow up with TS and send back to @gkasprow for him to investigate. However, this does not bode well to say the least...

hartytp, 2020-02-12:
@dnadlinger where can I find this dead channel now?

hartytp, 2020-02-12:
I guess if TS didn't verify the RF output, it's possible there is just a short somewhere due to the rework/TVS addition. So, maybe this is just a QC thing rather than an RF thing (still concerning, but not as bad as channels still dying I guess)...

hartytp, 2020-02-12:
@dnadlinger update from @wizath who calibrated/tested this unit....

> Based on information from github - chassis id 003B002F3037470535353239 and channel id 801F125DE78D, the channel was working fine before shipment (attaching calibration report).
Can you remove top cover and see if it helps? There was mechanical issue that caused SUP_CTRL board goldpins to short because of wrong top cover cutouts on older chassis models.

[801F125DE78D.pdf](https://github.com/sinara-hw/Booster/files/4192884/801F125DE78D.pdf)

Can you have a quick go at testing with the lid removed. If it still doesn't amplify then remove it and I'll post it back to @gkasprow

hartytp, 2020-02-13:
@dnadlinger did you have a chance to take the lid off yet?

dnadlinger, 2020-02-14:
Taking off the lid didn't change the behaviour, but immediately revealed the actual issue:
[image]

Note the tracks disappearing right underneath the pan head screws used for mounting the sup ctrl PCB extend. This isn't a subtle issue, see e.g. the one in the top left corner! The mechanical friction when tightening the screws normally is enough to scrape off the solder resist and expose the tracks, potentially shorting them to ground. The shiny bit in the following picture (from another channel on the same Booster) is a piece of exposed copper track:
[image]

On the above channel, both the top left and bottom right screws (in the orientation from the first picture with the silk screen upright) actually caused issues. Replacing the screws with M3 socket cap head bolts restored channel functionality, even when tightened quite aggressively.

Is this a known issue of the v1.0 mechanics? (I've noticed that the lid doesn't quite fit with standard-height hex bolts, so perhaps they were swapped out on assembly?) If no, then this is an embarrassing issue that needs to be fixed on all channels and for future revisions. If yes, why on earth was the v1.0 chassis kept for what is supposed to be the latest-and-greatest version, and why didn't anyone at least stop to re-think this when screwing down the bolts right onto the tracks of the freshly modded boards? Using a known-broken iteration of a system as basis for testing its reliability is a rather unique proposition, and not in a good way.

hartytp, 2020-02-14:
@dnadlinger I'm not aware of this being an issue with the V1.0 boards. It looks like someone's used a silly screw size for this due to lack of care during refurbishment. grrrr.....

@gkasprow please can you make these clearances big enough that we're not sensitive to the kind of issue?

hartytp, 2020-02-14:
Anyway, I'll forward this on to TS and CTI and point out the obvious: don't use screws with large head sizes/washers that will short out traces. That's a very basic QC failure.

closed 2020-02-14 by hartytp

dnadlinger, 2020-02-14:
So did you get confirmation this was a one-off mistake? I didn't want to disconnect/reconnect 16 more SMA cables to check.