sinara-hw/Booster #353: v1.4 RF PA PCB mounting screw clearance

dnadlinger, 2020-02-14:
The bottom-right (in silkscreen direction) mounting screw on the PA PCB is less than a millimeter from touching a MLCC resp. one of its mounting pads:

[image]

(Apologies for the bad picture quality.)

I haven't seen this cause an actual issue, but this is a bit too close for comfort.

Shifting away the whole section a couple of millimeters from the screw should be easy enough (after removing the unused pads.)

dnadlinger, 2020-02-14:
NB: I seem to have re-discovered https://github.com/sinara-hw/Booster/issues/323, but on a slightly different PCB revision and with even less clearance.

hartytp, 2020-02-14:
Thanks for reporting that @dnadlinger 

@gkasprow for v1.5 please can you go over **all** the clearances around the screws and board edge (for the RF boards, supplemental boards and main PCB) and make sure we leave a generous amount of space. The designs aren't that dense and I'd prefer not to have to rely on picking particular screws to avoid these kinds of issues.

hartytp, 2020-02-14:
See also https://github.com/sinara-hw/Booster/issues/352

gkasprow, 2020-03-02:
fixed it by adding screws to the BOM and increasing the components to screws clearance

closed 2020-03-02 by gkasprow