sinara-hw/Booster #354: v1.5 changelog

gkasprow, 2020-02-27:
- fixed #333 - changed voltage of all 100n caps to 50V
- both RF boards merged into one
- fixed #344  - added TVSs
- added fast start circuit #345

gkasprow, 2020-02-27:
I'm waiting with new release for design review.

gkasprow, 2020-03-02:
- added screws to the BOM
- aligned the top and bottom RF boards
- fixed cosmetic issues

hartytp, 2020-03-02:
@gkasprow thanks! Can you close the relevant issues as you go to make tracking progress easier?

gkasprow, 2020-03-02:
R38=33R

closed 2020-03-02 by gkasprow

hartytp, 2020-08-17:
> R38=33R

@gkasprow can you remind me what this change was?

hartytp, 2020-08-17:
I guess R38 is the one on the main board, which we increased from previous revisions? That would just be for better filtering on the FET bias and is not critical AFAICT.

hartytp, 2020-08-17:
Aah, no, it's https://github.com/sinara-hw/Booster/issues/337. Well, missing that doesn't scare me.