sinara-hw/Booster #355: ethernet fallback

hartytp, 2020-02-27:
If we have any problems with the Wiznet chip that we can't easily resolve, we can easily swap out the small PCB with ethernet + LEDs on with a new one with a microprocessor to act as an ethernet <> serial converter. Are there any changes we want to make to v1.5 to make that fallback plan easier? e.g. how would we flash that board?

gkasprow, 2020-02-27:
We would swap the Wiznet with other MAC+PHY chip. THere are a few of them already. it does not change anything to the main board. Only the interfeace board would need to be modified.

hartytp, 2020-02-27:
Okay cool. So you can route the Mac to the microprocessor using the ribbon cable and existing connections then?

gkasprow, 2020-02-27:
Exactly. We can use at least 4 different MAC+PHY chips

hartytp, 2020-02-27:
Cool! In that case we can easily swap if there are issues...

closed 2020-02-27 by hartytp