sinara-hw/Booster #356: Pre-production design review remarks

bjuszczyk, 2020-03-02:
**Common issues:**
- [ ] projects are not compiling correctly, there are minor issues that should be corrected before production 

**Common_Channel_RF:**
- [ ]  No rule for board outline clearance - copper on the edges will be removed during production, thus it has to be verified whether some important connections won't disappear 
- [ ]  Add bean shaped holes at PA_channel_SUP_CTRL and Common_Channel_RF PCBs for better boards adjustation 
- [ ] Increase components clearance (D3-L6; D6-IC8) - component clearance rule is marked as disabled for batch 
- [ ] PCB is not aligned with enclosure - mounting holes not matched  
- [ ] some components have incorrect component links - board variants could be affected  

**Main_board:**
- [ ]  Main_board has 1 difference between schematics and PCB

**PA_channel_SUP_CTRL:**
- [ ]  Add bean shaped holes at PA_channel_SUP_CTRL and Common_Channel_RF PCBs for better boards adjustation

hartytp, 2020-03-02:
@bjuszczyk many thanks for the valuable feedback :)

gkasprow, 2020-03-02:
I added screws to the BOM. This should solve one of the issues forever.

hartytp, 2020-03-02:
> I added screws to the BOM. This should solve one of the issues forever.

Is it still worth double checking the clearances around the screws to see if we can easily increase them? If you're happy that this won't be an issue ever again can you close https://github.com/sinara-hw/Booster/issues/353 please?

gkasprow, 2020-03-02:
the difference was just a component class. I fixed it.
I also aligned the connectors precisely between top and bottom boards. Not that important because they are aligned during soldering, but it just looks better.

hartytp, 2020-03-02:
Nice! Thanks

gkasprow, 2020-03-02:
compilation errors were removed by adding specific ERC.

closed 2020-03-02 by gkasprow