sinara-hw/Booster #357: Dead channel

hartytp, 2020-04-12:
@dnadlinger has reported another dead channel. I'll leave it to him to post the logs etc, but my understanding is that:
1. It's an old Booster mechanics version, but updated to the v1.4 electronics and with the TVS diodes fitted
2 Symptom is that the gain has decreased by ~20dB on one channel
3. All currents (PA/pre-amp) look unchanged

@gkasprow please correct me if I'm wrong, but (3) seems to me to rule out an issue with the amps/biasing -- I don't see how we could get a change in the AC gain without a change in the quiescent current draw.

Assuming that's correct, it has to be one of: issue with the passive matching circuitry; issue with the switch/switch control voltages; short/open somewhere in the signal path.

Failure of the passive matching seems highly unlikely since I've already double checked the voltage ratings of all the caps and we have TVS anyway to prevent large spikes.

Particularly given that this unit has an old version of the mechanics + fiddly rework, it thus seems overwhelmingly likely the issue is mechanical. But, we won't know for sure until we can get back in the lab for an autopsy and probe the signal path with a scope to see what's actually going on (good time to play with our new active probe!).

@gkasprow so that we can debug this without sending it to you, do you have an adapter that we can use to run the channel outside the chassis?

hartytp, 2020-04-12:
One other comment: like all good bugs, it seems likely that the gain changes had multiple causes; there were definitely dead amps, but I some of the symptoms like gain meta-stability without changes in the quiescent current, seem more like mechanical issues (this might also explain why you couldn't reproduce the issues when we sent you the amp modules back and you ran them outside their normal mechanics). In hindsight, I think we maybe got a bit carried away looking for dead FETs and didn't spend enough time looking for other more obvious things...

hartytp, 2020-04-12:
On an optimistic, note, the other thing I'll add is that we've had ~50 channels in heavy use for quite a while now and this is the only suspicious thing we've seen. We've also fixed quite a few mechanical issues in recent mechanics, so hopefully this really is an isolated issue with this channel that's already fixed in newer versions. But, we won't know until we look at it...

dnadlinger, 2020-04-12:
Apologies for the delay – here is the status output:

```
PGOOD: 1
FAN SPEED: 20 %
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1
HWID		E7:8D	2B:2E	91:DF	91:E5	C6:B0	7B:26	6D:76	E5:92
INPWR [V]	0.09	0.00	0.00	0.55	0.05	0.10	0.03	0.00
TXPWR [V]	0.08	0.01	0.08	0.01	0.14	0.01	0.01	0.09
RFLPWR [V]	0.07	0.11	0.04	0.01	0.14	0.09	0.00	0.15
INPWR [dBm]	-19.00	-19.00	-19.00	-16.31	-19.00	-19.00	nan	-19.00
TXPWR [dBm]	5.00	5.00	5.00	5.00	5.00	5.00	5.00	5.00
RFLPWR [dBm]	-4.69	-3.68	-3.76	-3.21	-3.63	-4.41	-4.02	-4.26
I30V [A]	0.055	0.050	0.048	0.051	0.057	0.061	0.000	0.064
I5V0 [A]	0.223	0.231	0.222	0.262	0.231	0.220	0.259	0.228
5V0MP [V]	4.944	4.954	4.938	4.958	4.930	4.946	4.958	4.930
ON		1	1	1	1	1	1	1	1
SON		1	1	1	1	1	1	1	1
IINT		0	0	0	0	0	0	0	0
OINT		0	0	0	0	0	0	0	0
SINT		0	0	0	0	0	0	0	0
ADC1		132	14	132	13	228	14	17	139
ADC2		113	183	65	13	228	153	7	250
INTSET [dBm]	32.36	31.45	30.48	28.97	36.69	35.94	34.47	33.98
DAC1		4095	4095	4095	4095	4095	4095	4095	4095
DAC2		2578	2498	2405	2096	2931	2770	2477	2606
SCALE1		55	57	55	56	56	59	56	54
OFFSET1		424	341	404	314	526	301	374	396
BIASCAL		1453	1309	1475	1201	1357	1383	1391	1247
HWIS		57.67	56.42	57.33	56.75	58.75	59.33	58.17	57.58
HWIO		711.67	723.42	657.33	451.75	775.75	637.33	472.17	649.58
LTEMP		30.25	29.00	30.50	29.00	29.00	29.00	30.00	32.25
RTEMP		28.00	30.00	28.00	30.00	30.25	30.25	31.00	31.00
==============================================================================

> status 6
[status] e=1 s=1 r1=16 r2=8 tx=5.000 rf=-4.037 curr=0.000 t=31.00 i=0.03 ip=nan
```

Channel 6 is the dead one, and draws no current from the 30 V rail. :/

dnadlinger, 2020-04-12:
NB: This is after DML power-cycled the unit (I didn't initially suspect the channel to have died).

While we have units where the input detectors aren't calibrated on any channel, it's at least a funny coincidence that the only channel that reads `nan` is the broken one. (I haven't checked how the code looks – perhaps this can also a be consequence of off-scale values?)

hartytp, 2020-04-12:
@dnadlinger thanks for the detailed.

> Channel 6 is the dead one, and draws no current from the 30 V rail. :/ (Currents being normal was a misunderstanding between DML and me remotely as to which Boosters were connected to USB.)

ACK. Okay, that's a completely different symptom...

Note to self: if the gate bias voltage weren't applied correctly (e.g. due to a short) the current would go high and the LDO would go into fold back limiting. Not sure what the reading would be in that case.

@gkasprow please can you send us an adapter in any case so that we can debug this here?

I'm not sure what to make of that. To do: measure the bias voltage reaching the FET, measure the drain voltage reaching the FET (as close to the FET as possible) and see if these are sensible. If those are sensible but we have no current draw then unfortunately this could only be a dead FET. Let's hope that's not the case...

hartytp, 2020-04-12:
@dnadlinger did you get a chance to run the SCPI logger with the new firmware version? Would be good to see if this still crashes and then report back to Creotech. Debugging would be much much easier if we could gather long-term data on all units.

gkasprow, 2020-04-14:
The adapters are very simple.
 Just take RA 2.54mm header and solder precision 2.54mm socket. In this way, you would be able to install the upper board vertically.
Take two DSUB15HD - you can grab them from old VGA extension cable.
Connect the corresponding pins with 15x 30cm wires, in such way the RF module can be taken outside chassis.

gkasprow, 2020-04-14:
I'd check if 30V LED is on, first. If it is off, it means that the LDO is probably dead. One can also measure it on the connector pins after opening the RF box lid.
If LED is ON (or voltage is present), I'd check the gate voltage. If it is close to 0, it means that the RF FET is dead or some components are not contacting correctly. If it is higher than -3V it mesns that bias generator is not working or is disabled.

hartytp, 2020-04-19:
I really want to look at this one asap. If this turns out to be something relatively innocuous (mechanical issue we've already fixed in recent revisions) I'll feel a lot better.

hartytp, 2020-09-04:
Looking again at this...

```python
>>> for channel in range(8):
...     pprint(dev.get_status(channel))
... 
Status(detected=True, enabled=True, interlock=False, output_power_mu=0, reflected_power_mu=0, I29V=0.057347302, I6V=0.222391579, V5VMP=4.941941202, temp=27.0, output_power=5.0, reflected_power=-4.844155844, input_power=-19.0, fan_speed=20.0, error_occurred=False, hw_id='80:1f:12:5d:e7:8d', i2c_error_count=0)
Status(detected=True, enabled=True, interlock=False, output_power_mu=0, reflected_power_mu=0, I29V=0.048164236, I6V=0.230580161, V5VMP=4.950000048, temp=28.0, output_power=5.0, reflected_power=-3.857142857, input_power=-19.0, fan_speed=20.0, error_occurred=False, hw_id='80:1f:12:5d:2b:2e', i2c_error_count=0)
Status(detected=True, enabled=True, interlock=False, output_power_mu=0, reflected_power_mu=0, I29V=0.049101283, I6V=0.221871673, V5VMP=4.933882654, temp=28.0, output_power=5.0, reflected_power=-3.756410256, input_power=-19.0, fan_speed=20.0, error_occurred=False, hw_id='80:1f:12:5d:91:df', i2c_error_count=0)
Status(detected=True, enabled=True, interlock=False, output_power_mu=0, reflected_power_mu=0, I29V=0.053036884, I6V=0.261384841, V5VMP=4.954029024, temp=28.0, output_power=5.0, reflected_power=-3.207317073, input_power=-16.311475754, fan_speed=20.0, error_occurred=False, hw_id='80:1f:12:5d:91:e5', i2c_error_count=0)
Status(detected=True, enabled=True, interlock=False, output_power_mu=0, reflected_power_mu=0, I29V=0.058284346, I6V=0.230710138, V5VMP=4.925824106, temp=28.0, output_power=5.0, reflected_power=-3.631578947, input_power=-19.0, fan_speed=20.0, error_occurred=False, hw_id='80:1f:12:5d:c6:b0', i2c_error_count=0)
Status(detected=True, enabled=True, interlock=False, output_power_mu=0, reflected_power_mu=0, I29V=0.062407358, I6V=0.220051981, V5VMP=4.937911928, temp=28.0, output_power=5.0, reflected_power=-4.564102564, input_power=-19.0, fan_speed=20.0, error_occurred=False, hw_id='80:1f:12:5d:7b:26', i2c_error_count=0)
Status(detected=True, enabled=True, interlock=False, output_power_mu=0, reflected_power_mu=0, I29V=0.0, I6V=0.258525337, V5VMP=4.958058596, temp=30.25, output_power=5.0, reflected_power=-4.024691358, input_power=nan, fan_speed=20.0, error_occurred=False, hw_id='80:1f:12:56:6d:76', i2c_error_count=0)
Status(detected=True, enabled=True, interlock=False, output_power_mu=0, reflected_power_mu=0, I29V=0.083584625, I6V=0.227070774, V5VMP=4.925824106, temp=28.0, output_power=5.0, reflected_power=-4.189189189, input_power=-19.0, fan_speed=20.0, error_occurred=False, hw_id='80:1f:12:5d:e5:92', i2c_error_count=0)
```

Channel 6 indeed has no current on the 30V rail

hartytp, 2020-09-04:
@gkasprow it looks like there is a short on the 30V power rail. I don't recall seeing this as a failure mode of the FETs before, so I'm inclined to thing that it's eaither: 1) another mechanical issue 2) a failure of one of the pieces of rework when the TVS diodes were added.

I'll post it back to you for postmortem if that's ok?

hartytp, 2020-09-04:
@gkasprow that will go in the post to TS on Monday.

Let's make sure we understand what went wrong with this one and the other one with I2C errors before we produce any more Boosters. It's probably an issue with the old mechanics/a rework issue, but we should confirm that.

gkasprow, 2020-09-09:
I received the box.

gkasprow, 2020-09-09:
This is the status I receive
```

FAN SPEED: 20 %
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
HWID		21:4B	DE:0E	A8:53	5B:EA	FB:68	02:41	B5:46	1F:B8	
INPWR [V]	0.01	0.01	0.01	0.41	0.82	0.54	0.02	0.34	
TXPWR [V]	0.01	0.01	0.01	0.02	0.01	0.01	0.14	0.01	
RFLPWR [V]	0.01	0.07	0.01	0.07	0.19	0.01	0.00	0.01	
INPWR [dBm]	nan	nan	nan	nan	nan	nan	nan	nan	
TXPWR [dBm]	5.00	5.00	5.00	5.00	5.00	5.00	5.00	5.00	
RFLPWR [dBm]	-3.88	-5.28	-4.14	-4.70	-4.06	-2.81	-4.98	-2.84	
I30V [A]	0.048	0.044	0.043	0.041	0.047	0.048	0.040	0.043	
I5V0 [A]	0.230	0.244	0.243	0.235	0.232	0.232	0.231	0.235	
5V0MP [V]	4.947	4.950	4.977	4.976	4.957	4.974	4.948	4.963	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
ADC1		18	18	18	24	18	18	234	17	
ADC2		12	107	12	116	316	13	4	11	
INTSET [dBm]	37.99	37.99	37.99	38.00	37.99	33.00	33.99	29.99	
DAC1		4095	4095	4095	4095	4095	4095	4095	4095	
DAC2		3792	3806	3818	3821	3932	3341	3427	3100	
SCALE1		84	83	83	84	85	85	79	85	
OFFSET1		505	431	567	472	571	430	635	480	
BIASCAL		1149	1283	1271	1327	1473	1577	1327	1335	
HWIS		82.67	86.17	84.83	85.08	86.83	86.00	83.17	85.42	
HWIO		651.67	532.17	594.83	588.08	632.83	503.00	600.17	538.42	
LTEMP		20.00	21.50	21.50	21.50	20.00	20.00	21.50	20.00	
RTEMP		20.00	19.00	20.00	19.00	21.50	21.00	20.00	21.25	
==============================================================================
```

gkasprow, 2020-09-09:
this is the unit with MACADDR 80:1F:12:56:1F:21

gkasprow, 2020-09-09:
@hartytp what is wrong with it? Are you sure this box has issue with 30V rail?

hartytp, 2020-09-09:
The thing I returned to TS was a single booster channel (not a chassis) which should have MAC '80:1f:12:56:6d:76'. What was in the box you received?

gkasprow, 2020-09-09:
I received the entire Booster with Creotech logo

gkasprow, 2020-09-09:
It's marked with the label: LAB3-BOO1

hartytp, 2020-09-09:
@gkasprow aah, okay, there has been a misunderstanding!

That Booster unit was sent to TS for refurbishment (it's the last one we have without TVS diodes). That was sent the day before the single channel with the 30V fault. Sorry for the confusion!

gkasprow, 2020-09-11:
I received the broken channel from TS. The 30V rail is indeed shortened to GND. What is intriguing, the FET gate is shorted to GND with 4Ohm resistance. When I apply a negative voltage to the gate (-3V), the drain is still conducting. It means permanent damage to the transistor.  The TVSs are fitted correctly. I checked their top marks, they are valid components.
I noticed some floating metal debris of 2..3mm length - these are pieces of tin, gold metalization and aluminum.
The solder on the transistor pads looks overheated.
The transistor thermal pad was wetted correctly.
I don't remember FET being damaged in such a way. Even in first prototypes, when LDO overheated and regulator BJT breakdown, which resulted in a high current surge through the FET drain, it did not get damaged. It lost some gain, but not a permanent short circuit!
It's quite possible that TVSs were fitted to the FET which was already somehow broken during tests. Maybe these metal debris caused some short-circuit, but I can't imagine how it would damage the FET, with so many protection features.

hartytp, 2020-09-12:
hmmm...let's discuss this with Pawel.

gkasprow, 2020-09-13:
I will bring entire Booster + broken channel to TS on Tuesday.

hartytp, 2020-09-15:
Okay, this is not particularly satisfying.

- The FET in this unit is dead
- In general, TS have been strict about replacing all three amp chips in each Booster channel when fitting the TVS didoes to make sure the amps can't be exposed to damaging conditions. With these chips and the 29V LDO foldback limiting, it should be pretty difficult to damage a channel. However...
- Greg recognizes the soldering in this unit as being his own. That suggests that this channel may have been the one that Greg stress tested while playing with TVS diodes, so may have been quite badly damaged before the diodes were fitted and just took some time to fail. TS think this is quite possible. (FWIW, the enclosure for this module is visibly different to all the other channels in the chassis, suggesting that it wasn't from the same batch).

It's hard to see what else we can do here other than fit new amps and return the unit to its chassis.

Given that it's the only channel that's died now after some months of running, it does seem plausible that there is an issue with this specific channel rather than the overall design, but it's a shame we don't have better records of the rework applied to each unit...

closed 2020-09-15 by hartytp