sinara-hw/Booster #359: I2C errors on 002600213436470832383539; also takes down SCPI interface

dnadlinger, 2020-04-19:
As per the title, one of our refurbished v1.4 Boosters has developed an I2C error after a long while of uneventful continuous operation:

```
> logstash
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_BUSY error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_BUSY error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_BUSY error ch 6
[ERROR] Disabling channel 1 due to I2C error
[ERROR] I2C_TRANSMIT_BUSY error ch 6
> devid
[devid] 002600213436470832383539
> version
RFPA v1.4.1 d0f3d02, built Dec 24 2019 12:49:18, hv rev 3
> task-stats
Task		State	Priority	Stack	#
**************************************************
CLI            	R	1		462	5
IDLE           	R	0		490	8
FAN            	B	0		535	1
RF Interlock   	B	4		335	7
CH MEAS        	B	2		491	9
UDPServer      	S	3		2149	4
DHCP           	S	1		325	3
ADC            	S	3		717	2
Ext            	S	2		478	6
CH INFO        	S	1		2549	10
> i2cerr
		#0	#1	#2	#3	#4	#5	#6	#7
I2C ERR		0	0	0	0	0	0	4466654	0
```

```
PGOOD: 1
FAN SPEED: 20 %
CHANNELS INFO
==============================================================================
		#0> 	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1
HWID		B1:CF	1F:0B	3C:9C	90:A2	FE:82	42:66	5B:BF	54:57
INPWR [V]	0.47	0.87	0.06	0.01	0.03	0.09	0.81	0.72
TXPWR [V]	0.09	0.02	0.01	0.04	0.17	0.05	0.04	0.02
RFLPWR [V]	0.01	0.01	0.01	0.01	0.01	0.01	0.00	0.01
INPWR [dBm]	nan	nan	nan	nan	nan	nan	-8.94	-12.82
TXPWR [dBm]	25.26	31.15	5.00	5.00	5.00	5.00	24.57	27.36
RFLPWR [dBm]	14.85	18.23	-3.76	-3.11	-4.43	-3.71	13.91	11.76
I30V [A]	0.079	0.193	0.047	0.047	0.040	0.047	0.045	0.107
I5V0 [A]	0.231	0.265	0.262	0.248	0.249	0.260	0.266	0.272
5V0MP [V]	4.988	4.969	4.971	4.973	4.960	4.965	4.944	4.965
ON		1	1	1	1	1	1	0	1
SON		1	1	1	1	1	1	0	1
IINT		0	0	0	0	0	0	0	0
OINT		0	0	0	0	0	0	0	0
SINT		0	0	0	0	0	0	0	0
ADC1		141	30	12	71	271	82	72	28
ADC2		23	12	12	24	11	12	4	12
INTSET [dBm]	34.90	35.95	32.74	32.99	28.98	28.98	35.46	35.99
DAC1		4095	4095	4095	4095	4095	4095	4095	4095
DAC2		2640	2110	3558	3593	2628	2766	3126	3164
SCALE1		53	48	85	85	68	72	68	74
OFFSET1		403	227	502	505	590	526	542	399
BIASCAL		1101	1451	1777	2087	1731	1379	1861	1317
HWIS		57.25	46.17	82.42	84.42	69.92	75.25	71.83	70.92
HWIO		642.25	450.17	859.42	808.42	601.92	585.25	578.83	611.92
LTEMP		29.25	29.50	28.00	30.25	29.00	29.00	29.25	30.50
RTEMP		27.00	28.00	29.25	28.00	29.50	29.25	29.50	29.00
==============================================================================
```

This has also stopped it responding on port 5000 (which is not ideal, as it affects the other channels as well).

dnadlinger, 2020-04-19:
Apparently, trying to `enable 64` was a bad idea, as the VCP interface now hangs too and I can't try to reboot it remotely.

hartytp, 2020-04-19:
Bleurgh

dnadlinger, 2020-04-19:
Interestingly, this is the same channel as on the other Booster that failed (i.e. Alice/Bob twins, with very similar usage patterns, including continued rapid toggling of the RF for a long time due to background subtraction). It's probably just by chance (1 in 39), though, and the faults might turn out to be completely dissimilar.

hartytp, 2020-04-19:
> It's probably just by chance (1 in 39), though, and the faults might turn out to be completely dissimilar.

I don't see an obvious way that usage would affect I2C errors so long as you're not tripping / clearing interlocks often, so this would seem unrelated to the other issue. "Just" flaky firmware, no? If you have enough faults you find some correlations.

hartytp, 2020-04-19:
When do you think you'll be able to get in to power cycle it?

dnadlinger, 2020-04-19:
DML wanted to go in tomorrow, I think.

hartytp, 2020-04-22:
@dnadlinger did the reset go ok? Is this firmware or hardware...

dnadlinger, 2020-04-22:
> Is this firmware or hardware...

That's pretty impossible to say; it was transient though.

hartytp, 2020-04-22:
ACK. Well the main thing is that it wasn't an issue with the RF or bias paths (which is really what I meant). Hopefully, https://github.com/sinara-hw/Booster/issues/357 does turn out to be a mechanical/rework issue that's already fixed for the next HW revision and we can put that mess behind us.

There is definitely a firmware issue here (an I2C error should not take down the SCPI interface under any circumstances) and a firmware issue could explain everything you see easily -- particularly given the history of I2C firwmare bugs.

hartytp, 2020-09-04:
Most likely software/can't reproduce. Closing as the current firmware is now a dead end

closed 2020-09-04 by hartytp