sinara-hw/Booster #362: General status of project

gji, 2020-05-26:
I'm looking into this RF amplifier box for a cold atom experiment. Based on the wiki, it seems like there are suppliers, some specs, and things are working. Scanning through some of the issues it seems like there are still problems that make this less than ideal to use in the lab. Is this product at the state where it's advisable to hold off on purchasing?

hartytp, 2020-05-27:
Hi @gji!

That's a reasonable summary of the current situation: since early revisions, the analog performance (gain stability, phase noise, cross-talk, etc) of Booster has been really good, but we've had two kinds of issues: frimware and reliability.

On the firmware front: the current firmware is still unreliable and prone to crashing. We're in discussions with M-Labs/Quartiq about getting high-quality firmware written, but I can't give you an eta for that yet. Until that's done, Booster is usable but somewhat annoying.

On the reliability front: we had quite a few channels of Booster die during development. We believe that we've now identified and resolved all the issues that caused this. Our lab has had ~50 channels in continuous use for a few months now. One channel seems to have died, but the others are all working without issue. Because of the pandemic we haven't had a chance to investigate what caused that one channel to stop working. Hopefully it's something silly like a mechanical issue (it's an old unit that has been patched repeatedly to bring it to the latest design revision) but we can't be sure until we get back into the lab and have a look.

tl;dr If you want something that works reliably out of the box then I'd suggest holding off for a bit

closed 2020-05-27 by hartytp

gji, 2020-05-28:
I see, thanks! Firmware bugs aren't a showstopper to us buying a test unit, especially if updates are eventually planned. What's more of an issue for cold atom experiments are the spikes in the phase/amplitude noise plot in #95 (thought the noise floor looks really nice) - the timescales in our lattice means we care about anything between 1 Hz and 1 MHz. I'll comment in that issue if that makes more sense, but do you know if newer models have fixed that? The spikes around 1 kHz really look like a fan connected to the supply line. I guess the RBW is quite small so maybe it's really not a problem.

We might just purchase a unit and play around with it, as we have some experience hunting down spurs.

gkasprow, 2020-05-28:
Any contribution to the project is very welcome. The more users, the better the project will be. The fans are PWM-modulated and this can be changed by SW.

hartytp, 2020-05-28:
@gji Have a look at the spur measurement [here](https://github.com/sinara-hw/Booster/issues/2#issuecomment-384916098). There aren't many close-in spurs. The worse ones I measured in that data set were ~-100dBc -- and subsequent measurements suggested that these spurs were present on the signal source I was using, so take that as an upper bound.

Do your experiments need better performance than that?

Re the phase noise plots you mention:
- I don't recall what the measurement bandwidth was, so one has to be careful translating that data to spur heights for a narrow-band signal
- we weren't too careful with grounding in that measurement and my experience has generally been that to remove spurs at the -100dBc level or below, one has really has to watch the grounding. So, again, I'd take that as an upper limit
- we have added a bit more filtering on the power and bias lines since that revision (mainly to further suppress cross-talk, but should also help spurs) so the current revision may be better

Anyway, those are just some thoughts off the top of my head but, as Greg says, more data and more users would be great!

hartytp, 2020-05-28:
> The spikes around 1 kHz really look like a fan connected to the supply line. I guess the RBW is quite small so maybe it's really not a problem.

If you feel like digging into the spurs then go for it. As Greg says, we can easily test that hypothesis by turning the fans off temporarily and seeing if the spurs change. I suspect this is not fan related since the fans are run from a different supply to the main PA and there is heavy filtering (both LDOs and RF filtering) on all supply/bias lines.

The other suspicion I had was RFI from external sources getting mixed up. For us -100dBc isn't an issue so we didn't bother chasing this down.

hartytp, 2020-05-28:
FWIW, I actually suspect we can consider scrapping the fans in future if we want to reduce the cost of Booster.

The total power consumption is something like 2W per channel (PA is 50mA*30V=1.5W), so 16W total. The RF modules, where all the power is dissipated, are really well heat sunk to the chassis. I did try disabling the fans at one point. The chassis got pleasantly warm to the touch, but that's about it.

gji, 2020-05-28:
Ah I see, yeah the NSD plot is tricky to decipher - it looks like those peaks at 200kHz have a lot more energy. While I can't say for certain we'd care at the -100dBc level, we have seen effects from spurs at the -80dBc level caused by RF amplifiers using poorly-filtered switching supplies for negative bias voltages. 

I'll go ahead and purchase a unit, I'm sure we'll find a use for them as not all of our applications are so sensitive.