sinara-hw/Booster #364: new Ethernet MAC+PHY

gkasprow, 2020-08-15:
@sbourdeauducq please check the schematics of the new Boster User Interface board with ENC424J600.
The reset line is not used anymore.
[Booster_User_IFC.PDF](https://github.com/sinara-hw/Booster/files/5078603/Booster_User_IFC.PDF)

gkasprow, 2020-08-15:
@akaminska when Sebastien confirms, we can produce these modules and send them to Mlabs.

HarryMakes, 2020-08-17:
Hi @gkasprow, page 15 of the [ENC424J600 datasheet](http://ww1.microchip.com/downloads/en/devicedoc/39935b.pdf) seems to suggest that the Eth-SPI controller has a weak internal pull-up (2.2V) on the CSn pin when SPI mode is selected, so is the 12k4 pull-up resistor on SPI1_NSS still needed?

And to select SPI as the interface, the datasheet suggests a <=100k pull-up resistor for INTn/SPISEL (a pin multiplexed with the SPI select function SPISEL, and the interrupt function INTn), which I suspect is to replace an active logic high output from the microcontroller. If the interrupt function INTn is to be used for Booster, then the microcontroller side pin (PG6) indeed needs to be left floating such that the Eth controller can drive it low when there's an interrupt. 

However, when I was developing the Pmod NIC100 module with ENC424J600, I didn't intend to use interrupt so I chose to drive the SPISEL pin from the microcontroller instead of leaving it floating - it works. The [NIC100 datasheet](https://reference.digilentinc.com/_media/reference/pmod/pmodnic100/pmodnic100_rm.pdf) also suggests that this pin can be left floating upon startup in order to select SPI, but this doesn't seem to be explicitly stated on the ENC424J600 datasheet. So I'm not sure if it is best to pull-up INTn/SPISEL pin or not.

gkasprow, 2020-08-18:
@HarryMakes thanks for review. I know there is weak pullup, but generally I don't trust such pullups - they may stop working with time when dirt makes the current patk. Since we have 12k1 already in the BOM, I added it.
I added the SIN/SPISEL 12k1 pullup.

hartytp, 2020-09-04:
@jordens did you decide to go for this chip in the end, or stick with the wiznet?

jordens, 2020-09-04:
Stick with the wiznet for now.

hartytp, 2020-09-04:
fine by me if that's what you prefer as the developer. Let's close this issue then.

gkasprow, 2020-09-04:
anyway, we are producing 10pcs currently. You can get Booster with any of them or even both :)

sbourdeauducq, 2020-09-05:
Thanks @gkasprow I plan to continue with the new module.

gkasprow, 2022-02-07:
both versions are available so closing

closed 2022-02-07 by gkasprow