sinara-hw/Booster #365: v1.3 changes

hartytp, 2020-08-17:
@gkasprow I thought we'd had all of our Booster electronics (not mechanics) upgraded to v1.4 but it appears that some of our Boosters still use patched v1.3 boards  (e.g. diodes fitted and interlocks patched). I'm trying to do the git hub archeology to see what the differences really are to see if the patches mean the v1.3 boards are now functionally equivalent to v1.5.

Could you do me a favor and confirm whether *all changes from v1.3 to v1.5* are listed in these issues:
- https://github.com/sinara-hw/Booster/issues/354
- https://github.com/sinara-hw/Booster/issues/331
- https://github.com/sinara-hw/Booster/issues/209

hartytp, 2020-08-17:
AFAICT the v1.4rc2 -> v1.5 changes (other than the TVS diodes) don't seem particularly interesting. The LT3045 missing fast start means that the FET bias will come on quite slowly when the device first powers but I suspect that it will come on enough to prevent damage to anything.

hartytp, 2020-08-17:
> The revision 1.41 is purely cosmetic. No changes to electric schematics nor BOM. Cleaning out a few issues:

so no need to worry there either.

hartytp, 2020-08-17:
>> @gkasprow is this change list complete? If so, I think that there are no changes to the supplemental control board or RF modules apart from the ones that Creotech have already fixed on our v1.3 boards. So, I think that if we want to fix all known bugs at some point in the future, we can do that by just swapping out the main PCBs and not touching the RF modules. Is that correct?

> that's correct. We maintain compatibility between v1.3 and v1.4 .

So, no RF module changes between these two hw revisions. There do seem to be a few changes

gkasprow, 2020-08-17:
It looks like this is a complete list.
The soft-start functionality is not critical. The only issue that may potentially matter in patched v1.3 HW is the voltage rating of C6. But the effect will be binary - either work properly or short circuit the 28V rail, but LDO will survive it.
I'm just looking at the v1.3 design and the C6 has three PN specified
[image]

All of them are 50V rated.

in rev1.4 it was replaced by GRM155R61H104KE19D which is also 50V rated

hartytp, 2020-08-17:
Yeah, looks like there were a lot of patches for things that were marginally stable on the main board (LDOs not quite having enough capacitance). But, I suspect we will get away with those (worst case they'd cause some low-level spurs, not catastrophic failure so I'm less concerned)

hartytp, 2020-08-17:
The ADCs were run outside their abs max voltage, so it's quite possible they will fail after some time.

The big change was that we changed the current-sense InAmps to completely different ones. This breaks software compatibility. The Quartiq firmware is currently not funded to support v1.3 so we'll need to decide what to do about this.

hartytp, 2020-08-17:
@gkasprow summarising then, am I right in thinking that for our v1.3 units with TVS and patched interlock, the RF modules do indeed seem to be equivalent to v1.5. So, if we want to upgrade the unit we'd just need to swap out the main PCBs (which is quite easy afaict).

gkasprow, 2020-08-17:
yes.

hartytp, 2020-08-17:
PS those detailed changelogs made this process much easier than I was expecting. Thanks for that!

closed 2020-08-17 by hartytp