sinara-hw/Booster #366: ADS7924 AVDD supply lower than expected

ryan-summers, 2020-08-19:
While working with booster, I was noticing that the reported measurement for P5V0 from the ADS7924 channel supply monitor was appearing low.

After further investigation, it appears that this is caused by a below-expected voltage on AVDD being supplied to the ADS7924.

Measuring voltage on the 3V3 side of the referenced resistor measures ~3.4V
[image]

On the AVDD supply side, the voltage measures 2.65V.

This only occurs when the ADS7924 is configured for "auto-scan without sleep" conversion mode. If the ADS7924 is configured to use sleep, this issue is not present.

jordens, 2020-08-19:
The datasheet wants 5-10 R here.
This causes two issues:

* Abs max rating violation of DVDD <= AVDD
* Gain error (even at minimum AVDD current it's already 20 mV error, at max conversion rate it's the above mentioned 0.8 V)

hartytp, 2020-08-19:
Thanks both for the detailed testing of this.

I assume that resistance was chosen based on the 8uA max current consumption specified in the [data sheet](https://www.ti.com/lit/ds/symlink/ads7924.pdf?ts=1597838142265&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FADS7924)  (2.5ms cycle time)

[image]

This resistor is way bigger than necessary, so can easily be decreased in the next revision as you suggest.

Can we work around this satisfactorily for now in firmware by slowing down the cycle time?

hartytp, 2020-08-19:
(this probably explains some of the issues that @wizath had with this ADC...)

ryan-summers, 2020-08-19:
There's a firmware workaround by using the auto-scan-with-sleep function. The key here is that the voltage rail only sags if we have the ADC sampling at full-bore.

gkasprow, 2022-02-07:
replaced with 10R

closed 2022-02-07 by gkasprow